robotology-legacy/gym-ignition #5: Unmet dependencies for JointForce.hh, JointPosition.hh

gonzalezJohnas, 2019-03-20:
While compiling I got this unmet dependencies however I have the libignition-gazebo installed, I checked also in the different branches of the repository  libignition-gazebo and couldn't find the files.

Which version of libignition are you using ?

diegoferigo, 2019-03-20:
This is the perfect example that falls in the most particular case described in the [README](https://github.com/robotology/gym-ignition/blob/6c59b013f8e031aad4b76a0368f718173b2cea5a/README.md#setup):

> We are working upon their latest development branches, and **in some cases we have own forks to implement specific features**.

Waiting that the developers upstream add the support for setting joint positions, I implemented it myself [here](https://github.com/diegoferigo/ign-gazebo). This is just a first attempt, it works fine but it is still a kind of a workaround.

If you're brave enough to build everything from sources, you can get useful information from [my development environment](https://github.com/diegoferigo/development-iit/blob/ad38337721ccfabdc35287d8b572c8aac332a69d/RL/Dockerfile#L140).

diegoferigo, 2019-04-04:
I think your question has been replied, feel free to comment or open a new issue if you need any other information.

Closing.

closed 2019-04-04 by diegoferigo