robotology-legacy/gym-ignition #7: Ignition robotics libraries and python: plugins symbol resolution problems

diegoferigo, 2019-03-22:
This issue reports a quite tricky behavior and involves an advanced knowledge of how a C++ software is assembled. First, I try to describe briefly the sequence of operations that are executed when environments are used from python, and provide some information of the architecture of upstream Ignition Gazebo. Second, I describe the problem I encountered and an explanation that @traversaro and me though about it. Last, I present how I solved the problem and comment possible issues that we might face in the future.  

### Architecture overview

The current initial support to use `gym-ignition` from other languages, python in this case, is implemented with [swig](http://swig.org/). Swig is responsible to create bindings of the C++ classes that form this project. Since the usage of the bindings is not very python-idiomatic, we also provide a [gym_ignition](https://github.com/robotology/gym-ignition/gym_ignition/) python package that implements the `gym.Env` interface of OpenAI gym. In this way, the bindings are hidden to the users, and we are more free to change them accordingly to our needs (potentially also using something different than swig).

When the users use the provided class, the python module created with swig is imported, and the shared library that contains the wrapping functions is loaded in memory. These wrapping functions then directly call the C++ classes of `gympp`.

If the aim is to use the environments running inside Gazebo (the only implementation currently available), the dynamic loader of the operating system at one point has to load the libraries that form ignition robotics in memory. This should be clear and straightforward. What's important to know for what follows is that the architecture implemented in Gazebo for adding `System`s to a simulation exploits a plugin-based pattern (shared libraries loaded during runtime, e.g. using `dlopen` on GNU/Linux and macOS). The environments provided by `gym-ignition` are implemented as gazebo plugins.

### Problem description

Calling the same code that creates and uses a `gympp::Environment` object works if called from a C++ executable, but fails if called from python.

The failure is related to a portion of code of the ignition libraries where a singleton is involved (particularly about the creation of the storage of a component of the ECM that involves a Factory class). Unfortunately, the coupling of singleton and plugins is extremely delicate.

I found out that, when called from python, the instance of this factory, that should be global ( i.e. shared by the whole program) was instead local to the plugins. In other words, the pointer to this factory object was not unique. This creates many problems and breaks the logic of components registration.

Ignition plugins are handled by [`ign-plugin`](https://bitbucket.org/ignitionrobotics/ign-plugin). They use [`dlopen(path, RTLD_LAZY | RTLD_LOCAL)`](https://bitbucket.org/ignitionrobotics/ign-plugin/src/911d4fbe6da191455a32c9aefe4b43f7bac02d5a/loader/src/Loader.cc#lines-457) to load plugins, but despite the LOCAL resolution, in C++ everything works fine.

The only difference of the code executed from C++ and python is that the executable in the first case links directly against the ignition libraries, which are loaded _before_ that execution starts (and, consequently, before the plugins are loaded). In the second case, instead, the ignition libraries are loaded by swig. Now, even though I'm not 100% sure of what follows, swig from python loads the bindings library using `RTLD_LOCAL` (not sure it due to the default python behavior or hardcoded somewhere). This means that the ignition libraries which are transitively loaded, have a local resolution. Then, later on during the execution of the environment, when `gympp` loads the plugins, `dlopen` re-loads the ignition libraries, since they were not global in the first place. And this might explain why the factory object is not shared.

### Solution

If the ignition libraries were already loaded in memory as global _before_ the plugins are loaded, `dlopen` would directly use them and it would not attempt to load them again. Changing the default behavior `dlopen` in python is possible using [setdlopenflags], and this workaround does the trick :tada: More in detail, we should alter the flags _before_ loading the `gympp` python module of the swig bindings:

```python
sys.setdlopenflags(sys.getdlopenflags() | os.RTLD_GLOBAL)
``` 

#### Sources:

- http://man7.org/linux/man-pages/man3/dlopen.3.html
- https://stackoverflow.com/questions/7195656/linking-scope-of-loaded-symbols-when-using-swig-to-wrap-c-for-python
- https://sourceforge.net/p/swig/mailman/message/9894693/

[setdlopenflags]: https://docs.python.org/3/library/sys.html#sys.setdlopenflags

traversaro, 2019-03-22:
> Now, even though I'm not 100% sure of what follows, swig from python loads the bindings library using RTLD_LOCAL (not sure it due to the default python behavior or hardcoded somewhere).

For reference, I think the Python-code related to dlopen is in https://github.com/python/cpython/blob/338d54f0a59dc5e5b6c9e7397340169f3a3f8ea4/Python/dynload_shlib.c#L66

closed 2019-03-22 by diegoferigo

diegoferigo, 2019-03-26:
The complete solution we are adopting is the following:

```python
import sys
if sys.platform.startswith('linux') or sys.platform.startswith('darwin'):
    import os
    sys.setdlopenflags(sys.getdlopenflags() | os.RTLD_GLOBAL)
import gympp
```

diegoferigo, 2019-03-28:
I just had related dlopen problems when `gym_ignition` is used together with `tensorflow`. If both modules are imported, `import gym_ignition` must be placed **after** `import tensorflow`.

After reading https://github.com/tensorflow/tensorflow/issues/6568, I suspect that the workaround reported in this issue for `gym_ignition` interferes somehow with the internal logic of `tensorflow`.

traversaro, 2019-03-28:
A slightly less invasive workaround may be store the original value of `sys.getdlopenflags()`, and restore it once `gympp` has been loaded.

diegoferigo, 2019-03-28:
That was my first attempt. I was able to load the `gympp` module, but although I did reset the flags, I was still having segfaults. I actually discovered by chance after many tries that the order was important. It might be worth to investigate this more, but for the time being let's apply this constraint of the import order.

traversaro, 2019-09-26:
Related issue: 
* https://github.com/RobotLocomotion/drake/pull/12108
* https://github.com/RobotLocomotion/drake/issues/12073
* https://github.com/pytorch/pytorch/issues/3059

diegoferigo, 2020-12-02:
Another related issue https://github.com/casadi/casadi/issues/1012