robotology-legacy/gym-ignition #21: Singletons and plugins: dlclose affects singleton state

diegoferigo, 2019-05-09:
As described already in #7, the combo singleton plus plugins (plus python) is quite delicate. I am facing a very subtle problem when multiple environment are sequentially created in python in the same process. Take as example the following `pytest`, that creates 2 identical environments, overriding the first `env` variable and hence closing the resources of the first one:

```python
import gym
import gym_ignition
from gym_ignition.utils import logger

# Set gym verbosity
logger.set_level(gym.logger.WARN)

def test_create_cpp_environment():
     # Environment #1
    env = gym.make("CartPoleIgnition-v0")
    assert env, "Failed to create CartPoleIgnition-v0 environment"

    action = env.action_space.sample()
    assert isinstance(action, int), "The sampled action is empty"

    observation = env.observation_space.sample()
    assert observation.size > 0, "The sampled observation is empty"

    observation = env.reset()
    assert observation.size > 0, "The observation is empty"

    state, reward, done, _ = env.step(action)
    assert state.size > 0, "The environment didn't return a valid state"

    # Environment #2
    env = gym.make("CartPoleIgnition-v0")
    assert env, "Failed to create CartPoleIgnition-v0 environment"

    action = env.action_space.sample()
    assert isinstance(action, int), "The sampled action is empty"

    observation = env.observation_space.sample()
    assert observation.size > 0, "The sampled observation is empty"

    observation = env.reset()
    assert observation.size > 0, "The observation is empty"

    state, reward, done, _ = env.step(action)
    assert state.size > 0, "The environment didn't return a valid state"
```

In this case the system allocates two different `IgnitionEnvironment` objects containing the gazebo server, which under the hood `dlopen`s the same plugin library. This means that two instances of the plugin are loaded in the same time. Beyond this, there's a `EnvironmentCallbacksSingleton` object, contained in a shared library, that stores resources of the plugins.

It can be expected that the shared library containing the singleton stays loaded for the entire lifecycle of both plugins. However, when the first one is deleted, the `dlclose` also calls the singleton destructor (which is a very weird behavior, singleton should not be destroyed by design). This behaviour breaks the second environment, which is still alive.

**Solution:** Remove the pimpl pattern from the `EnvironmentCallbacksSingleton` such that the destructor can be called without causing harm, and preserve the state even after possible destructor call by converting all the private attributes to `static` (there's just one).

cc @traversaro Not sure if you want to add any detail, particularly about you investigation on `cxa_atexit`

traversaro, 2019-05-09:
If someone wants to investigate more in detail this, I add some useful resources in  this comment.

## Gazebo Classic related issues 
In Gazebo Classic, dlclose is never called on the loaded plugins (see https://bitbucket.org/osrf/gazebo/src/7c2936c66a23e6c6707fe4c7790458e75824407e/gazebo/common/Plugin.hh#lines-103). This is probably connected to some problem similar to this, see related issues: 
* https://bitbucket.org/osrf/gazebo/issues/1026/osx-seg-faults-during-plugin-teardown
* https://bitbucket.org/osrf/gazebo/issues/1066

## ISO C++ faqs
ISO C++ faqs' question "What’s the “static initialization order ‘fiasco’ (problem)”?" and following questions : https://isocpp.org/wiki/faq/ctors#static-init-order

traversaro, 2019-05-09:
### `__cxa_atexit` 
> However, when the first one is deleted, the dlclose also calls the singleton destructor (which is a very weird behavior, singleton should not be destroyed by design).

For complete clarity, apparently global/static objects contained  or linked in a library loaded via `dlopen` have  their destructor registered (on system that follow the Itanium ABI)  with the `__cxa_atexit` function (see https://itanium-cxx-abi.github.io/cxx-abi/abi.html#dso-dtor-runtime-api).

## Nifty counter 
I am not completely sure, but the "Nifty" or "Schwarz" counter seems to be a pattern related to this problem
* https://en.wikibooks.org/wiki/More_C%2B%2B_Idioms/Nifty_Counter
* https://stackoverflow.com/questions/9251763/c-static-initialization-via-schwartz-counter/9252072#9252072

## `P1247R0`
The C++ wg21 proposal "Disabling static destructors: introducing `no_destroy` and `always_destroy` attributes seems to be related to this problem as well: http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2018/p1247r0.html .

PR #22 (2019-05-13): Fix dlclose and plugins
Fixes #21
closed 2019-05-13 by diegoferigo