robotology-legacy/gym-ignition #31: Investigate color support in mesh files

diegoferigo, 2019-05-31:
As anticipated in https://github.com/robotology/gym-ignition/pull/4, the colors are not properly displayed in out iCub models.

traversaro, 2019-05-31:
The graphic engine is OGRE or OGRE2 ?

diegoferigo, 2019-05-31:
I tried both, same result. Upstream they changed the scene plugin during the past 10 days (new one is `Scene3D`) but it segfaults. Probably something is wrong in my compilation from sources of the entire suite. I tried their binaries but they are not yet updated.

traversaro, 2019-05-31:
Which source of light do you have in your scene? If you just have environment light and no point or direction source of light, a uniform color for the meshes is actually the expected outcome of the rendering process.

diegoferigo, 2019-05-31:
I had the same suspect as well. The new world upstream added new light elements and I tried to do the same on our worlds. The only difference is that I can see the iCub shadow:

[image]

But note that this is using the old `Scene3D` plugin.

traversaro, 2019-05-31:
Have you tried rotating the model? If the model is in shadow w.r.t. to the only source of light, the only remaining light is indeed the diffuse one.

diegoferigo, 2019-05-31:
Just tried, no change. Good tentative, though.

traversaro, 2019-06-01:
Things I would try to investigate more the problem: 
* A simple cube whose visual shape is given by primitive SDF shapes, instead of an external mesh (such as https://bitbucket.org/ignitionrobotics/ign-gazebo/src/default/test/media/test_model.sdf)
* A simple cube loaded from an STL model, such as the one in https://github.com/xml3d/xml3d-stl-plugin/blob/master/examples/resources/cube.stl 
* Try the simple cube converted in dae via Blender ( [cube.dae.zip](https://github.com/robotology/gym-ignition/files/3244035/cube.dae.zip) )
* A simple collada mesh model, something like the duck in https://bitbucket.org/ignitionrobotics/ign-gazebo/src/default/test/worlds/mesh.sdf

diegoferigo, 2019-12-12:
I wanted to give a quick shot to this problem. I tried to add the following to each visual elements of the links:

```xml
<material>
    <diffuse>1 1 1 1</diffuse>
</material>
```

And this is the result:

[image]

I'm not an expert on this, but as far as I know STL files do not support defining material properties. Probably the default behavior of the gazebo renderer changed, because on Gazebo Classic the same model is properly rendered (all white with shadows, similar to the result of the gif above).

traversaro, 2019-12-22:
So, it a nutshell in ignition-gazebo it seems that the default material for primitive shapes is different from the default material used for STL meshes. This sounds like a bug in ign-gazebo or some dependent library.

diegoferigo, 2019-12-22:
Yes, something similar. I couldn't find the source of the problem, but I'm not yet familiar with all the rendering-related codebase of the simulator.

diegoferigo, 2020-02-14:
We adopted the `diffuse` elements workaround, as reported in https://github.com/dic-iit/gym-ignition-models/pull/1 and particularly https://github.com/dic-iit/gym-ignition-models/pull/1/commits/4e8b9404e73917ef69b7252f07502788a6da4596.

We experienced a mesh problem also with the Panda model. The model was correctly inserted in the simulation but the visual elements were transparent. Citing @paolo-viceconte:

> I discovered that both in https://github.com/mkrizmancic/franka_gazebo and in https://github.com/cpezzato/panda_simulation the visual meshes (i.e. all the *.dae files) have the transparent tag set to A_ONE, which seems to be ok only in old versions of gazebo. One possible solution is to set it to RGB_ZERO, as detailed in http://answers.gazebosim.org/question/20068/collada-model-is-not-visible-in-gazebo-89-if-transparent-a_one/.

Closing.

closed 2020-02-14 by diegoferigo