robotology-legacy/gym-ignition #33: Discussion on the Robot abstraction

diegoferigo, 2019-06-24:
Currently the Robot interface has only the support of getting and setting joint values. It is enough for the simple environments we have developed, but definitely it has to be extended. I open this issue as placeholder for the discussion about how to structure the interface.

An initial list of feature we need to have is the following:

- Joints (almost finalized)
  - Angle
  - Velocity
  - Acceleration
  - PID
- Links
  - Pose (frame orientation + position)
- Sensors
  - IMU
  - Camera
  - F/T (not yet supported by Ignition Gazebo)
- Other
  - Floating base
  - Contact wrenches

The existing interfaces are the following: [`gympp::Robot`](https://github.com/robotology/gym-ignition/blob/master/gympp/include/gympp/Robot.h) and [`gym_ignition.base.Robot`](https://github.com/robotology/gym-ignition/blob/master/gym_ignition/base/robot.py).

Currently the robot interface is unique. Likely this is not scalable, there would be too many methods. We should think how to split it in many interfaces while maintaining consistency between python and C++.

Reminder: the python interface will switch to numpy arrays soon

DanielePucci, 2019-06-24:
@diegoferigo joint torques may be added?

diegoferigo, 2019-06-24:
I thought they were already in the list, of course joint torques.

diegoferigo, 2020-05-05:
Excluding sensors, all the other abstractions have been implemented in the new ScenarI/O bindings #158. Closing.

closed 2020-05-05 by diegoferigo