robotology-legacy/gym-ignition #38: Develop an example to compare Runtime classes

diegoferigo, 2019-09-17:
At the moment the supported runtimes are the following:

- `GazeboRuntime`
- `PyBulletRuntime` (almost ready)

It would be nice comparing simple environments like the existing CartPole, or an even simpler Pendulum. Without contacts, the two physics engines behaviors should mostly match.

diegoferigo, 2019-09-26:
We should also add a comparison with a ground truth. A possibility might be a custom runtime that directly integrates the equation of motion of the pendulum.

For a fair comparison, we should model the pendulum as a cylinder and not as a mass point as commonly done.

diegoferigo, 2019-09-26:
Equations of the pendulum have been derived in #43.

traversaro, 2019-09-26:
If you think that it is feasible, I would also test in the pendulum case the basic coulomb + viscous friction model used in SDF. While not properly documented in the specification (see http://sdformat.org/spec?ver=1.6&elem=joint#axis_dynamics) its precise meaning is quite shared in the community (see https://bitbucket.org/osrf/gazebo/issues/381/ode-joints-dont-support-friction-parameter#comment-12156423), but I suspect that different implementation could have subtly different behaviours.

diegoferigo, 2019-09-26:
I already considered to extend the tests to include friction. Modifying the equations should be easy, it can be a second step though. I want to verify first that simulations are comparable in the simplest case.

diegoferigo, 2019-10-07:
Implemented in https://github.com/robotology/gym-ignition/issues/38 in the simple case without friction.

closed 2019-10-07 by diegoferigo