robotology-legacy/gym-ignition #58: Double free if statically compiled

diegoferigo, 2019-10-09:
For the PyPI distribution, all the C++ libraries of Gym-Ignition are statically compiled. When the user installs the `sdist` package, the installed shared library of the SWIG bindings includes all the other components [1].

Everything seems to work just fine, and tests pass correctly. Though, during resource deallocation, only when the project is compiled as static there's a [`double free or corruption`](https://github.com/robotology/gym-ignition/commit/886676d37973cb4c647f9b8c22f8d07a0252e4de/checks?check_suite_id=257062563#step:7:45) segmentation fault. More details in the [valgrind analysis](https://github.com/robotology/gym-ignition/commit/886676d37973cb4c647f9b8c22f8d07a0252e4de/checks?check_suite_id=257062563#step:8:11319).

I temporarily disabled the check of `pytest` in the PyPI CD pipeline in 886676d37973cb4c647f9b8c22f8d07a0252e4de.

---

[1] The Gazebo plugins are an exception, they are installed separately.

traversaro, 2019-10-09:
Are there any singletons in gym-ignition? Possibly related issue: https://github.com/robotology/codyco-modules/issues/44 .

traversaro, 2019-10-10:
I think it is some kind of tricky interaction between how and when constructors/destructors are called (and by who) in static libraries when used as part of dynamically loaded library, see some related material (not directly addressing the problem):
* https://isocpp.org/wiki/faq/ctors#static-init-order
* https://forums.ogre3d.org/viewtopic.php?t=69311 
* https://groups.google.com/forum/#!topic/android-ndk/xr_h7ediIhY

diegoferigo, 2019-10-10:
Accordingly to Valgrind, the incriminated data structure seems to be the following:

https://github.com/robotology/gym-ignition/blob/8b9cc79862b6c5eabea8ce10bf02bb996be3e22b/ignition/src/TaskSingleton.cpp#L18

This `TaskSingleton` is there since the very beginning of the project development. I recently touched it in https://github.com/robotology/gym-ignition/pull/53 just to change its name. Checking the code, I noticed the following mismatches wrt the other singletons (that work fine):

1. There's no Impl class
1. I use the `ignition::common::Singleton<T>` class

Adding the Impl class had no effects. Tests were passing but the error described above is still there.

Curiously, implementing the singleton pattern internally and **removing the inheritance from the gazebo implementation solves the problem** :thinking: I suspect that there's something wrong in how they deal with the internal static object in the case of static compilation.

A this point is also suspect that the problems we had in #21 were related also to this finding.

PR #66 (2019-10-10): Fix termination segfault when statically compiled
Closes #58
closed 2019-10-10 by diegoferigo