robotology-legacy/gym-ignition #98: Error at compiling time

raffaello-camoriano, 2019-11-06:
Trying to build `gym-ignition` in Dev mode as outlined in the readme file, I stumbled in a build error when running
```
cmake --build .
```

Output: 

```shell
(gym-ignition-env) kammo@kammo-XPS-15-9560:~/repos/gym-ignition/build$ cmake --build .
Scanning dependencies of target tiny-process-library
[  2%] Building CXX object _deps/tinyprocesslibrary-build/CMakeFiles/tiny-process-library.dir/process.cpp.o
[  5%] Building CXX object _deps/tinyprocesslibrary-build/CMakeFiles/tiny-process-library.dir/process_unix.cpp.o
[  8%] Linking CXX shared library ../../lib/libtiny-process-library.so
[  8%] Built target tiny-process-library
Scanning dependencies of target gympp
[ 11%] Building CXX object gympp/CMakeFiles/gympp.dir/src/Space.cpp.o
[ 14%] Building CXX object gympp/CMakeFiles/gympp.dir/src/Random.cpp.o
[ 17%] Linking CXX shared library ../lib/libgympp.so
[ 17%] Built target gympp
Scanning dependencies of target RobotSingleton
[ 20%] Building CXX object ignition/CMakeFiles/RobotSingleton.dir/src/RobotSingleton.cpp.o
[ 22%] Linking CXX shared library ../lib/libRobotSingleton.so
[ 22%] Built target RobotSingleton
Scanning dependencies of target IgnitionRobot
[ 25%] Building CXX object ignition/CMakeFiles/IgnitionRobot.dir/src/IgnitionRobot.cpp.o
[ 28%] Linking CXX shared library ../lib/libIgnitionRobot.so
[ 28%] Built target IgnitionRobot
Scanning dependencies of target TaskSingleton
[ 31%] Building CXX object ignition/CMakeFiles/TaskSingleton.dir/src/TaskSingleton.cpp.o
[ 34%] Linking CXX shared library ../lib/libTaskSingleton.so
[ 34%] Built target TaskSingleton
Scanning dependencies of target ECMSingleton
[ 37%] Building CXX object ignition/CMakeFiles/ECMSingleton.dir/src/ECMSingleton.cpp.o
[ 40%] Linking CXX shared library ../lib/libECMSingleton.so
[ 40%] Built target ECMSingleton
Scanning dependencies of target GazeboWrapper
[ 42%] Building CXX object ignition/CMakeFiles/GazeboWrapper.dir/src/GazeboWrapper.cpp.o
[ 45%] Linking CXX shared library ../lib/libGazeboWrapper.so
[ 45%] Built target GazeboWrapper
Scanning dependencies of target IgnitionEnvironment
[ 48%] Building CXX object ignition/CMakeFiles/IgnitionEnvironment.dir/src/IgnitionEnvironment.cpp.o
[ 51%] Linking CXX shared library ../lib/libIgnitionEnvironment.so
[ 51%] Built target IgnitionEnvironment
Scanning dependencies of target PhysicsSystem
[ 54%] Building CXX object plugins/Physics/CMakeFiles/PhysicsSystem.dir/Physics.cpp.o
/home/kammo/repos/gym-ignition/plugins/Physics/Physics.cpp: In member function ‘void gympp::plugins::Physics::Impl::CreatePhysicsEntities(const ignition::gazebo::v2::EntityComponentManager&)’:
/home/kammo/repos/gym-ignition/plugins/Physics/Physics.cpp:293:10: error: parse error in template argument list
     _ecm.EachNew<components::World, components::Name, components::Gravity>(
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/kammo/repos/gym-ignition/plugins/Physics/Physics.cpp:295:19: error: reference to ‘components’ is ambiguous
             const components::World* /* _world */,
                   ^~~~~~~~~~
/home/kammo/repos/gym-ignition/plugins/Physics/Physics.cpp:110:52: note: candidates are: namespace components = ignition::gazebo::v2::ignition::gazebo::v2::components;
 namespace components = ignition::gazebo::components;
                                                    ^
In file included from /usr/include/ignition/gazebo2/ignition/gazebo/EntityComponentManager.hh:35:0,
                 from /usr/include/ignition/gazebo2/ignition/gazebo/System.hh:23,
                 from /home/kammo/repos/gym-ignition/plugins/Physics/Physics.h:29,
                 from /home/kammo/repos/gym-ignition/plugins/Physics/Physics.cpp:26:
/usr/include/ignition/gazebo2/ignition/gazebo/Entity.hh:44:15: note:                 namespace ignition::gazebo::v2::components { }
     namespace components {}
               ^~~~~~~~~~
/home/kammo/repos/gym-ignition/plugins/Physics/Physics.cpp: In lambda function:
/home/kammo/repos/gym-ignition/plugins/Physics/Physics.cpp:312:11: error: expected ‘{’ before ‘;’ token
         });
           ^
/home/kammo/repos/gym-ignition/plugins/Physics/Physics.cpp: In member function ‘void gympp::plugins::Physics::Impl::CreatePhysicsEntities(const ignition::gazebo::v2::EntityComponentManager&)’:
/home/kammo/repos/gym-ignition/plugins/Physics/Physics.cpp:312:11: error: expected ‘)’ before ‘;’ token
/home/kammo/repos/gym-ignition/plugins/Physics/Physics.cpp:314:10: error: parse error in template argument list
     _ecm.EachNew<components::Model, components::Name, components::Pose, components::ParentEntity>(
[... 897 lines cut ...]
                 from /home/kammo/repos/gym-ignition/plugins/Physics/Physics.cpp:26:
/usr/include/ignition/gazebo2/ignition/gazebo/EntityComponentManager.hh:330:20: note: candidate: template<class ... ComponentTypeTs> void ignition::gazebo::v2::EntityComponentManager::Each(typename ignition::gazebo::v2::EntityComponentManager::identity<std::function<bool(const long unsigned int&, const ComponentTypeTs* ...)> >::type) const
               void Each(typename identity<std::function<
                    ^~~~
/usr/include/ignition/gazebo2/ignition/gazebo/EntityComponentManager.hh:330:20: note:   template argument deduction/substitution failed:
/home/kammo/repos/gym-ignition/plugins/Physics/Physics.cpp:1171:10: error: template argument 1 is invalid
         });
          ^
In file included from /usr/include/ignition/gazebo2/ignition/gazebo/System.hh:23:0,
                 from /home/kammo/repos/gym-ignition/plugins/Physics/Physics.h:29,
                 from /home/kammo/repos/gym-ignition/plugins/Physics/Physics.cpp:26:
/usr/include/ignition/gazebo2/ignition/gazebo/EntityComponentManager.hh:347:20: note: candidate: template<class ... ComponentTypeTs> void ignition::gazebo::v2::EntityComponentManager::Each(typename ignition::gazebo::v2::EntityComponentManager::identity<std::function<bool(const long unsigned int&, ComponentTypeTs* ...)> >::type)
               void Each(typename identity<std::function<
                    ^~~~
/usr/include/ignition/gazebo2/ignition/gazebo/EntityComponentManager.hh:347:20: note:   template argument deduction/substitution failed:
/home/kammo/repos/gym-ignition/plugins/Physics/Physics.cpp:1171:10: error: template argument 1 is invalid
         });
          ^
plugins/Physics/CMakeFiles/PhysicsSystem.dir/build.make:62: recipe for target 'plugins/Physics/CMakeFiles/PhysicsSystem.dir/Physics.cpp.o' failed
make[2]: *** [plugins/Physics/CMakeFiles/PhysicsSystem.dir/Physics.cpp.o] Error 1
CMakeFiles/Makefile2:503: recipe for target 'plugins/Physics/CMakeFiles/PhysicsSystem.dir/all' failed
make[1]: *** [plugins/Physics/CMakeFiles/PhysicsSystem.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

diegoferigo, 2019-11-07:
Hi @raffaello-camoriano, which compiler are you using? I suspect you do not have enabled GCC8.

There are many ways to tell CMake to use a specific compiler. In my opinion on Ubuntu the simplest way is to export the following env vars **before** the first time you configure the project:

```sh
export CC=gcc-8
export CXX=g++-8
```

When you configure the project, you should see in the very beginning something like this:

```
$ cmake ..
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
[...]
```

raffaello-camoriano, 2019-11-07:
> Hi @raffaello-camoriano, which compiler are you using? I suspect you do not have enabled GCC8.
> 
> There are many ways to tell CMake to use a specific compiler. In my opinion on Ubuntu the simplest way is to export the following env vars **before** the first time you configure the project:
> 
> ```shell
> export CC=gcc-8
> export CXX=g++-8
> ```

Thanks, I did some further investigation.

Here are a few observations, with pointers to potential issues and some suggestions.

Part 1: Non-complying CMake Version Gets Installed by `gym-ignition` 
--------
I was quite sure about having enabled it, since I was following the instructions step-by-step.

Since I'm not sure, I repeated the procedure from scratch after removing the build folder.

First problem I encounter now is with the cmake version. 

```
(gym-ignition-env) kammo@kammo-XPS-15-9560:~/repos/gym-ignition/build$ cmake -DCMAKE_INSTALL_PREFIX=/ust/local ..
CMake Error at CMakeLists.txt:5 (cmake_minimum_required):
  CMake 3.12 or higher is required.  You are running version 3.10.2

-- Configuring incomplete, errors occurred!
```
Note that I had installed cmake v 3.15 using `sudo snap install cmake --classic` and soft-linking the snap-installed version via `sudo ln -s /snap/bin/cmake cmake`

This maybe means that one of the previous commands I ran from `gym-ignition`'s instructions overwrote my soft link and installed a non-compatible verions.

Maybe one of the following?
```
sudo apt install ignition-blueprint
pip3 install gym-ignition
```

Part 2: No Instructions to Install `g++-8`
--------
Besides this, I proceeded by manually removing the old cmake from `/user/bin` and replacing it again by a soft link to the `/snap/bin` version, as follows:

```shell
(gym-ignition-env) kammo@kammo-XPS-15-9560:/usr/bin$ sudo rm cmake
(gym-ignition-env) kammo@kammo-XPS-15-9560:/usr/bin$ sudo ln -s /snap/bin/cmake cmake
(gym-ignition-env) kammo@kammo-XPS-15-9560:/usr/bin$ cmake --version
cmake version 3.15.5

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

and then proceeded with the `gym-ignition` dev build instructions to see if the rest of the process would work, as follows:

```shell
(gym-ignition-env) kammo@kammo-XPS-15-9560:~/repos/gym-ignition/build$ export CC=gcc-8
(gym-ignition-env) kammo@kammo-XPS-15-9560:~/repos/gym-ignition/build$ export CXX=g++-8
(gym-ignition-env) kammo@kammo-XPS-15-9560:~/repos/gym-ignition/build$ cmake -DCMAKE_INSTALL_PREFIX=/usr/local ..
CMake Error at /snap/cmake/148/share/cmake-3.15/Modules/CMakeDetermineCXXCompiler.cmake:48 (message):
  Could not find compiler set in environment variable CXX:

  g++-8.

Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
See also "/home/kammo/repos/gym-ignition/build/CMakeFiles/CMakeOutput.log".
```

[... 621 lines cut ...]
libpasteurize
numpy
numpy-1.17.3.dist-info
past
pip
pip-19.3.1.dist-info
pkg_resources
pybullet-2.5.6.dist-info
pybullet.cpython-36m-x86_64-linux-gnu.so
pybullet_data
pybullet_envs
pybullet_utils
__pycache__
pyglet
pyglet-1.3.2.dist-info
scipy
scipy-1.3.1.dist-info
setuptools
setuptools-41.6.0.dist-info
six-1.13.0.dist-info
six.py
wheel
wheel-0.33.6.dist-info
``` 
So there's something wrong also with the python installation.

diegoferigo, 2019-11-08:
> First problem I encounter now is with the cmake version.

> This maybe means that one of the previous commands I ran from gym-ignition's instructions overwrote my soft link and installed a non-compatible verions.

I really doubt that this happened, depending on where you have your install prefix of the CMake project you might not even need administration privileges that can mess around system folders (such as where you symlinked the new CMake version). ~In general, I do recommend to update CMake as documented in the README, through the official PPA.~ I just realized that this information is not there as I thought. I will add it. In the meantime, check the [CI docker image](https://github.com/robotology/gym-ignition/blob/960a856cabcbae53c03f587671dda6e67affa018/.docker/Dockerfile.ci#L16-L17) that reproduces the environment required to test the project in developer mode. Snap is bit tricky if you don't know how it works, I still prefer using PPAs.

> So, one of the issues was probably that I didn't have g++-8 installed.

I am already aware of it, thanks for reporting it. The package `gcc-8` does not install also the C++ compiler. The README will be updated accordingly.

> Then, I ran cmake --build . --target install, which resulted in an error:

In general, install prefix is user choice. Of course, if you choose a folder not owned by your user, you are responsible to grant CMake the right privileges.

> Note: setup.py is in the parent of the build dir, so I suggest to add cd .. to the instructions. I know it's obvious, but for first-time users with little experience it may generate unnecessary future noise.

Right, thanks! I agree that inexperienced used might get confused.

> I was first wondering if the `<installprefix>` was referring to the project's virtual environment or with the cmake install directory prefix, which in my case was `/usr/bin`.

Ignoring that I highly encourage to not use `/usr/bin/` as install prefix (the most similar right choice is `/usr` or, better, `/usr/local`), the name _install prefix_ is quite specific to CMake. Though, I understand that it might confuse inexperienced users.

From the log you attached in step 4, it seems that your install prefix is indeed `/usr/local` (the default when you don't specify anything). If you see the installed files you can clearly see the following:

```
[...]
-- Installing: /usr/local/lib/python3.6/site-packages/gympp_bindings.py
```

Given that in your case `<installprefix>=/usr/local`, the README is correctly stating the following: `export PYTHONPATH=<installprefix>/lib/python3.6/site-packages`.

diegoferigo, 2019-11-08:
@raffaello-camoriano Addressing the instruction in #99, can you please have a look?

traversaro, 2019-11-08:
> This is expected. I linked the file from the repository because you have either to download and execute it from the command line, or, in developer mode, browse to the right folder and execute the file. I think this falls within the minimal knowledge we require users have.

I think adding a sentence like that (explaining the expected way of using `launch_cartpole.py`) would drastically improve docs, and not add a lot of maintenance effort. I may not be a Python expert, but I was also quite confused on how I was expected to run that script.

diegoferigo, 2019-11-08:
Thanks for the feedback, I updated the PR to with a more clear explanation.

raffaello-camoriano, 2020-02-26:
Tried again to follow the guide and while installing ignition-gazebo from binaries on my machine, and this step:
```bash
sudo apt-get install ignition-citadel
```
yields a dependency conflict:
```bash
The following packages have unmet dependencies:
 ignition-citadel : Depends: libignition-launch2-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```
I tried to run `sudo apt install libignition-launch2-dev`, resulting in 
```
The following packages have unmet dependencies:
 libignition-launch2-dev : Depends: ignition-gazebo3 but it is not installable
E: Unable to correct problems, you have held broken packages.
```
And finally, `sudo apt install ignition-gazebo3`:
```
Package ignition-gazebo3 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'ignition-gazebo3' has no installation candidate
```

<details>

```bash
(base) kammo@kammo-XPS-15-9560:~/repos/stable-baselines$ sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
[sudo] password for kammo: 
(base) kammo@kammo-XPS-15-9560:~/repos/stable-baselines$ wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
--2020-02-26 12:19:15--  http://packages.osrfoundation.org/gazebo.key
Resolving packages.osrfoundation.org (packages.osrfoundation.org)... 52.52.171.73
Connecting to packages.osrfoundation.org (packages.osrfoundation.org)|52.52.171.73|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1772 (1,7K) [application/octet-stream]
Saving to: ‘STDOUT’

-                          100%[======================================>]   1,73K  --.-KB/s    in 0,004s  

2020-02-26 12:19:16 (431 KB/s) - written to stdout [1772/1772]

OK
(base) kammo@kammo-XPS-15-9560:~/repos/stable-baselines$ sudo apt update
Hit:1 http://security.ubuntu.com/ubuntu bionic-security InRelease
Hit:2 http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu bionic InRelease
Ign:3 http://dl.google.com/linux/chrome/deb stable InRelease                                             
Ign:4 http://linux.dropbox.com/ubuntu bionic InRelease                                                   
Hit:5 http://it.archive.ubuntu.com/ubuntu bionic InRelease                                               
Hit:6 http://dl.google.com/linux/chrome/deb stable Release                                               
Get:7 http://linux.dropbox.com/ubuntu bionic Release [6.600 B]                                           
Hit:8 http://ppa.launchpad.net/malteworld/ppa/ubuntu bionic InRelease                                    
Get:9 http://it.archive.ubuntu.com/ubuntu bionic-updates InRelease [88,7 kB]                             
Hit:11 http://packages.osrfoundation.org/gazebo/ubuntu-stable bionic InRelease                           
Hit:12 https://download.docker.com/linux/ubuntu bionic InRelease                                         
Hit:13 https://nvidia.github.io/libnvidia-container/ubuntu18.04/amd64  InRelease                         
Hit:15 https://nvidia.github.io/nvidia-container-runtime/ubuntu18.04/amd64  InRelease                    
Hit:16 https://packages.microsoft.com/repos/ms-teams stable InRelease                                    
Hit:17 http://ppa.launchpad.net/yktooo/ppa/ubuntu bionic InRelease                                       
Hit:18 https://nvidia.github.io/nvidia-docker/ubuntu18.04/amd64  InRelease                               
Get:19 http://it.archive.ubuntu.com/ubuntu bionic-backports InRelease [74,6 kB]       
Get:20 http://it.archive.ubuntu.com/ubuntu bionic-updates/main amd64 DEP-11 Metadata [307 kB]
Get:21 http://it.archive.ubuntu.com/ubuntu bionic-updates/main DEP-11 48x48 Icons [73,8 kB]
Get:22 http://it.archive.ubuntu.com/ubuntu bionic-updates/main DEP-11 64x64 Icons [140 kB]
Get:23 http://it.archive.ubuntu.com/ubuntu bionic-updates/universe amd64 DEP-11 Metadata [264 kB]
Get:24 http://it.archive.ubuntu.com/ubuntu bionic-updates/universe DEP-11 48x48 Icons [195 kB]           
Get:25 http://it.archive.ubuntu.com/ubuntu bionic-updates/universe DEP-11 64x64 Icons [434 kB]           
Get:26 http://it.archive.ubuntu.com/ubuntu bionic-updates/multiverse amd64 DEP-11 Metadata [2.464 B]     
Get:27 http://it.archive.ubuntu.com/ubuntu bionic-backports/universe amd64 DEP-11 Metadata [7.980 B]     
Fetched 1.593 kB in 11s (142 kB/s)                                                                       
[... 16 lines cut ...]
E: Unable to correct problems, you have held broken packages.
(base) kammo@kammo-XPS-15-9560:~/repos/stable-baselines$ sudo apt install libignition-launch2-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libignition-launch2-dev : Depends: ignition-gazebo3 but it is not installable
E: Unable to correct problems, you have held broken packages.
(base) kammo@kammo-XPS-15-9560:~/repos/stable-baselines$ sudo apt install ignition-gazebo3
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package ignition-gazebo3 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'ignition-gazebo3' has no installation candidate
```
</details>

diegoferigo, 2020-02-26:
I encountered similar problems recently, and it was caused by out-of-sync packaging of the gazebo ppa. Can you temporarily try to use the `ubuntu-prerelease` channel? Following the official installation instructions, you have to change only the following line:

```
sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-prerelease `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-prerelease.list'
```

Remember to remove (or disable) the `gazebo-stable` channel first, and execute `apt-get update` before trying to reinstall the `ignition-citadel` meta package.

diegoferigo, 2020-05-05:
@raffaello-camoriano I'm closing this issue. Let me know if you still find installation problems.

closed 2020-05-05 by diegoferigo