robotology-legacy/gym-ignition #100: MacOs Mojave 10.14.6 Compilation

CarlottaSartore, 2019-11-08:
Trying to compile gym-ignition on `macOS Mojave 10.14.6`

Installed `gazebo-ignition` via` homebrew` by following https://ignitionrobotics.org/docs/latest/install.

 ```brew install ignition-blueprint```

Needed to unlink sdformta6 

```brew unlink sdformat6```

for using sdformat8
 
 Installed `gcc-8 `

```brew install gcc@8```

and exported 

```
export CC=gcc-8
export CXX=g++-8
```

Cloned the repository, inside the repository, configure and make 

```
mkdir build && cd build
cmake .. -DCMAKE_INSTALL_PREFIX=/usr/local
cmake --build .
```

**CMakeWarning**
```
  Could not find a configuration file for package "DART" that is compatible
  with requested version "6.10.0".

  The following configuration files were considered but not accepted:

    /usr/local/share/dart/cmake/DARTConfig.cmake, version: 6.9.2
```

**Compilation Error**
```
[ 17%] Linking CXX shared library ../lib/libgympp.dylib
Undefined symbols for architecture x86_64:
  "ignition::common::Logger::operator()(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)", referenced from:
      gympp::spaces::Discrete::contains(gympp::data::Sample const&) const in Space.cpp.o
      gympp::spaces::Box::contains(gympp::data::Sample const&) const in Space.cpp.o
ld: symbol(s) not found for architecture x86_64
collect2: error: ld returned 1 exit status
```

By putting to false (not solving the problem since ignition-common3 is present in the system)
```
if(ignition-common3_FOUND))
    target_compile_definitions(gympp PUBLIC USE_IGNITION_LOGS)
    target_link_libraries(gympp PUBLIC
        ignition-common3::ignition-common3)
endif()
```

The compilation continued 
```
[ 20%] Building CXX object ignition/CMakeFiles/RobotSingleton.dir/src/RobotSingleton.cpp.o
/Users/csartore/gym-ignition/ignition/src/RobotSingleton.cpp: In member function 'std::weak_ptr<gympp::Robot> gympp::gazebo::RobotSingleton::getRobot(const string&) const':
/Users/csartore/gym-ignition/ignition/src/RobotSingleton.cpp:56:5: error: 'assert' was not declared in this scope
     assert(pImpl->robots.at(robotName));
     ^~~~~~
/Users/csartore/gym-ignition/ignition/src/RobotSingleton.cpp:56:5: note: 'assert' is defined in header '<cassert>'; did you forget to '#include <cassert>'?
/Users/csartore/gym-ignition/ignition/src/RobotSingleton.cpp:11:1:
+#include <cassert>

/Users/csartore/gym-ignition/ignition/src/RobotSingleton.cpp:56:5:
     assert(pImpl->robots.at(robotName));
     ^~~~~~
```

By adding the include asked 

```
Undefined symbols for architecture x86_64:
  "google::protobuf::MessageLite::ParseFromIstream(std::basic_istream<char, std::char_traits<char> >*)", referenced from:
      ignition::gazebo::v2::serializers::VectorDoubleSerializer::Deserialize(std::basic_istream<char, std::char_traits<char> >&, std::vector<double, std::allocator<double> >&) in IgnitionRobot.cpp.o
  "ignition::math::v6::PID::Update(double, std::chrono::duration<double, std::ratio<1l, 1l> > const&)", referenced from:
      gympp::gazebo::IgnitionRobot::update(std::chrono::duration<double, std::ratio<1l, 1l> >) in IgnitionRobot.cpp.o
  "ignition::common::env(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)", referenced from:
```

I have cheked and `google::protobuf::MessageLite::ParseFromIstream` exists in the related header located in `/usr/local/Cellar/protobuf/3.10.0/include/google/protobuf/message_lite.h`

traversaro, 2019-11-08:
By quickly inspecting the ignition-gazebo homebrew formulas, it seems that they are  compiled using the normal system compiler, so probably I would avoid setting manually gcc8 when compiling gym-ignition .

diegoferigo, 2019-11-08:
@CarlottaSartore What is in Mojave the default compiler? The compiler our CI infrastructure tests are the following:

- gcc 8
- clang 7
- clang 8

Maybe as @traversaro wrote the default version of the compiler on your OSX distribution is already good enough.

CarlottaSartore, 2019-11-08:
The default compiler is `AppleClang 11.0.0.11000033 ` and with that, I succeeded in compiling gym ignition.
I still have to run the example and check that everything is working fine but at least the compilation worked quite easy without setting manually gcc8.

diegoferigo, 2019-11-08:
Cool, let us know if the examples work fine. With some work me maybe could even think to officially support osx. Though, before announcing it we should at least also 1) distribute a binary wheel for osx, and that might take some effort, and 2) have a osx maintainer (@CarlottaSartore :eyes: ).

traversaro, 2019-11-08:
> Though,  before announcing it we should at least also 1) distribute a binary  wheel for osx, and that might take some effort, and 2) have a osx  maintainer (@CarlottaSartore 👀 ).

Before deciding to distribute a binary wheel for macOS, I would first do to a due diligence on the public  dependencies of ignition-gazebo, to see if they include boost or other problematic dependencies, as homebrew is unfortunately a [rolling release](https://en.wikipedia.org/wiki/Rolling_release) system, so the ABI of the dependencies can change without any advice, breaking any binary that depends on such ABI. The ABI changes should not be a problem for ignition libraries as each major version is considered as a different homebrew package, but for transitive dependencies this could be a problem. See https://github.com/ros-infrastructure/bloom/issues/254#issuecomment-379359911 for a related comment.

IeiuniumLux, 2019-11-10:
@CarlottaSartore, out of curiosity, were you at least able to launch **Ignition-Gazebo** on the MacOs Mojave from command line? _e.g:_

`$ ign gazebo -r rolling_shapes.sdf`

According to this [PR](https://bitbucket.org/ignitionrobotics/ign-gazebo/issues/25/apple-support-for-ign-command-line-tool), the `ign gazebo` command line tool is not working on Mac.

@diegoferigo @traversaro, does the `cartpole` example make usage of Ignition GUI Simulator?

diegoferigo, 2019-11-10:
> @diegoferigo @traversaro, does the cartpole example make usage of Ignition GUI Simulator?

Gym-Ignition executes simulations typically in headless mode. The framework, in line with the OpenAI Gym interface, provides an optional `render()` method that spawns a new process and allows visualizing the simulated environment in a graphical frontend. In the cartpole demo the GUI is enabled by default, though commenting out the render method the execution would happen entirely in the console.

This issue describes the very first attempt on MacOs done by @CarlottaSartore, and up to today there are no further updates beyond what reported here.

What's unfortunate is that we rely to the [`ign gazebo -g`](https://github.com/robotology/gym-ignition/blob/960a856cabcbae53c03f587671dda6e67affa018/ignition/src/GazeboWrapper.cpp#L334-L335) command line to spawn the GUI process. Therefore, if the command line is not working properly on MacOs and Windows, the environments will not have rendering capabilities on those OSs.

IeiuniumLux, 2019-11-10:
@diegoferigo thanks for the clarification.  Let's hope that the coming release of Ignition-Gazebo ([Citadel](https://ignitionrobotics.org/docs/all/releases)) in December addresses the support for MacOS and Windows since it works great on Ubuntu 18 (as shown below).  Thank you for contributing this great component and making this RL capability available to Ignition.

[image]

traversaro, 2019-11-11:
> What's unfortunate is that we rely to the [`ign gazebo -g`](https://github.com/robotology/gym-ignition/blob/960a856cabcbae53c03f587671dda6e67affa018/ignition/src/GazeboWrapper.cpp#L334-L335) command [...]

A  possible workaround (at least for Blueprint) is to use `ign-gazebo -g` on macOS, even if deprecated, as it is the only one currently working.

diegoferigo, 2019-11-12:
Just to put another option on the table, there's also `ign gui`, even though I suspect that it is an alias of `ign gazebo -g` (or better, the other way around).

IeiuniumLux, 2019-11-12:
Neither one works for me on the MacOS thought.

```
~$ ign gazebo -g
Library error for [/usr/local/Cellar/ignition-gazebo2/2.11.0/lib/libignition-gazebo2-ign.2.11.0.dylib]: can't load /usr/local/Cellar/ignition-gazebo2/2.11.0/lib/libignition-gazebo2-ign.2.11.0.dylib

~$ ign gui
/usr/local/Cellar/ignition-gui2/2.0.1_2/lib/ruby/ignition/cmdgui2.rb:189: [BUG] Segmentation fault at 0x0000000000000000
ruby 2.3.7p456 (2018-03-28 revision 63024) [universal.x86_64-darwin18]
```

diegoferigo, 2019-11-13:
@IeiuniumLux I would just try to ping in the upstream's issue. As a temporary workaround, as commented there, installing ruby from homebrew and changing the shebang should work.

traversaro, 2019-11-14:
By the way, I actually meant to use `ign-gazebo -g` (with the dash between `ign` and `gazebo`) that as far as I understand should be available also on macOS in ignition-gazebo2 .

IeiuniumLux, 2019-11-14:
@traversaro `ign-gazebo-gui` tool is deprecated as shown below.

```
$ ign-gazebo -g
[Err] [gui_main.cc:129] The ign-gazebo-gui tool is deprecated, and is replaced by `ign gazebo`
```

@diegoferigo I'll give that a try. Thx!

IeiuniumLux, 2019-11-15:
@diegoferigo regarding changing `changing the shebang`; which one?

diegoferigo, 2019-11-16:
> As a test, I installed another ruby implementation with brew install ruby and modified the shebang line of /usr/local/bin/ign to point to that other ruby in /usr/local, and it was successful!

I was referring to [this comment](https://bitbucket.org/ignitionrobotics/ign-gazebo/issues/25/apple-support-for-ign-command-line-tool#comment-53976669).

IeiuniumLux, 2019-11-16:
I edited the `ign` file under `/usr/local/bin` and changed the top line `#!/usr/local/opt/ruby/bin/ruby` as suggested in that comment, but I get the error below when I try to launch it on the MacOS.

```
$ export IGN_CONFIG_PATH=/usr/local/share/ignition
$ ign gazebo shapes.sdf -v 4
```

```
Error while loading the library [/usr/local/lib//libignition-physics1-dartsim-plugin.1.dylib]: dlopen(/usr/local/lib//libignition-physics1-dartsim-plugin.1.dylib, 5): Library not loaded: /usr/local/opt/assimp/lib/libassimp.4.dylib
  Referenced from: /usr/local/lib//libignition-physics1-dartsim-plugin.1.dylib
  Reason: image not found
[Err] [Physics.cc:271] Unable to load the /usr/local/lib//libignition-physics1-dartsim-plugin.1.dylib library.
```

traversaro, 2019-11-16:
I am afraid that @diegoferigo did not actually tried the workaround mention in that comment, so probably it would make more sense to ask to @scpeters directly.

scpeters, 2019-11-17:
dartsim@6.10.0 is a forked version with some extra friction parameters that we have been using and hosting on our homebrew tap. It is keg-only so as to not interfere with the official package, but you need to set some extra environment variables to use it:

* https://bitbucket.org/osrf/release-tools/src/c6d88a6f19e5779f56ad6ec2b54fa7e3ba473d62/jenkins-scripts/lib/project-default-devel-homebrew-amd64.bash#lines-123:128

IeiuniumLux, 2019-11-17:
As soon as I set this:

```
export DYLD_FALLBACK_LIBRARY_PATH=${DYLD_FALLBACK_LIBRARY_PATH}:/usr/local/opt/dartsim@6.10.0/lib:/usr/local/opt/octomap/local
```

Then I get this:

`Library error for [/usr/local/Cellar/ignition-gazebo2/2.11.0/lib/libignition-gazebo2-ign.2.11.0.dylib]: can't load /usr/local/Cellar/ignition-gazebo2/2.11.0/lib/libignition-gazebo2-ign.2.11.0.dylib`

scpeters, 2019-11-17:
that looks like the following bug, and I just replied to your  comment there 

* https://bitbucket.org/ignitionrobotics/ign-gazebo/issues/25/apple-support-for-ign-command-line-tool

diegoferigo, 2019-11-17:
@CarlottaSartore [Steven's comment](https://github.com/robotology/gym-ignition/issues/100#issuecomment-554714843) about dart 6.10 applies also to your setup.

diegoferigo, 2019-11-17:
Thank you @scpeters for chiming in! Let's move the discussion about the command line to the [upstream issue](https://bitbucket.org/ignitionrobotics/ign-gazebo/issues/25/apple-support-for-ign-command-line-tool).