robotology-legacy/gym-ignition #120: Error testing computed_torque_fixed_base.py

paolo-viceconte, 2020-01-17:
Hi,

I tried to set up gym-ignition locally on my Ubuntu 18.04 according to the guidelines in the README and I managed to run the cartpole example.

After that, I tried to integrate the code for the controllers in #102 and locally run the tests included in the PR via the following steps:
- as suggested in the new README, I installed `libeigen3-dev` and [iDynTree](https://github.com/robotology/iDynTree) with the CMake option `IDYNTREE_USES_PYTHON` enabled (I also added to the PYTHONPATH enviromental variable the install path of the iDynTree.py file as required)
- in order to align the local code with the unmerged PR, I executed `git fetch origin pull/102/head:102` and then `git checkout 102` before repeating the 'Build and install the CMake project' section of the gym-ignition setup guidelines (being not an expert git user, I don't know whether there is a more appropriate way)

If I try to run the tests included in the PR using `pytest`, all the tests succeed except for `computed_torque_fixed_base.py` which returns the error
```
tests/python/test_controller_fixed_base.py:None (tests/python/test_controller_fixed_base.py)
ImportError while importing test module '/home/iitcublap185/gym-ignition/tests/python/test_controller_fixed_base.py'.
Hint: make sure your test modules/packages have valid Python names.
E   ImportError: dynamic module does not define module export function (PyInit__iDynTree)
```   
@diegoferigo is the test supposed to succeed right? The issue could be related to the `pytest` version (I currently have the 5.3.2 installed), but I still don't get why it appears only for this specific test. I didn't do any modification to the code for the time being. Any suggestion is appreciated, thanks.

traversaro, 2020-01-17:
What happens if you try to run: 
~~~
import iDynTree
~~~
in the Python prompt?

paolo-viceconte, 2020-01-17:
Very same error. By looking at the complete traceback there seems to be a switch from python 2.7 to 3.6 (required by gym-ignition): there could be a compatibility issue?
```Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/iDynTree.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "/home/iitcublap185/venv/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 658, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 571, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 922, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: dynamic module does not define module export function (PyInit__iDynTree)
```

traversaro, 2020-01-17:
I think the problem is that you compiled iDynTree bindings with Python 2 support, while you should compiled them with Python 3. Can you try to re-compiled iDynTree with the `IDYNTREE_USES_PYTHON_VERSION` CMake option set to 3? 
See https://github.com/robotology/idyntree/blob/d5bbceab9e8ae52943b31199f6e024d12ff52159/bindings/python/CMakeLists.txt#L4 .

paolo-viceconte, 2020-01-24:
Compiling iDynTree with the `IDYNTREE_USES_PYTHON_VERSION` CMake option set to 3 and properly updating the PYTHONPATH enviromental variable with 
```
export PYTHONPATH=$PYTHONPATH:<prefix>/lib/python3.6/site-packages
```
in place of
```
export PYTHONPATH=$PYTHONPATH:<prefix>/lib/python2.7/dist-packages/
```

solves the import problem. Closing.

closed 2020-01-24 by paolo-viceconte