robotology-legacy/gym-ignition #127: Simplify the logic to apply of link wrenches

diegoferigo, 2020-02-01:
As soon as [ignitionrobotics/ign-gazebo#509](https://bitbucket.org/ignitionrobotics/ign-gazebo/pull-requests/509/added-link-addworldwrench/diff) will be merged, we can simplify a lot the logic to apply link wrenches introduced in #124.

traversaro, 2020-02-01:
As usual, pay attention to the details on how this quantities are expressed, see https://bitbucket.org/ignitionrobotics/ign-gazebo/pull-requests/509/added-link-addworldwrench/diff#comment-134092323 .

raffaello-camoriano, 2020-02-02:
> As usual, pay attention to the details on how this quantities are expressed, see https://bitbucket.org/ignitionrobotics/ign-gazebo/pull-requests/509/added-link-addworldwrench/diff#comment-134092323 .
> cc @paolo-viceconte

So at the current stage `AddWorldForce` is not compliant with screw theory standards. What about other uses of wrenches in `gazebo-ign`, are they compliant?
Please let us know if they clarify whether the application point is the link CoM. Thank you!

diegoferigo, 2020-02-12:
> As usual, pay attention to the details on how this quantities are expressed

Yes thanks for pointing it out. Using the low-level component, the wrench is applied to the frame of the link. Instead, using these helpers, forces are applied to the center of mass.

In general, the CoM is more intuitive since it is not dependent on how the model has been designed. As soon as the upstream PR get merged and we port our code to use the Link helper, we should also change the Python documentation from link frame to CoM (before freezing the APIs :slightly_smiling_face:).

diegoferigo, 2020-06-15:
The new ScenarI/O APIs of #158 refactored also the logic for applying link wrenches:

https://github.com/robotology/gym-ignition/blob/28d44844ef9e1273ac3433b31159fcf6ba33a277/cpp/scenario/gazebo/src/Link.cpp#L450-L464

Now all the methods call `Link::applyWorldWrench` that applies a wrench to the CoM of the link.

closed 2020-06-15 by diegoferigo