robotology-legacy/gym-ignition #128: Add Inverse Kinematics class

diegoferigo, 2020-02-03:
Comparing with other frameworks, gym-ignition still lacks an IK implementation.

Since we will already depend on iDynTree for the upcoming controllers (#102), we might take advantage of the powerful IK provided there and wrap it in a Python class to simplify its usage and expose / accept numpy objects.

cc @paolo-viceconte

DanielePucci, 2020-02-03:
It may also be the moment to port the IB-IK (Integration-Based IK) developed by @lrapetti to iDynTree @diegoferigo @paolo-viceconte

diegoferigo, 2020-02-03:
It would be really great, but unfortunately that's not a small task. The whole IK class in iDynTree would need a big polishing, and it could be the right occasion to create a common interface with multiple implementation. In my opinion the IB-IK is the perfect solution in the case we reach a bottleneck with the one based on IPOPT. I would start with that before diving into something that, given the current priorities, could be a demanding task.

traversaro, 2020-02-04:
> It would be really great, but unfortunately that's not a small task. The whole IK class in iDynTree would need a big polishing, and it could be the right occasion to create a common interface with [...]

General comment not directly related to whether we integrated IB-IK in iDynTree now or later: I would not conflate the task of bringing IB-IK to iDynTree  (eventually exposing it via bindings) and the task of wrap the IPOPT-based IK and the IB-IK IK under a a common IK interface (either the existing one or the new one). Bringing IB-IK in iDynTree is easy and  quick, while providing a common interface is hard, so conflating them does not help in the task  planning.

diegoferigo, 2020-02-05:
> General comment not directly related to whether we integrated IB-IK in iDynTree now or later: I would not conflate the task of bringing IB-IK to iDynTree (eventually exposing it via bindings) and [...]

I agree with your line @traversaro, though if we allocate a person on this task, the familiarity with the IK code that they gain would create the perfect setting for the refactoring. I don't think that the common interface would be extremely hard (we can just keep the current one to avoid breaking current code), the challenging part is refactoring the NLP-IK because if I remember it really needs a polishing.

diegoferigo, 2020-02-11:
It's still OT with this issue in gym-igniton, but today I had a quick discussione with @S-Dafarra for related topics and we realized that having a common interface for the NLP-IK and IB-IK is quite challenging. The way the two implementation are configured are very different, in fact the NLP-IK has way more methods since it allows to specify all the constraints and cost. In general, the applications where the two IK can be used are quite different, e.g. NLP one targets a setting that can be very constrained, where the IB instead targets a very simple setup and no constraints / cost are supported.

The only common method are `solve` and maybe all the getters. I'm not sure how such interface could be useful. In light on this finding, maybe a simple port of the current implementation of the IB-IK inside idyntree is enough for the time being.

S-Dafarra, 2020-02-11:
> where the IB instead targets a very simple setup and no constraints / cost are supported.

I would say that constraints can be added also to the IB-IK but only at the velocity level. I would also add that applications using the NLP-IK require to have a solution "one-shot", while those using IB-IK can accept a solution that will **converge** to the desired one after several cycles.

lrapetti, 2020-02-11:
> I would say that constraints can be added also to the IB-IK but only at the velocity level.

in our application we add also constraints at the kinematic level using position dependent velocity constraints (hyperbolic tangent) + saturation in the integrator.

diegoferigo, 2020-02-11:
Thanks for the details! :)

diegoferigo, 2020-02-11:
Closed via #137

closed 2020-02-11 by diegoferigo