robotology-legacy/gym-ignition #168: World files cannot load models via URI (to fuel.ignitionrobotics.org)

wolfgangschwab, 2020-04-05:
Using a world file that contains URIs to fuel.ignitionrobotics.org generates errors. 

I added some code to my local copy of the repository to solve this issue. So, models are automatically loaded from the website. If you are interested in the solution, I can generate a Pull request with this.

traversaro, 2020-04-05:
Hi @wolfgangschwab, 
  thanks a lot for opening the issue. @diegoferigo can confirm this, but I guess that fuel URIs should be supported, so I think we are indeed interested in fixing this issue, so feel free to provide the details of your solution. Furthermore, I guess it would be useful to know which version of gym-ignition you are using, and if you have a self-contained example that reproduces the issue it would be great too.

diegoferigo, 2020-04-05:
With the released versions of gym-ignition I never tried to load world with models included as Fuel URIs, I would not be surprised if something would not work. However, thinking about how the world model is processed, it should be `ignition::gazebo::Server` responsibility to download resources from fuel. If you already found a missing piece from our side, I'd be happy to check it.

**Spoiler alert:** If you didn't notice, we're in the process of a big C++ refactoring #155, and in the `refactoring` branch I added the possibility to download Fuel models as function calls directly from C++ and Python:

https://github.com/robotology/gym-ignition/blob/a303413174aaa3ff6d111f014f53ac856d7674ce/cpp/scenario/gazebo/include/scenario/gazebo/utils.h#L48-L49

Fuel support is one of many new features that are coming soon. The `refactoring` branch is not yet usable, but I think it could hit the nightly channel in few weeks.

wolfgangschwab, 2020-04-05:
The major point was, that callback-functions were missing in GazeboWrapper::findAndLoadSdf . So I added 

```
    sdf::setFindCallback(myFetchResource);
    ignition::common::addFindFileURICallback(myFetchResourceUri);
```
Additionally I had to write the callback-functions and add the ignition fuel-tools package. For world files this works. With robots I still have problems. But currently I'm looking more how laser-sensors (Lidar) can be used. As I understand currently Laser sensors are not implemented. Do you have plans to do so in the near future?

diegoferigo, 2020-04-06:
@wolfgangschwab thanks a lot for the suggestion, I'll have a look!

> But currently I'm looking more how laser-sensors (Lidar) can be used. As I understand currently Laser sensors are not implemented. Do you have plans to do so in the near future?

I general, sensors are not yet supported by the framework. In the next months we have other features we need with more urgency, I doubt anyone would start looking into sensors integration in the next few months. If you're interested in contributing, I'd be happy to discuss in a dedicated issue how to implement them.

wolfgangschwab, 2020-04-06:
Thanks, I will open a new issue for the Lidar-Sensors. 

Regarding the fix for using URI with fuel: Just tell me, if you would like to get additional info or a Pull request.

diegoferigo, 2020-04-06:
> Regarding the fix for using URI with fuel: Just tell me, if you would like to get additional info or a Pull request.

Go ahead with a PR, you can target directly `master` for this fix. In the meantime, I check if I have to do the same also in the new architecture of the `refactoring` branch. Thanks a lot!

> Thanks, I will open a new issue for the Lidar-Sensors.

Perfect, thanks!

wolfgangschwab, 2020-04-09:
I raised Pull Request #174 to solve this issue.

diegoferigo, 2021-03-31:
Closed via #309

PR #309 (2021-03-31): Resolve Fuel resources when included in models and worlds
Closes: #296
Closes: #174
Closes: #168

After digging into it, all that was needed is to specify a callback. I chose to implement it slightly differently than #174

Instead of adding fuel support to the GazeboWrapper, I added it directly to the (lower level?) `GazeboSimulator`. Loading fuel models requires a `FuelClient` which became an attribute of `GazeboSimulator` to avoid constantly re-instantiating it (I mimic this behavior from ign-gazebo).

I didn't find any C++ unit tests to add a test for downloading fuel models, so I didn't ad [... cut]
closed 2021-03-31 by diegoferigo