robotology-legacy/gym-ignition #203: swig package version 4.0 needed but not found

wolfgangschwab, 2020-05-18:
I tried to install the new Scenario Version by using the `devel` branch. The README describes that the swig package should be installed:

`Install SWIG with apt install swig`

This installs version 3.0.12 but later the Version 4.0 is required. I tried to install directly the swig package but I need a hint to get this working. Which environment variable is need to be set to which path?

An update of the README would be helpful too. ;-)

wolfgangschwab, 2020-05-18:
Dropping the build folder solved my issue. 

Amending the README, that Version 4.0 is required and 'apt install swig' will not do the job, would be helpful.

diegoferigo, 2020-05-19:
@wolfgangschwab I have almost finalized the new instructions (tracked in #202), I will open a new PR soon. Sorry but these side-tasks have a low priority at the moment.

The devel branch needs swig 4. In Ubuntu bionic you have to install it from sources (see [Dockerfile](https://github.com/robotology/gym-ignition/blob/370031022b7553b74150c39b29c33e56a2fb3a4f/.docker/Dockerfile.cicd-devel#L58-L74) for a quick overview), in Focal instead it's already in the official repository. Note that as soon as we'll have Ubuntu 20.04.1 and Ignition Gazebo will be packaged for the new LTS, we're going to drop the support to Bionic.

diegoferigo, 2020-05-19:
@wolfgangschwab I opened #204 with the current status of the new instructions (they should be almost complete) and already pushed them to the website https://robotology.github.io/gym-ignition.

I hope that they will be helpful, let me know if something is still missing or does not work.

closed 2020-05-19 by diegoferigo