robotology-legacy/gym-ignition #206: Malformed RPATH in developer installation

diegoferigo, 2020-05-30:
I'm not yet sure why this does not happen in my local setup, but on a clean Ubuntu installation (docker container) the Developer Installation has some problems. In particular, I see that few plugins installed in `<prefix>/lib/scenario/plugins` do not find libraries installed in `<prefix>/lib/`.

Example:

```bash
$ ldd /usr/local/lib/scenario/plugins/libControllerRunner.so | grep found
        libComputedTorqueFixedBase.so => not found
        libScenarioGazebo.so => not found
```

In this installation mode, the RPATH is configured as follows:

https://github.com/robotology/gym-ignition/blob/22d6980a5509892ca6d141f81313efbd0221720a/CMakeLists.txt#L109-L115

---

### Solution

The solution is quite easy: adding the plugin folder to `LIB_DIRs` as follows:

```cmake
    add_install_rpath_support(
        BIN_DIRS "${CMAKE_INSTALL_FULL_BINDIR}"
        LIB_DIRS "${CMAKE_INSTALL_FULL_LIBDIR}"
                 "${CMAKE_INSTALL_FULL_LIBDIR}/scenario/plugins"
                 "${Python3_SITELIB}"
        INSTALL_NAME_DIR "${CMAKE_INSTALL_FULL_LIBDIR}"
        DEPENDS ENABLE_RPATH
        USE_LINK_PATH)
```

Though, I don't think this would be strictly necessary since no one should link against the plugin shared libraries and their location is not important. Maybe the RPATH module enables itself only for libraries contained in the listed directories? The [documentation](https://github.com/robotology/ycm/blob/master/modules/AddInstallRPATHSupport.cmake) says:

```cmake
#  - ``LIB_DIRS`` list of directories to be added to the RPATH. These
#    directories will be added "relative" w.r.t. the ``BIN_DIRS`` and
#    ``LIB_DIRS``.
```

@traversaro any clue? If you don't have a prompt answer no problem, I will open a PR with the fix anyways. I wanted to report this problem here for future reference.

traversaro, 2020-05-30:
It is not necessary to add the directories that contains plugin to the RPATH in general, but in this case it seems that `ComputedTorqueFixedBase` and `ScenarioGazebo` are not used as plugins, but rather as shared library that are dynamically linked: 
https://github.com/robotology/gym-ignition/blob/370031022b7553b74150c39b29c33e56a2fb3a4f/cpp/scenario/plugins/ControllerRunner/CMakeLists.txt#L33 

To avoid the need to have this additional RPATH settings you can either avoid to have libraries that are both modules and dynamically linked libraries, or if you have them install them in the directory where the shared libraries are normally installed.

diegoferigo, 2020-05-30:
In fact they are normal shared libraries that are installed in the regular `${CMAKE_INSTALL_FULL_LIBDIR}`, sorry if this was not clear in my original comment. The only plugin here is `libControllerRunner.so` that is correctly in a subfolder.

More in detail:

```bash
$ tree /usr/local/lib/ 
/usr/local/lib/
├── libComputedTorqueFixedBase.so
├── libECMSingleton.so
├── libGazeboSimulator.so
└── scenario
    └── plugins
        ├── libControllerRunner.so
        ├── libECMProvider.so
        ├── libJointController.so
        └── libPhysicsSystem.so
```

traversaro, 2020-05-30:
Ah ok, then yes, I would not expect this behavior.

diegoferigo, 2020-05-31:
That seems strange also to me :man_shrugging: Couldn't find enough time to debug it, the quick fix works fine. Thanks for the feedback anyway :)

PR #207 (2020-05-31): Fix RPATH in developer installation
Fixes #206
closed 2020-05-31 by diegoferigo