robotology-legacy/gym-ignition #209: CMake Error after updating to newest Ignition-citadel version

wolfgangschwab, 2020-06-03:
As I still have problems to get the devel branch working, I'm still stuck with the master branch version. There I get an error after updating to the newest version of ignition-citadel. When running the cmake command:
`cmake -DCMAKE_INSTALL_PREFIX=/home/developer/other/gym-ignition_ws/gym-ign-install/ ..`

I get an error:
`CMake Error at plugins/Physics/CMakeLists.txt:25 (target_compile_definitions):
  Error evaluating generator expression:

    $<TARGET_SONAME_FILE:ignition-physics2::ignition-physics2-dartsim-plugin>

  No target "ignition-physics2::ignition-physics2-dartsim-plugin"`
Do you have a hint how to solve this?

diegoferigo, 2020-06-04:
Hi @wolfgangschwab, https://github.com/robotology/gym-ignition/pull/208 fixed this problem, can you try to update to the latest version?

wolfgangschwab, 2020-06-04:
Hi @diegoferigo , thanks for the feedback. I updated my devel-version, but still it does not work. When I start the cartpole case, one cpu is running at 100% but nothing happens. Is the fix also in the master version available?

diegoferigo, 2020-06-05:
I didn't get whether you're using the `master` or the `devel` branch. #208 targets devel, master is currently quite behind and I'm not sure when I find time to port the fix, it's not straightforward this time.

wolfgangschwab, 2020-06-05:
@diegoferigo, can you give me a hint how to solve my problem with the devel-branch ?

diegoferigo, 2020-06-05:
What do you mean with nothing happens? The GUI does not open? I realized that the rendering of the example has been disabled, can you try to uncomment the render line of the example? Note that there's currently a problem of ogre2 on ubuntu bionic #211.

wolfgangschwab, 2020-06-05:
What I mean is no GUI is opening but also no output is printed in the terminal.

diegoferigo, 2020-06-05:
For the GUI you have to enable rendering, what's weird is if you don't have any output even with rendering disabled. I expect that the following line prints the episode reward:

https://github.com/robotology/gym-ignition/blob/28d44844ef9e1273ac3433b31159fcf6ba33a277/examples/python/launch_cartpole.py#L73

What does it happen if you run the code in an interactive session, e.g. using `ipython`? Then, can you report how you installed gym-ignition and its dependencies?

wolfgangschwab, 2020-06-06:
To ensure that there is no problem with the build folder. I dropped the build folder and the gym-ign-install folder and rebuild everything. Now I get the following error:

> developer@wolfgang-desktop-2:~/other/gym-ignitionV2_ws$ python3 ~/other/gym-ignitionV2_ws/gym-ignition/examples/python/launch_cartpole.py
Traceback (most recent call last):
  File "/home/developer/other/gym-ignitionV2_ws/gym-ignition/examples/python/launch_cartpole.py", line 8, in <module>
    from gym_ignition.utils import logger
  File "/home/developer/other/gym-ignitionV2_ws/gym-ignition/python/gym_ignition/__init__.py", line 22, in <module>
    import scenario_bindings
  File "/usr/lib/python3/dist-packages/scenario_bindings.py", line 11, in <module>
    import _scenario_bindings
ImportError: /usr/lib/python3/dist-packages/../../../../home/developer/other/gym-ignitionV2_ws/gym-ign-install/lib/libGazeboSimulator.so: undefined symbol: _ZNK8scenario7plugins6gazebo12ECMSingleton8hasWorldERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE

Any hint for this problem?

diegoferigo, 2020-06-08:
> What does it happen if you run the code in an interactive session, e.g. using ipython? Then, can you report how you installed gym-ignition and its dependencies?

Sorry but if you don't tell me how you installed gym-ignition it's quite difficult to me to guess the origin of the problem. We have different installation procedures and the resulting structure in the filesystem is quite different.

wolfgangschwab, 2020-06-08:
Sorry, I thought this problem is independent from the installation. 

I 'm using a docker image for both versions of gym-ignition (master branch and devel branch). The commands that extended an existing image for the use with gym-ignition are:
[gym-ignition-docker.txt](https://github.com/robotology/gym-ignition/files/4745664/gym-ignition-docker.txt)

In this docker image I use the following commands:
[gym-ignitionV2-installation.txt](https://github.com/robotology/gym-ignition/files/4745618/gym-ignitionV2-installation.txt)

diegoferigo, 2020-06-08:
I had a quick look to your setup and I think there are few parts that are not clear to me (possibly unrelated to the problem of posted):

1. You create a virtualenv but the line `RUN /bin/bash -c "source $HOME/venv/bin/activate"` just activates it temporarily. You either need to source it from your bashrc or extend the PATH variable prepending `/venv/bin`.
1. The devel branch of gym-ignition, as explained in the [documentation](https://robotology.github.io/gym-ignition/devel/installation/nightly.html) (your setup seems to be Nightly Channel in Developer Mode), does not work with the packaged version of Ignition Gazebo. You have to build it using colcon and the `tags.yaml` that we provide.

Back to your problem, I think that the `LD_LIBRARY_PATH` variable that you export is messing up something. In general, its usage should be discouraged. If you notice, despite you have created a virtualenv, when you install gym-ignition with pip, the python package get installed somehow in another folder (I assume this one `export PYTHONPATH=/home/developer/other/gym-ignitionV2_ws/gym-ign-install/lib/python3.6/site-packages` since you exported the pythonpath).

If the python package installed in a virtualenv, and the CMake project in a custom prefix (as you do), the only thing you need you have to do is enabling the virtualenv and importing `gym_ignition` (or the low level `scenario_bindings`). When you load the python package, it load `_scenario_bindings.so` that transitively loads all the depending shared libraries (you can check if they are found using `ldd /path/to/_scenario_bindings.so`).

I would suggest to check these things. You can have a look as a reference to my personal setup I keep in this [Dockerfile](https://github.com/diegoferigo/development-iit/blob/master/RL/Dockerfile).

wolfgangschwab, 2020-06-14:
Hi @diegoferigo,
I'm trying to get the new version running but I still have many problems although I used the Dockerfile that you referred to. 

Main problem: 
I understand that I need to use a special version of ignition-citadel. So I used the commands from the dockerfile where the tag.yaml is used. The download works but the colcon command does not work. There are many warnings regarding cmake policies 0072 an 0079 which I ignored. But there are still stderr-outputs for 3 packages:

```
--- stderr: ignition-tools                                                                    
CMake Warning:
  Manually-specified variables were not used by the project:

    BUILD_TESTING

make: *** No rule to make target 'man'.  Stop.
---

```

```
-- stderr: sdformat9                                                                                                                                 
CMake Warning at CMakeLists.txt:224 (message):
  -- BUILD WARNINGS

CMake Warning at CMakeLists.txt:226 (message):
  -- 	Python psutil package not found.  Memory leak tests will be skipped

CMake Warning at CMakeLists.txt:228 (message):
  -- END BUILD WARNINGS

CMake Warning:
  Value of sdf_headers contained a newline; truncating

CMake Warning:
  Value of sdf_headers contained a newline; truncating

---
```

```
--- stderr: ignition-rendering3                                         
CUDA_TOOLKIT_ROOT_DIR not found or specified
-- BUILD WARNINGS
-- 	Cannot build component [optix] - Missing: OptiX
-- END BUILD WARNINGS

---
```

For my understanding the first one (regarding ignition-tools) is the important one. In the end the command `ign` is not available. 
How do you compile the newest ignition-citadel version?

diegoferigo, 2020-06-14:
I'm not a `colcon` expert but it my understanding is that it redirects all warning and errors to stderr. It seems that none of the output you posted is an actual error, all seem just warnings. I double checked them with the automatic build of the docker images that happen every week in our Continuous Integration pipeline, and [those warnings are there](https://github.com/robotology/gym-ignition/runs/748344265?check_suite_focus=true#step:4:6106) but the compilation of Ignition Gazebo proceeds successfully.

I think that you still didn't catch the real error, can you compare your full output with the one I just linked from CI?

wolfgangschwab, 2020-06-15:
Thank you for the hint. I now used the [Dockerfile.cicd-devel](https://github.com/robotology/gym-ignition/blob/devel/.docker/Dockerfile.cicd-devel) to generate an image. With this image I'm now able to build the devel branch and run the cartpole example:

```
root@wolfgang-desktop-2:/home/developer/other/gym-ignitionV2_ws/gym-ignition# python3 /home/developer/other/gym-ignitionV2_ws/gym-ignition/examples/python/launch_cartpole.py
WARN: Box bound precision lowered by casting to float32
Reward episode #0: 311.2632461916006
Reward episode #1: 813.8782971369577
Reward episode #2: 387.70479129875974
Reward episode #3: 895.25279191711
Reward episode #4: 531.8814380035373
Reward episode #5: 379.3318306853364
Reward episode #6: 590.1304833814571
Reward episode #7: 708.7281827391934
Reward episode #8: 431.16611148149207
Reward episode #9: 773.257460593163
```
So, looks good. 
As the next test I uncommented the two `env.render` commands to get a gui, but then I get an error. 

```
root@wolfgang-desktop-2:/home/developer/other/gym-ignitionV2_ws/gym-ignition# python3 /home/developer/other/gym-ignitionV2_ws/gym-ignition/examples/python/launch_cartpole_gui.py
WARN: Box bound precision lowered by casting to float32
[Err] [SystemLoader.cc:66] Failed to load system plugin [libignition-gazebo-scene-broadcaster-system.so] : couldn't find shared library.
Library error for [/workspace/install/lib/libignition-gazebo3-ign.a]: can't load /workspace/install/lib/libignition-gazebo3-ign.a
```
Is this expected or do I have an issue with my setup?

diegoferigo, 2020-06-15:
How did you build the image? Did you use the [docker-compose.yml](https://github.com/robotology/gym-ignition/blob/devel/.docker/docker-compose.yml) file?

I realized that the following should be ON if you want to get by default the CI configuration:

https://github.com/robotology/gym-ignition/blob/28d44844ef9e1273ac3433b31159fcf6ba33a277/.docker/Dockerfile.cicd-devel#L7

Otherwise GUIs will not work. The reason is that GUI are not compiled by colcon if the compilation is static (I think it is an upstream bug).

wolfgangschwab, 2020-06-15:
No, I directly used the Dockerfile.cicd-devel. 

Which command should I use?

diegoferigo, 2020-06-15:
`docker-compose build ci-devel`

wolfgangschwab, 2020-06-15:
OK, thanks. Using this docker solved some issues but the displaying of the GUI still had some problems. But I was able to solve them on my own. 
So, I close this issue.

closed 2020-06-15 by wolfgangschwab

diegoferigo, 2020-06-15:
I'm glad you managed to fix it. Was the GUI problem related to #211 or something else? If it could be helpful to other users feel free to add more details.

wolfgangschwab, 2020-06-16:
My problems doesn't seem to be related to #211. I got errors that the Display : 1 could not be displayed (or so). In the end I solved the problems by using another Nvidia docker image, which also includes opengl, as the basis by amending the `docker-compose.yml`. I added the following lines at the end:

```
  se-devel:
    build:
      args:
        from: nvidia/opengl:1.0-glvnd-devel-ubuntu18.04
        BUILD_SHARED_LIBS: "ON"
        CMAKE_BUILD_TYPE: Debug
      context: .
      dockerfile: Dockerfile.cicd-devel
    image: sopheng-gym-ign-devel:latest
```
Additionally I had to create an ignition folder (as already mentioned in another issue):
```
cd /root
mkdir .ignition
cd .ignition
mkdir gazebo
```

diegoferigo, 2020-06-16:
Yes, if you have an Nvidia-based setup that's the right way. The `ci-devel` image was designed to run in Github's servers in headless mode w/o GPU support.

I opened https://github.com/robotology/gym-ignition/pull/213 that solves the dot folder problem.