robotology-legacy/gym-ignition #210: Unable to run the example

TadielloM, 2020-06-04:
Hi,
I was trying to run the example of the cartpole but when I tried to run it I get the following error:
```standard_init_linux.go:211: exec user process caused "exec format error"```
I've seen that could be some incompatibility with the architecture. However, I'm running a Ubuntu 18.04 x86_64 on an Intel core i7, is the example on docker using a different OS?
How can I try to fix it?

diegoferigo, 2020-06-04:
Hi @TadielloM, we're in the process of updating and moving the documentation from the README (where is not very updated after the last refactoring) to the website.

In particular, the instruction you followed were referring to the `master` branch. I think we removed the shebang from the Python script some while ago, and never updated the instructions afterwards. In any case, there are few caveats in Ubuntu bionic regarding ogre2 that need to be addressed in the docker image so I don't have a fix on-the-fly.

I recommend to follow the [User Installation](https://robotology.github.io/gym-ignition/devel/installation/nightly.html#user-installation) of the devel branch and run the cartpole example from there. Those setup instruction have been written recently and they're not bullet proof yet, let me know if you find any trouble. I will open a new issue to describe how to handle the ogre2 problem on bionic.

diegoferigo, 2020-06-04:
Workaround for ogre2 described in https://github.com/robotology/gym-ignition/issues/211.

TadielloM, 2020-06-08:
Hi @diegoferigo, at the end, I was able to run the code locally but not in the docker container. I didn't have ```$HOME/.ignition/gazebo/``` folder. 
Probably I had some dependency missing.
However, I suggest to modify the demo part on the README temporarily or modify the container to make it work "out-of-the-box".

diegoferigo, 2020-06-08:
> Hi @diegoferigo, at the end, I was able to run the code locally but not in the docker container. I didn't have $HOME/.ignition/gazebo/ folder.

This is a upstream problem of Ignition Gazebo. That folder is a typical unix-like hidden folder in the user's home that stores the user configuration of the simulator. Somehow Ignition Gazebo does not create it if the folder does not exist. In my docker containers it's typically folder I want to keep as persistent and I mount it as a volume, but in the most general case you can solve this problem waiting an upstream fix just using `mkdir $HOME/.ignition/gazebo/` right after you create the container.

> However, I suggest to modify the demo part on the README temporarily or modify the container to make it work "out-of-the-box".

Yes thanks, it's in my TODO list.

Reaper151, 2020-06-17:
@diegoferigo I can't run launch_cartpole.py too. I install gym-ignition stable channel as developer in virtual environment.

My issue is nothing output in terminal after debug this line `env = randomizeers.cartpole_no_rand.CartpoleEnvNoRandomizations(env=make_env)` or 
`env = randomizers.cartpole.CartpoleEnvRandomizer(env=make_env, seed=42, num_physics_rollouts=5)`.

I got below snippet when I kill python script with `Ctrl+c`.
`WARN: Box bound precision lowered by casting to float32
Segmentation fault (core dumped)`

diegoferigo, 2020-06-17:
@Reaper151 There's something strange in your setup, if you installed the stable channel as developer you should not have the `CartpoleEnvNoRandomizations`. It was introduced recently and it's only present in the `devel` branch (nightly channel).

Master up to today is quite old and I warmly recommend to use the nightly channel (it requires building Ignition Robotics from source).

Reaper151, 2020-06-17:
@diegoferigo Oh yes. I clone devel repo but I install stable developer. However, I'm building Ignition libraries. I'll get back to you when I'm done installation and test that demo again.

diegoferigo, 2020-06-17:
If you keep having problems with the demo script, please enable debug messages uncommenting the following line and post the output

https://github.com/robotology/gym-ignition/blob/ea59cb9c4c52d2a64bb96eb3f0ead4d3bd1f8574/examples/python/launch_cartpole.py#L13

Reaper151, 2020-06-17:
Unfortunately,  @diegoferigo  I still can't run demo. 
For whatever this might worth, I already check integrity of purely Ignition and Ignition-ROS.

`[Msg] [GazeboSimulator.cpp:388] No world file passed, using the default empty world
[Dbg] [GazeboSimulator.cpp:643] Physics profile:
max_step_size=0.001
real_time_factor=3.40282e+38
real_time_update_rate=-1
[Msg] Loading SDF string. File path not available.
[Dbg] [EntityComponentManager.cc:627] Using components of type [2251689575469537287] / [ign_gazebo_components.World].
[Dbg] [EntityComponentManager.cc:627] Using components of type [13994732549916512682] / [ign_gazebo_components.Name].
[Dbg] [EntityComponentManager.cc:627] Using components of type [12592746352568925681] / [ign_gazebo_components.Gravity].
[Dbg] [EntityComponentManager.cc:627] Using components of type [13224937992534617849] / [ign_gazebo_components.MagneticField].
[Dbg] [EntityComponentManager.cc:627] Using components of type [3630648173860223239] / [ign_gazebo_components.PhysicsEnginePlugin].
[Dbg] [EntityComponentManager.cc:627] Using components of type [8753193699724811771] / [ign_gazebo_components.Wind].
[Dbg] [EntityComponentManager.cc:627] Using components of type [12173050716021724529] / [ign_gazebo_components.WorldLinearVelocity].
[Dbg] [EntityComponentManager.cc:627] Using components of type [15943768124495574352] / [ign_gazebo_components.WorldLinearVelocitySeed].
[Dbg] [EntityComponentManager.cc:627] Using components of type [3297509811873971798] / [ign_gazebo_components.ParentEntity].
[Dbg] [EntityComponentManager.cc:627] Using components of type [17100615127981600159] / [ign_gazebo_components.Scene].
[Dbg] [EntityComponentManager.cc:627] Using components of type [17605309075052480649] / [ign_gazebo_components.Atmosphere].
[Dbg] [EntityComponentManager.cc:627] Using components of type [8064491505919932473] / [ign_gazebo_components.Level].
[Dbg] [EntityComponentManager.cc:627] Using components of type [2668898242563798256] / [ign_gazebo_components.DefaultLevel].
[Dbg] [EntityComponentManager.cc:627] Using components of type [11371360182141354106] / [ign_gazebo_components.LevelEntityNames].
[Dbg] [ECMProvider.cpp:83] World 'default' successfully processed by ECMProvider
[Dbg] [SimulationRunner.cc:840] Loaded system [scenario::plugins::gazebo::ECMProvider] for entity [1]
[Dbg] [EntityComponentManager.cc:627] Using components of type [4981278897826323946] / [ign_gazebo_components.WorldSdf].
[Dbg] [EntityComponentManager.cc:627] Using components of type [3866641186784191835] / [ign_gazebo_components.Light].
[Dbg] [EntityComponentManager.cc:627] Using components of type [6612894081701502240] / [ign_gazebo_components.Pose].
[Msg] Loaded level [3]
[Msg] Serving world controls on [/world/default/control] and [/world/default/playback/control]
[Msg] Serving GUI information on [/world/default/gui/info]
[Msg] World [default] initialized with [default_physics] physics profile.
[Msg] Serving world SDF generation service on [/world/default/generate_world_sdf]
[Dbg] [GazeboSimulator.cpp:668] Starting the gazebo server
[Msg] Found no publishers on /stats, adding root stats topic
[Msg] Found no publishers on /clock, adding root clock topic
[Dbg] [SimulationRunner.cc:403] Creating PostUpdate worker threads: 1
[Dbg] [EntityComponentManager.cc:627] Using components of type [14753161793664412980] / [ign_gazebo_components.SimulatedTime].
[Dbg] [EntityComponentManager.cc:627] Using components of type [13081221419881009263] / [ign_gazebo_components.Timestamp].
[Dbg] [signals.cpp:94] Setting callback for signal SIGINT
[Dbg] [signals.cpp:94] Setting callback for signal SIGTERM
[Dbg] [signals.cpp:94] Setting callback for signal SIGABRT
[Dbg] [EntityComponentManager.cc:627] Using components of type [6687176221774458630] / [ign_gazebo_components.Model].
[Dbg] [EntityComponentManager.cc:627] Using components of type [8546580419506082455] / [ign_gazebo_components.Static].
[Dbg] [EntityComponentManager.cc:627] Using components of type [9712747055438129860] / [ign_gazebo_components.WindMode].
[Dbg] [EntityComponentManager.cc:627] Using components of type [5661073481138181711] / [ign_gazebo_components.SelfCollide].
[Dbg] [EntityComponentManager.cc:627] Using components of type [11683062252779233161] / [ign_gazebo_components.SourceFilePath].
[Dbg] [EntityComponentManager.cc:627] Using components of type [5081358965268446661] / [ign_gazebo_components.Link].
[Dbg] [EntityComponentManager.cc:627] Using components of type [8112400427272910195] / [ign_gazebo_components.Inertial].
[Dbg] [EntityComponentManager.cc:627] Using components of type [16454635107327670381] / [ign_gazebo_components.Visual].
[Dbg] [EntityComponentManager.cc:627] Using components of type [13011964647677164955] / [ign_gazebo_components.CastShadows].
[Dbg] [EntityComponentManager.cc:627] Using components of type [13440282432131634483] / [ign_gazebo_components.Transparency].
[Dbg] [EntityComponentManager.cc:627] Using components of type [17121648710877364109] / [ign_gazebo_components.Geometry].
[Dbg] [EntityComponentManager.cc:627] Using components of type [9853217982010720764] / [ign_gazebo_components.Material].
[Dbg] [EntityComponentManager.cc:627] Using components of type [17938588655714334139] / [ign_gazebo_components.Collision].
[Dbg] [EntityComponentManager.cc:627] Using components of type [9225962031573086509] / [ign_gazebo_components.CollisionElement].
[... cut ...]
[Dbg] [Physics.cpp:534] Loaded [ignition::physics::dartsim::Plugin] from library [/home/reaper/venv_DRL/repos/gym-ignition-nightly/install/lib/ign-physics-2/engine-plugins/libignition-physics-dartsim-plugin.so]
[Dbg] [SimulationRunner.cc:751] Loaded system [scenario::plugins::gazebo::Physics] for entity [1]
[Wrn] [ECMProvider.cpp:72] Resources of world default already inserted
[Dbg] [SimulationRunner.cc:840] Loaded system [scenario::plugins::gazebo::ECMProvider] for entity [1]
WARN: Box bound precision lowered by casting to float32
Segmentation fault (core dumped)
`

diegoferigo, 2020-06-17:
I just tried on my machine with last `devel`'s commit and it works fine. Let's try to understand what's wrong on your setup.

Can you write the precise sequence of commands you executed to 1) install the CMake project 2) Create and enable the virtual environment 3) Install the Python package?

Reaper151, 2020-06-17:
1. I create virtual environment according to this [link](https://robotology.github.io/gym-ignition/devel/installation/nightly.html#virtual-environment-optional) then I activate and stay in virtual environment the whole time of installation.

2. I follow [this instructions](https://robotology.github.io/gym-ignition/devel/installation/nightly.html#ignition-robotics) then I complete Ignition nightly installation by start with [install dependencies](https://ignitionrobotics.org/docs/citadel/install_ubuntu_src#install-dependencies) and finish with [using the workspace](https://ignitionrobotics.org/docs/citadel/install_ubuntu_src#using-the-workspace).

3. I test Ignition with getting started command ( shape.sdf ) and test Ignition-ROS with (diff-drives) and both succeed.

4. Then I follow developer installation for nightly but at `cmake ..`.
I encounter with swig incompatibility even though I done [swig setup](https://robotology.github.io/gym-ignition/devel/installation/nightly.html#ignition-robotics) beforehand. However, I continue  `cmake ..` but I still encounter with swig incompatibility, so I just delete whole `build` folder and it works.

5. I installed python package with `pip install -e .`

diegoferigo, 2020-06-17:
The steps seem correct. Does it segfault also if you execute line by line in an interactive session the following? If yes, which line?

```python
import gym_ignition_models
from gym_ignition.utils import scenario
from gym_ignition_environments import models

gazebo, world = scenario.init_gazebo_sim()
cartpole = models.cartpole.CartPole(world=world)

cartpole.get_joint("pivot").reset(position=0.5, velocity=2.0)

for _ in range(10):
    gazebo.run()
    print(cartpole.get_joint("pivot").position())
```

This is the simplest standalone code you can execute (I have few similar tutorials pending).

---

Regarding the swig problem, I suspect that after step 2) you have `swig3` installed from apt, and after step 4) you also have `swig4` installed from sources. For some reason the first one is detected, that on bionic it is too old. Did you perhaps run CMake in gym-ignition build folder *before* installing swig from sources? In that case, `swig3` is detected and cached. Even if you install `swig4` later, it won't get detected and the cached resources are used.

Solution: do not install `swig` from apt. I will update the install guide accordingly.

Reaper151, 2020-06-17:
I probably do run CMake in gym-ignition build folder before built SWIG from sources which leads to SWIG incompatibiliy. Then I built swig from sources, ran CMake again, failed again, then I deleted whole build folder and ran CMake turns out it passed ( I wouldn't said it works because it's not lol)

Sincerely,
Wiwat Tulapornpipat

Reaper151, 2020-06-18:
I fresh installed ignition-nightly and gym-ignition-nightly with steps below. (SPOILED: still not working)

1. Delete whole gym-ignition, ignition-nightly folder
2. Export CC and CXX environment variables
3. Build [SWIG ](https://robotology.github.io/gym-ignition/devel/installation/nightly.html#swig)
4. Get Ignition-nightly [sources](https://robotology.github.io/gym-ignition/devel/installation/nightly.html#ignition-robotics)
5. Install [dependencies](https://ignitionrobotics.org/docs/citadel/install_ubuntu_src#install-dependencies)
6. Built Ignition-nightly libraries
7. Activate ignition-nightly workspace
8. Test Ignition nightly with (shape.sdf) and Ignition-nightly integrate with ROS (diff_drive.launch) -> Succeed
9. Clone gym-ignition devel repo
10.Follow developer installation --> SWIG Incompatibility  --> Delete "build folder in gym-ignition-nightly folder" --> Built SWIG as step 3 --> `sudo cmake ..` and get "Configuration done" prompt
10.1 I suspect that there is no done or succeed prompt when I use `cmake --build .` and `cmake --build . --target install`

> cmake --build .
>
> /usr/bin/cmake -S/home/reaper/venv_DRL/repos/gym-ignition-nightly -B/home/reaper/venv_DRL /repos/gym-ignition-nightly/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/reaper/venv_DRL/repos/gym-ignition-nightly/build/CMakeFiles /home/reaper/venv_DRL/repos/gym-ignition-nightly/build//CMakeFiles/progress.marks
/usr/bin/make -s -f CMakeFiles/Makefile2 all
/usr/bin/cmake -E cmake_progress_start /home/reaper/venv_DRL/repos/gym-ignition-nightly/build/CMakeFiles 0
>
> ---------------------------------------------------------------------------
> cmake --build . --target install
>
> /usr/bin/cmake -S/home/reaper/venv_DRL/repos/gym-ignition-nightly -B/home/reaper/venv_DRL/repos/gym-ignition-nightly/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/reaper/venv_DRL/repos/gym-ignition-nightly/build/CMakeFiles /home/reaper/venv_DRL/repos/gym-ignition-nightly/build//CMakeFiles/progress.marks
/usr/bin/make -s -f CMakeFiles/Makefile2 all
/usr/bin/cmake -E cmake_progress_start /home/reaper/venv_DRL/repos/gym-ignition-nightly/build/CMakeFiles 0
/usr/bin/make -s -f CMakeFiles/Makefile2 preinstall
Install the project...
/usr/bin/cmake -P cmake_install.cmake
-- Install configuration: "Release"
> 
> ---------------------------------------------------------------------------
10.2 But I got "Successfully installed gym-ignition"
>pip install -e .

>Obtaining file:///home/reaper/venv_DRL/repos/gym-ignition-nightly
Requirement already satisfied: gym>=0.13.1 in /home/reaper/venv_DRL/repos/gym (from gym-ignition==1.0b1.post1.dev495) (0.17.2)
Requirement already satisfied: numpy in /home/reaper/venv_DRL/lib/python3.6/site-packages (from gym-ignition==1.0b1.post1.dev495) (1.18.5)
Requirement already satisfied: gym_ignition_models in /home/reaper/venv_DRL/lib/python3.6/site-packages (from gym-ignition==1.0b1.post1.dev495) (0.1.dev76)
Requirement already satisfied: lxml in /home/reaper/venv_DRL/lib/python3.6/site-packages (from gym-ignition==1.0b1.post1.dev495) (4.5.1)
Requirement already satisfied: scipy in /home/reaper/venv_DRL/lib/python3.6/site-packages (from gym>=0.13.1->gym-ignition==1.0b1.post1.dev495) (1.4.1)
Requirement already satisfied: pyglet<=1.5.0,>=1.4.0 in /home/reaper/venv_DRL/lib/python3.6/site-packages (from gym>=0.13.1->gym-ignition==1.0b1.post1.dev495) (1.5.0)
Requirement already satisfied: cloudpickle<1.4.0,>=1.2.0 in /home/reaper/venv_DRL/lib/python3.6/site-packages (from gym>=0.13.1->gym-ignition==1.0b1.post1.dev495) (1.3.0)
Requirement already satisfied: future in /home/reaper/venv_DRL/lib/python3.6/site-packages (from pyglet<=1.5.0,>=1.4.0->gym>=0.13.1->gym-ignition==1.0b1.post1.dev495) (0.18.2)
Installing collected packages: gym-ignition
  Running setup.py develop for gym-ignition
Successfully installed gym-ignition

11. I ran your simplest standalone code.
> Python 3.6.9 (default, Apr 18 2020, 01:56:04) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>.> import gym_ignition_models
>.> from gym_ignition.utils import scenario
>.> from gym_ignition_environments import models
>.> gazebo, world = scenario.init_gazebo_sim()
[Msg] [GazeboSimulator.cpp:597] Using default empty world
[Msg] Loading SDF string. File path not available.
[Msg] Loaded level [3]
[Msg] Serving world controls on [/world/default/control] and [/world/default/playback/control]
[Msg] Serving GUI information on [/world/default/gui/info]
[Msg] World [default] initialized with [default_physics] physics profile.
[Msg] Serving world SDF generation service on [/world/default/generate_world_sdf]
[Msg] Found no publishers on /stats, adding root stats topic
[Msg] Found no publishers on /clock, adding root clock topic
[Msg] [Model.cpp:148] Model: [5] ground_plane
[Msg] [Model.cpp:152] Links:
[Msg] [Link.cpp:128]   [6] link
[... 90 lines cut ...]
[Dbg] [EntityComponentManager.cc:627] Using components of type [9853217982010720764] / [ign_gazebo_components.Material].
[Dbg] [EntityComponentManager.cc:627] Using components of type [17938588655714334139] / [ign_gazebo_components.Collision].
[Dbg] [EntityComponentManager.cc:627] Using components of type [9225962031573086509] / [ign_gazebo_components.CollisionElement].
[... cut ...]
**### [Err] [Physics.cpp:501] Failed to find plugin [libignition-physics-dartsim-plugin.so]. Have you checked the IGN_GAZEBO_PHYSICS_ENGINE_PATH environment variable?**
[Dbg] [SimulationRunner.cc:751] Loaded system [scenario::plugins::gazebo::Physics] for entity [1]
[Wrn] [ECMProvider.cpp:72] Resources of world default already inserted
[Dbg] [SimulationRunner.cc:840] Loaded system [scenario::plugins::gazebo::ECMProvider] for entity [1]
**### WARN: Box bound precision lowered by casting to float32
Segmentation fault (core dumped)**

Reaper151, 2020-06-18:
I also tried docker demo but it's working too.

> rocker --x11 --nvidia diegoferigo/gym-ignition /github/examples/python/launch_cartpole.py
Traceback (most recent call last):
  File "/home/reaper/.local/lib/python3.6/site-packages/urllib3/connectionpool.py", line 677, in urlopen
    chunked=chunked,
  File "/home/reaper/.local/lib/python3.6/site-packages/urllib3/connectionpool.py", line 392, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.6/http/client.py", line 1264, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.6/http/client.py", line 1310, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.6/http/client.py", line 1259, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.6/http/client.py", line 1038, in _send_output
    self.send(msg)
  File "/usr/lib/python3.6/http/client.py", line 976, in send
    self.connect()
  File "/home/reaper/.local/lib/python3.6/site-packages/docker/transport/unixconn.py", line 43, in connect
    sock.connect(self.unix_socket)
PermissionError: [Errno 13] Permission denied

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/reaper/.local/lib/python3.6/site-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/home/reaper/.local/lib/python3.6/site-packages/urllib3/connectionpool.py", line 725, in urlopen
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
  File "/home/reaper/.local/lib/python3.6/site-packages/urllib3/util/retry.py", line 403, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "/home/reaper/.local/lib/python3.6/site-packages/urllib3/packages/six.py", line 734, in reraise
    raise value.with_traceback(tb)
  File "/home/reaper/.local/lib/python3.6/site-packages/urllib3/connectionpool.py", line 677, in urlopen
    chunked=chunked,
  File "/home/reaper/.local/lib/python3.6/site-packages/urllib3/connectionpool.py", line 392, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.6/http/client.py", line 1264, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.6/http/client.py", line 1310, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.6/http/client.py", line 1259, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.6/http/client.py", line 1038, in _send_output
    self.send(msg)
  File "/usr/lib/python3.6/http/client.py", line 976, in send
    self.connect()
  File "/home/reaper/.local/lib/python3.6/site-packages/docker/transport/unixconn.py", line 43, in connect
    sock.connect(self.unix_socket)
urllib3.exceptions.ProtocolError: ('Connection aborted.', PermissionError(13, 'Permission denied'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/reaper/.local/bin/rocker", line 11, in <module>
    sys.exit(main())
  File "/home/reaper/.local/lib/python3.6/site-packages/rocker/cli.py", line 44, in main
    extension_manager.extend_cli_parser(parser, default_args)
  File "/home/reaper/.local/lib/python3.6/site-packages/rocker/core.py", line 89, in extend_cli_parser
    p.register_arguments(parser, default_args)
  File "/home/reaper/.local/lib/python3.6/site-packages/rocker/extensions.py", line 113, in register_arguments
    parser.add_argument('--network', choices=[n['Name'] for n in client.networks()],
  File "/home/reaper/.local/lib/python3.6/site-packages/docker/api/network.py", line 38, in networks
    res = self._get(url, params=params)
  File "/home/reaper/.local/lib/python3.6/site-packages/docker/utils/decorators.py", line 46, in inner
    return f(self, *args, **kwargs)
  File "/home/reaper/.local/lib/python3.6/site-packages/docker/api/client.py", line 230, in _get
    return self.get(url, **self._set_request_timeout(kwargs))
  File "/home/reaper/.local/lib/python3.6/site-packages/requests/sessions.py", line 543, in get
    return self.request('GET', url, **kwargs)
  File "/home/reaper/.local/lib/python3.6/site-packages/requests/sessions.py", line 530, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/reaper/.local/lib/python3.6/site-packages/requests/sessions.py", line 643, in send
    r = adapter.send(request, **kwargs)
  File "/home/reaper/.local/lib/python3.6/site-packages/requests/adapters.py", line 498, in send
    raise ConnectionError(err, request=request)
requests.exceptions.ConnectionError: ('Connection aborted.', PermissionError(13, 'Permission denied'))

diegoferigo, 2020-06-18:
> I also tried docker demo but it's working too.

This error seems `rocker` related. None related to the image itself.

> I fresh installed ignition-nightly and gym-ignition-nightly with steps below. (SPOILED: still not working)

Ok great, this is more informative. Are you sure you sourced the colcon workspace before running the commands that failed? It seems not finding the variable `IGN_GAZEBO_PHYSICS_ENGINE_PATH` that is not our responsibility. This path is used to find the plugin of DART that is installed by ign-physics (actually, `this->dataPtr->pluginPathEnv` contains the default `IGNITION_PHYSICS_ENGINE_INSTALL_DIR` that should be set the CMake). For some reason on your setup this is not detected automatically. Let's do a test, what if you `find /path/to/the/workspace/install | grep libignition-physics-dartsim-plugin.so`, and then export `IGN_GAZEBO_PHYSICS_ENGINE_PATH=<thispath>`? We can figure out later what's missing after things start working.

Reaper151, 2020-06-18:
After I export 'IGN_GAZEBO_PHYSICS_ENGINE_PATH' it still failed to find physics plugin.
I done your substitution code but I stuck at SWIG again.
No matter I install SWIG with [your instructions](http://lfs.linux-sysadmin.com/blfs/view/svn/general/swig.html) or [this instructions ](http://lfs.linux-sysadmin.com/blfs/view/svn/general/swig.html) after make clean 'build' folder, I can't pass SWIG issue.

BTW, I have gym-gazabo in my system too. I don't know whether this cause the issue or not.
Now, I'm thinking about create fresh venv and do it all again but from what I understand even I do that I'll get SWIG incompatibility anyway because it isn't pip package.

diegoferigo, 2020-06-18:
> I done your substitution code but I stuck at SWIG again.

I believe that if you `sudo apt remove swig` you would solve this issue. Having two versions of the same library in different places of your fs (`/usr/lib` and `/usr/local/lib` in your case) is a bit borderline if you are not very familiar and know how to handle it. If you're lucky, no other packages depend on the `swig` deb package and you can uninstall it easily.

> BTW, I have gym-gazabo in my system too.

This means that you also have Gazebo Classic installed, with all its dependencies that despite have been designed to coexist with Ignition Gazebo, it makes your setup quite peculiar. For these setup I typically advocate for docker-based solutions because it helps a lot on removing cross-talk with other projects. However, let's try to make it work without it and find what's the root cause. Often, without docker, cleaning properly a system is not straightforward and minor details that we cannot know are crucial.

As I wrote above, I would recommend using the nightly installation in user mode, it reduces the steps. The drawback is that you no longer have access to the build folder, therefore swig 4 must be detected by default.

Reaper151, 2020-06-18:
How can I ensure that swig 4 is detected by default? 
I noticed that dependencies installation command for Bionic which you suggest has swig in it. So, SWIG4 installation procedure should be under Ignition-nightly procedure.
>sudo apt-get install cmake freeglut3-dev libavcodec-dev libavdevice-dev libavformat-dev libavutil-dev libdart6-collision-ode-dev libdart6-dev libdart6-utils-urdf-dev libfreeimage-dev libgflags-dev libglew-dev libgts-dev libogre-1.9-dev libogre-2.1-dev libprotobuf-dev libprotobuf-dev libprotoc-dev libqt5core5a libswscale-dev libtinyxml2-dev libtinyxml-dev pkg-config protobuf-compiler python qml-module-qt-labs-folderlistmodel qml-module-qt-labs-settings qml-module-qtquick2 qml-module-qtquick-controls qml-module-qtquick-controls2 qml-module-qtquick-dialogs qml-module-qtquick-layouts qml-module-qtqml-models2 qtbase5-dev qtdeclarative5-dev qtquickcontrols2-5-dev ruby ruby-ronn uuid-dev libzip-dev libjsoncpp-dev libcurl4-openssl-dev libyaml-dev libzmq3-dev libsqlite3-dev libwebsockets-dev  **swig** ruby-dev -y

When I ran `pip list | grep ignition`, it listed `gym-ignition` not nightly though. 
However, it listed `gym-ignition-nightly` after I install gym-ignition-nightly as user but my physics path environment variable also broken!! 

I went back to built gym-ignition-nightly as developer again and my physics path is fine but as  I mention `pip list` gave `gym-ignition 1.0b1.post1.dev495` to me.

I'm so confused now. TT

diegoferigo, 2020-06-18:
> I noticed that dependencies installation command for Bionic which you suggest has swig in it. So, SWIG4 installation procedure should be under Ignition-nightly procedure.

Yes, this is what I meant in https://github.com/robotology/gym-ignition/issues/210#issuecomment-645454311. I will update the docs accordingly.

> When I ran pip list | grep ignition, it listed gym-ignition not nightly though.

It's packaging burden, don't mind about names. When we upload packages to PyPI we do some renaming. Otherwise we couldn't use a single `setup.py` file for both channels. This approach is quite common in the Python world.

I'm confused as well, the simplest configuration to debug your setup problem is the following. Please do this from scratch otherwise we're going to loop forever:

- Clean for good your system. Be sure you have a virtualenv only for this project and don't have any ignition-related filesin your home, `/usr/`, and in `/usr/local`. It's not uncommon that using `sudo` as you did, some files have been installed in a system folder without noticing. Use `find <folder> | grep ignition` to check. If you don't need Gazebo Classic, uninstall it. Be sure you don't have any ignition library installed with `dpkg -l *ignition* | grep ^ii`.
- Use a new virtualenv only for gym-ignition
- Uninstall swig with apt and install it from sources as explained in the installation guide of gym-ignition
- Install Ignition Gazebo dependencies omitting `swig`
- Install Ignition Gazebo with colcol in a folder of your home without using sudo
- Clone the devel branch of gym-ignition
- Install the devel branch with `pip install .`

After all of this, you should have in `pip list` a package `gym-ignition 1.0b1.post1.dev495`.

Reaper151, 2020-06-18:
I used Synaptic with completely removal action with all of ignition and check with `dpkg -l *ignition* | grep ^ii` , `find /usr/ | grep ignition` , `find /usr/local | grep ignition`

May I put ignition-nightly in this order `/home/reaper/venv_drl/repos/ignition-nightly` because I would like to contain all of repos related to this venv in the same place?

diegoferigo, 2020-06-18:
This is something I didn't get from your setup. The python virtualenv is one thing that should remain standalone. A possible structure if you want a single folder is the following:

```
DRL/
├── repos
│   ├── gym-ignition
│   └── gym-mynewnenv
├── venv
└── workspace
```

- `venv` is the Python virtual environment created with `virtualenv`.
- `repos` all the sources, in your case only gym-ignition but then if your plan is to develop an environment with it, you could add your own `gym-<mynewnev>` repo.
- `workspace` is the colcol workspace. I would keep it separated because it's not only sources as the repos in `repos` but it also contains the install tree (in `workspace/install`).

Reaper151, 2020-06-18:
Finally, I got something make sense but I noticed there's  `Segmentation fault (core dumped)` again.

> WARN: Box bound precision lowered by casting to float32
Reward episode #0: 311.2632461916006
Reward episode #1: 813.8782971369577
Reward episode #2: 387.70479129875974
Reward episode #3: 895.25279191711
Reward episode #4: 531.8814380035373
Reward episode #5: 379.3318306853364
Reward episode #6: 590.1304833814571
Reward episode #7: 708.7281827391934
Reward episode #8: 431.16611148149207
Reward episode #9: 773.257460593163
**double free or corruption (fasttop)
Segmentation fault (core dumped)**

diegoferigo, 2020-06-18:
Ok we're on the right way. Now the output is finally correct. Let's try to solve now that double free that does not occur in any of our systems.

Does it occur also if you run the minimal code of https://github.com/robotology/gym-ignition/issues/210#issuecomment-645454311?

Reaper151, 2020-06-19:
I got `double free or corruption (fasttop)` even when I ran `ign gazebo shapes.sdf` maybe I should try to install ign again.

diegoferigo, 2020-06-19:
Lucky case, debugging that would have been a pain. Did you validate it with the `ldd` command?,

traversaro, 2020-06-19:
We had a similar ROS2 conflict problem (even if there was at configuration time) in https://github.com/robotology/gym-ignition/issues/118#issuecomment-575372388 .

Reaper151, 2020-06-20:
@diegoferigo Yes, I have exact same results. I found that libdart6 depends on liboctomap1.8 for collision after I remove  liboctomap1.8 instead of remove ros-melodic-octomap because ros-melodic-octomap is crucial in ros-melodic but this didn't work. So, I removed ros-melodic-octomap, installed Ignition library/ gym-ignition then reinstall ros-melodic and other packages.

diegoferigo, 2020-06-22:
I'm glad you managed to solve it. Even though it works, every time you need to compile again the project, your setup will break again, and removing the octomap package every time could be an annoying process. Maybe you can find another workaround playing with few symlinks, `LD_LIBRARY_PATH`, or `LD_PRELOAD`. Closing.

closed 2020-06-22 by diegoferigo