robotology-legacy/gym-ignition #214: Using Panda ingym-ignition

Reaper151, 2020-07-03:
I tried to use Panda as a substitution environment of cartpole in gym-ignition but I can't only find randomizer for cartpole and pendulum. 
Do I have to create every python config files -> randomizer, tasks ??

diegoferigo, 2020-07-03:
The resources of [`gym_ignition_environments`](https://github.com/robotology/gym-ignition/tree/devel/python/gym_ignition_environments) package are meant to become a demo / tutorial in the future and only show the usage pattern we envision for the framework. We don't aim to create and maintain environments as in [openai/gym](https://github.com/openai/gym/tree/master/gym/envs/robotics), but rather provide a powerful framework that enables researchers to easily create their own.

I moved the issue to the gym-ignition repo since it's not related to the models.

diegoferigo, 2020-07-07:
A possible starting point for using the Panda is  [`test_custom_controllers.py`](https://github.com/robotology/gym-ignition/blob/devel/tests/test_scenario/test_custom_controllers.py). In this test we use the Panda to test a fixed-base computed torque controller. Another possibility to move the robotic arm is tuning and applying position PIDs.

closed 2020-07-14 by diegoferigo