robotology-legacy/gym-ignition #218: Randomized inertia tensor does not satisfy the triangle inequality

diegoferigo, 2020-07-15:
The recently introduced SDF randomizer #177, part of the bigger randomization plan defined in #41, allows to randomize mostly all entries of an SDF model taking advantage of XPaths.

This is applied also to the [inertia parameters](http://sdformat.org/spec?ver=1.7&elem=link#inertial_inertia) that define the inertia matrix. However, sampling the parameters from a random distribution sometimes leads to a parameters that do not satisfy the triangle inequality, obtaining an invalid model (check performed in [`ignition::math::MassMatrix3::ValidMoments`](https://github.com/ignitionrobotics/ign-math/blob/9ade6165e790f766f8bbe6c0d8dc78f24ff88c69/include/ignition/math/MassMatrix3.hh#L455-L466).

Refer to [`test_sdf_randomizer`](https://github.com/robotology/gym-ignition/blob/c2fd6390aba839b64d0de0d63748de4baad62067/tests/test_gym_ignition/test_sdf_randomizer.py) for usage examples.

We should add a new randomization method specific for the inertia tensor that samples from a physically consistent distribution, something like:

```python
randomizer.new_randomization() \
    .at_xpath("*/link/inertial/inertia") \
    .is_inertia(True, only_diagonal=True) \
    .method(Method.Coefficient) \
    .sampled_from(Distribution.Gaussian, GaussianParams(mean=1, variance=0.1)) \
    .add()
```

so that `ixx`, `iyy` and `izz` are sampled consistently.

HorvathDawson, 2021-10-13:
I am working on implementing a version of this for my own use that I could create a pull request for after.  I have two ideas on how to achieve this without adding bias. 

1. Only allow coefficient scaling. Then scale each  principal inertia by this value. (the issue with this is that it will only randomize the inertia magnitude instead of shifting the centre of mass around randomly. This is a bit restrictive.
2. To avoid this another option would be to perform some random method to each principal inertia independently then check if these values satisfy the inequality. If they do not satisfy the inequality then repeat until they do. This will have a longer run time but avoids adding accidental bias into the chosen values by forcing the constraints before randomization. (This would need a max number of attempts)  

Do either of these options sound like a good way to implement this randomization?

HorvathDawson, 2021-10-14:
I made a small google colab which also includes a randomization using the constraints of the triangle inequalities to avoid permutations.  [https://colab.research.google.com/drive/1BQRnzZ69jncUPdaVWVzR6ZIeWmAugGf3?usp=sharing](url)

However,I am not convinced this avoids biasing the output to the clipped values... There might be a much more intelligent way to do this.

AndrejOrsula, 2021-10-14:
Hello @HorvathDawson, great to see that someone is looking into this.

I implemented a simple randomization of inertial properties within the context of `Gym-Ignition` some time ago. I used the approach you described in your first point (my assumption/limitation was that all objects have a uniform density).

> 1. Only allow coefficient scaling. Then scale each principal inertia by this value. (the issue with this is that it will only randomize the inertia magnitude instead of shifting the centre of mass around randomly. This is a bit restrictive.)

Therefore, I thought I would share the code. It's an old and messy Python script, but there might be something useful (or not). Here, all links of models are assumed to have a mesh geometry (one or more `visual`), and [trimesh](https://trimsh.org) is used for most of the mesh analysis and manipulation.

- [randomization of inertial properties](https://github.com/AndrejOrsula/drl_grasping/blob/2a58185048328d8f7f0a9b7e1f743e77e73863e9/drl_grasping/utils/model_collection_randomizer.py#L465-L479)

And then also some other stuff within this context that might be relevant for your application (or not).

- [estimation of model's inertial properties](https://github.com/AndrejOrsula/drl_grasping/blob/2a58185048328d8f7f0a9b7e1f743e77e73863e9/drl_grasping/utils/model_collection_randomizer.py#L353-L381) + [simple checking and writing into file](https://github.com/AndrejOrsula/drl_grasping/blob/2a58185048328d8f7f0a9b7e1f743e77e73863e9/drl_grasping/utils/model_collection_randomizer.py#L287-L295)
- [randomization of model's scale](https://github.com/AndrejOrsula/drl_grasping/blob/2a58185048328d8f7f0a9b7e1f743e77e73863e9/drl_grasping/utils/model_collection_randomizer.py#L423-L463)

closed 2025-01-30 by diegoferigo