robotology-legacy/gym-ignition #244: Default CMake installation procedure result in file being installed outside of CMAKE_INSTALL_PREFIX

traversaro, 2020-09-25:
The bindings file thanks to the logic in https://github.com/robotology/gym-ignition/blob/ae2154690356431e71203201448bb9f3bbad5c3e/bindings/CMakeLists.txt#L43 are installed in `Python3_SITELIB` . However, as noted inhttps://github.com/robotology/gym-ignition/pull/170#issuecomment-610969105 in a non-virtualenv enviroment, `Python3_SITELIB` points to python directories under `/usr`, that should never be used for installing files.

diegoferigo, 2020-09-25:
Thanks for reporting @traversaro! We warmly recommend installing the package inside a virtualenv, but I understand that not every user follows our recommendations :smile: 

I think this is the expected behaviour. When the active Python interpreter is the system interpreter, there are indeed files installed under `/usr`. If you don't want this and don't want to use folder not owned by your user, you should enable a Python environment where you have writing permissions.

This does not mean, however, that we cannot improve it. There is a Python tree inside the user's home that can be used. Although, as far as I know, also the most recent CMake versions do not recognize it. When a virtualenv is not enabled, if we want to go in this direction, we should check if we have writing access to the existing Python tree, and if not, revert to the tree in the user's home. To be honest, I don't like it much. It means again installing outside `CMAKE_INSTALL_PREFIX`, but in this case in a place where CMake can write, all behind the neck of the user. The logic of: _we install the bindings for the active python interpreter, if this is the system one you need sudo_ it seems more clear to me.

traversaro, 2020-09-26:
I think you suggestions go in the direction of solving the problem "When emulating the behavior of pip install and there is no active environment, the compilation and install with default options on Ubuntu 20.04 install stuff in the `/usr`, while nothing from the user should be install in that prefix (the prefix for global user installed software is `/usr/local` ". 

However, my problem is quite different, even if probably I did not framed it correctly in the original post: if I am already installing the project in a "workspace" fashion (i.e. by installing all the related software packages in a given user directory) but using a tool different from `virtualenv`, for example using a [YCM-based superbuild](http://robotology.github.io/ycm/gh-pages/v0.11/manual/ycm-superbuild.7.html), [colcon](https://colcon.readthedocs.io/en/released/) or [conda](https://docs.conda.io/en/latest/), I want to be able to obtain the normal behavior of "do not install anything outside of `CMAKE_INSTALL_PREFIX`". This can be achieved in several ways, for example permitting user to optionally specify manually `BINDINGS_INSTALL_PREFIX`, or having a dedicated option to toggle the value of  `BINDINGS_INSTALL_PREFIX`.

> We warmly recommend installing the package inside a virtualenv, but I understand that not every user follows our recommendations 😄

If not following a recommendation result in a failure, then the recommendation is probably actually a requirement!

traversaro, 2020-09-26:
To clarify the point of the issue, I updated the title of the issue.

traversaro, 2020-09-26:
> I think you suggestions go in the direction of solving the problem "When emulating the behavior of pip install and there is no active environment, the compilation and install with default options on [...]

> > We warmly recommend installing the package inside a virtualenv, but I understand that not every user follows our recommendations 😄
> 
> If not following a recommendation result in a failure, then the recommendation is probably actually a requirement!

Actually, now that I read in https://docs.python.org/3/install/#how-installation-works, I realize that probably by default `setup.py install` installs under `/usr`, so indeed the system is correctly working as expected, from that point of view there is nothing unexpected, the actual point of the issue was another.

diegoferigo, 2020-09-27:
> > I think you suggestions go in the direction of solving the problem "When emulating the behavior of pip install and there is no active environment, the compilation and install with default options on Ubuntu 20.04 install stuff in the `/usr`, while nothing from the user should be install in that prefix (the prefix for global user installed software is `/usr/local`) ".
> 
> > > We warmly recommend installing the package inside a virtualenv, but I understand that not every user follows our recommendations smile
> > 
> > 
> > If not following a recommendation result in a failure, then the recommendation is probably actually a requirement!
> 
> Actually, now that I read in https://docs.python.org/3/install/#how-installation-works, I realize that probably by default `setup.py install` installs under `/usr`, so indeed the system is correctly working as expected, from that point of view there is nothing unexpected, the actual point of the issue was another.

If you execute the `setup.py` script with sudo, the prefix is inside `/usr`, correct. Then, I completely agree that `/usr/local/lib` should be used instead of `/usr/lib` and this has to be fixed. This is not the default behaviour that can be obtained by CMake though. I propose to 1) locate the active interpreter 2) if it is not `/usr/bin/*` proceed (venv, custom installation, ...), otherwise insert `/local/` in the install path.

> However, my problem is quite different, even if probably I did not framed it correctly in the original post: if I am already installing the project in a "workspace" fashion (i.e. by installing all [...]

Thanks for clarifying your main point. I think we discussed something similar in https://github.com/robotology/robotology-superbuild/pull/359 without converging to a final solution, how to handle this use case is still open. We can definitely trying to mimic the approach we already use in iDynTree and YARP also for this repo. It will simplify integrating it in the superbuild in the future. However, as I commented in the linked PR, this is something quite custom that we're used to do.

I keep this issue open and add this small task to the v1 list. I like the idea of overriding `BINDINGS_INSTALL_PREFIX`. This variable could be cached in CMake and, if the user specifies it in the configuration stage, overrides the logic we are currently using.

traversaro, 2020-09-27:
> However, as I commented in the linked PR, this is something quite custom that we're used to do.

diegoferigo, 2020-09-27:
Here you have a better knowledge of what other projects do. This does not mean that this is the preferred way to proceed :) This approach always requires exporting `PYTHONPATH` for importing the packages, and if your system has multiple python versions and your bindings are not ABI compatible with different versions of Python I suspect you get funny results.

I think there are reasons and specific use-cases why the most modern CMake versions started to detect virtualenv and custom Python installations.

traversaro, 2020-09-27:
> Here you have a better knowledge of what other projects do. This does not mean that this is the preferred way to proceed :) 

To clarify, I never wanted to change the default behavior, just have a way to install everything in `CMAKE_INSTALL_PREFIX` if required. The preferred way to proceed is something that depend on the context, and the default depends on the main target audience.

closed 2020-11-02 by diegoferigo

traversaro, 2021-02-28:
I now understand a bit more about the Python installation process (thanks to https://github.com/dic-iit/bipedal-locomotion-framework/issues/217 and the linked issues), and I think I may have find a way to avoid violating POLA (https://en.wikipedia.org/wiki/Principle_of_least_astonishment) for both C++/CMake and Python/venv users. 

First of all, a brief introduction to our two target groups.

## C++/CMake users
He/she typically installs projects with:
~~~ 
git clone https://github.com/<project>
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX=<local_install_prefix> -DCMAKE_BUILD_TYPE=Release ..
cmake --build . --config Release
cmake --install . --config Release
~~~

The (hidden) assumption that it may be convenient not to violate for this user is that files are not installed **outside** the specified `CMAKE_INSTALL_PREFIX` .

## Python/venv users
He/she typically installs projects with (correct me if I am wrong here, definitely not my area of expertise):
~~~ 
git clone https://github.com/<project>
pip install .
~~~

The (hidden) assumption here is that if a virtual environment is active, all the files are installed there and not in some set by default `CMAKE_INSTALL_PREFIX`.

## Proposal

If we agree that this are two main category of users we target, can't we simply keep the CMake options to install in the virtualenv directories off by default (so that they C++ users don't get them as they directly invoke cmake), while for the Python/venv users can't we call CMake as part of the setup.py script, enabling in that case the CMake options to install in the active environment?

traversaro, 2021-02-28:
cc @xEnVrE that could also be interested in this discussion as he is a mixed C++/Python user.

diegoferigo, 2021-03-01:
Thanks @traversaro for resuming the discussion. We never had much experience with Python, and particularly hybrid Python/C++ projects, and in the past couple of years we learned a lot (and make quite a few mistakes on the road). I believe that now that we started more integration with Python, we should fix the following two problems:

1) This one: unify the installation logic
2) PEP600 support (tooling is almost ready, it's matter of weeks)

The second is more about binary distribution, and it has a lower priority. It can also be kind of replaced by the recent effort of @traversaro to bring robotology projects in a custom conda channel.

This being said, I agree on the separation of the two target groups. If we keep using [diegoferigo/cmake-build-extension](https://github.com/diegoferigo/cmake-build-extension/) for the packaging, integrating CMake in the `setup.py` should be straightforward also for maintainers that do not have an extensive knowledge of Python packaging.

I have some follow up questions and observations:

1. When you talk about virtualenvs, do you also include the system's interpreter? What I mean is, in the case without a virtualenv and a custom `CMAKE_INSTALL_PREFIX`, do you suggest to avoid installing the python resources in the system's directories?
1. For gym-ignition, you're suggesting to make the behavior currently optionally enabled with `BINDINGS_INSTALL_PREFIX` (#263) the default, correct?
1. For the group C++/CMake users, are you suggesting that the users **always** have to manually update their `PYTHONPATH` in order to enable the discovery of the packages? (unless the prefix is the default `/usr/local` I guess)

Considering our distribution approach (robotology-superbuild and its binaries e.g. for windows, in the future maybe conda) I see good reasons to unify all the projects in such a way. It would simplify the maintenance of our distribution strategies. In this moment I don't see anything that could be blocking this change, I'm open to keep discussing.

traversaro, 2021-03-01:
> * When you talk about virtualenvs, do you also include the system's interpreter? What I mean is, in the case without a virtualenv and a custom `CMAKE_INSTALL_PREFIX`, do you suggest to avoid installing the python resources in the system's directories?
> 

Good point. If we agree that we don't want to support `pip install .` outside of virtualenv, I would stick to always install in `Python3_SITELIB`, what do you think? I am not sure how mantainable/feasible/easy is to emulate the installation in the user directory etc etc etc.

> * For gym-ignition, you're suggesting to make the behavior currently optionally enabled with `BINDINGS_INSTALL_PREFIX` (#263) the default, correct?
> 

I will try to answer this two points together. I think we can have two strategies here, as mentioned also in https://github.com/robotology/robotology-superbuild/issues/641#issuecomment-787463121 : 
* (A) A CMake option that permits to specify an absolute path for the bindings installation directory (as `CMAKE_INSTALL_PYTHON3DIR` for YARP, see  https://github.com/robotology/yarp/issues/2511#issuecomment-787429494)
* (B) A CMake option that permits to install directly in [`Python3_SITELIB` as returned by `find_package(Python3)`](https://cmake.org/cmake/help/latest/module/FindPython3.html#result-variables), (as `FRAMEWORK_DETECT_ACTIVE_PYTHON_SITEPACKAGES` for bfl, see https://github.com/dic-iit/bipedal-locomotion-framework/blob/381ad4623ce709ad7966e75412b45814152cf009/bindings/CMakeLists.txt#L7 )

In my opinion the default of any CMake option that is implemented should be such that files are only installed in `CMAKE_INSTALL_PREFIX`, so `CMAKE_INSTALL_PYTHON3DIR` being a relative path computed by `distutils.sysconfig.get_python_lib` for Option A, or `FRAMEWORK_DETECT_ACTIVE_PYTHON_SITEPACKAGES` for Option B . Then, this options will be set appropriately in the setup.py to ensure that the package is installed in `Python3_SITELIB` .

> * For the group C++/CMake users, are you suggesting that the users **always** have to manually update their `PYTHONPATH` in order to enable the discovery of the packages? (unless the prefix is the default `/usr/local` I guess)

Yes, that is what is  exactly what happens for native executable with PATH or with C++ libraries with CMAKE_PREFIX_PATH or PKG_CONFIG_PATH, so it seems consistent. Note that unfortunatly on Ubuntu the default `/usr/local`  will not work out of the box due to https://bugs.launchpad.net/ubuntu/+source/python3-stdlib-extensions/+bug/1832215, see https://github.com/robotology/yarp/issues/2511 for a related discussion.

traversaro, 2021-03-01:
>     2\. PEP600 support (tooling is almost ready, it's matter of weeks)
> 
> 
> The second is more about binary distribution, and it has a lower priority. It can also be kind of replaced by the recent effort of @traversaro to bring robotology projects in a custom conda channel.

Just FYI that kind of work is meant to permit selected projects (such as iDynTree) to be packaged also in conda-forge, and use those one in the robotology channel (that is what we already do for osqp, see https://github.com/traversaro/robotology-superbuild/blob/master/conda/cmake/CondaGenerationOptions.cmake#L21 ).

diegoferigo, 2021-03-01:
> Good point. If we agree that we don't want to support `pip install .` outside of virtualenv, I would stick to always install in `Python3_SITELIB`, what do you think? I am not sure how [...]

We should highly discourage using `sudo pip`. I think that in most cases there's already a big red warning when one attempts doing that. In fact, when Python/venv users call `pip install .` and there's no active venv, the package is installed in `$HOME/.local`, and this is ok. In my comment, even if it was not clear, I meant for C++/CMake users. 

So, by default, also in this case, users have to update their `PYTHONPATH`. If this is the case, since we're finding a common approach that could be used by many projects, I would suggest to expose a CMake's cache variable `<project>_USE_PYTHON_SYSTEM_INSTALL_PREFIX` that populates `<project>_PYTHON_SYSTEM_INSTALL_PREFIX` (allowing users to manually specify it if misdetected).

> > * For gym-ignition, you're suggesting to make the behavior currently optionally enabled with `BINDINGS_INSTALL_PREFIX` (#263) the default, correct?
> 
> I will try to answer this two points together. I think we can have two strategies here, as mentioned also in [robotology/robotology-superbuild#641 (comment)](https://github.com/robotology/robotology-superbuild/issues/641#issuecomment-787463121) :
> 
>     * (A) A CMake option that permits to specify an absolute path for the bindings installation directory (as `CMAKE_INSTALL_PYTHON3DIR` for YARP, see  [robotology/yarp#2511 (comment)](https://github.com/robotology/yarp/issues/2511#issuecomment-787429494))
> 
>     * (B) A CMake option that permits to install directly in [`Python3_SITELIB` as returned by `find_package(Python3)`](https://cmake.org/cmake/help/latest/module/FindPython3.html#result-variables), (as `FRAMEWORK_DETECT_ACTIVE_PYTHON_SITEPACKAGES` for bfl, see https://github.com/dic-iit/bipedal-locomotion-framework/blob/381ad4623ce709ad7966e75412b45814152cf009/bindings/CMakeLists.txt#L7 )
> 
> 
> In my opinion the default of any CMake option that is implemented should be such that files are only installed in `CMAKE_INSTALL_PREFIX`, so `CMAKE_INSTALL_PYTHON3DIR` being a relative path computed by `distutils.sysconfig.get_python_lib` for Option A, or `FRAMEWORK_DETECT_ACTIVE_PYTHON_SITEPACKAGES` for Option B . Then, this options will be set appropriately in the setup.py to ensure that the package is installed in `Python3_SITELIB` .

Ok I replied the above without reading all your previous comment, I think we reached the same conclusion even if we named the variables differently. I think it could be useful producing a gist or similar with a draft of the guidelines, I have to do an extensive mental recap every time I reply in order to consider of all the cases we want to cover xD 

> > * For the group C++/CMake users, are you suggesting that the users **always** have to manually update their `PYTHONPATH` in order to enable the discovery of the packages? (unless the prefix is the default `/usr/local` I guess)
> 
> Yes, that is what is exactly what happens for native executable with PATH or with C++ libraries with CMAKE_PREFIX_PATH or PKG_CONFIG_PATH, so it seems consistent. Note that unfortunatly on Ubuntu the default `/usr/local` will not work out of the box due to https://bugs.launchpad.net/ubuntu/+source/python3-stdlib-extensions/+bug/1832215, see [robotology/yarp#2511](https://github.com/robotology/yarp/issues/2511) for a related discussion.

That's ok. If users like me that for developing reason 1) don't install the package with pip and 2) do want a CMake install tree that does not need any system configuration (should also be the case of https://github.com/robotology/yarp/issues/2511, bug excluded), they can enable the `<project>_USE_PYTHON_SYSTEM_INSTALL_PREFIX` or `FRAMEWORK_DETECT_ACTIVE_PYTHON_SITEPACKAGES` variables, depending what we'll pick.

PR #263 (2020-11-02): Update CMake project
- Cleans CMake after gympp removal #261.
- Closes #244. Note that installing the bindings in a custom location does not show by default in ccmake, you have to manually pass `-DBINDINGS_INSTALL_PREFIX=/path/to/folder/scenario` and only then it is shown in ccmake.
- Update to CMake 3.16 since it's the version currently shipped in Ubuntu Focal