robotology-legacy/gym-ignition #247: reset_base_world_angular_velocity and reset_base_world_linear_velocity not equivalent to reset_base_world_linear_velocity

traversaro, 2020-09-27:
If I run the following script (commit from `test/rbd`) branch:
~~~
import numpy as np
from gym_ignition import utils
import idyntree.bindings as idt
from gym_ignition.rbd import idyntree
from scenario import gazebo as scenario_gazebo
from gym_ignition.utils.scenario import init_gazebo_sim

# Configure numpy output
np.set_printoptions(precision=10, suppress=True, floatmode='fixed')

scenario_gazebo.set_verbosity(scenario_gazebo.Verbosity_info)

urdf = f"""
<robot name="oneLink">
    <link name="base_link">
        <inertial>
            <origin xyz="0 0 0.0" rpy="0 0 0"/>
            <mass value="1"/>
            <inertia ixx="2e-5" ixy="0" ixz="0" iyy="2e-5" iyz="0" izz="2e-5"/>
        </inertial>
    </link>
</robot>
"""
urdf_file = utils.misc.string_to_file(urdf)

# Get the simulator and the world
step_size=1e-6;
print(step_size)
gazebo, world = init_gazebo_sim(step_size=step_size)
assert world.set_gravity([0, 0, 0])

# Show the GUI
# import time
# gazebo.gui()
# gazebo.run(paused=True)
# time.sleep(3)

# Insert the model
world.insert_model(urdf_file)
gazebo.run(paused=True)
model = world.get_model(model_name="oneLink").to_gazebo()

# Reset the state of the model
model.to_gazebo().reset_base_pose([0, 0, 0], [1, 0, 0, 0])
model.to_gazebo().reset_base_world_angular_velocity([0.0, 0, 2.0])
model.to_gazebo().reset_base_world_linear_velocity([0.0, 0, 1.0])

gazebo.run(paused=True)

kindyn = idyntree.kindyncomputations.KinDynComputations(model_file=urdf_file)

# ~~> t = k
# ---------

kindyn.set_robot_state_from_model(model=model, world_gravity=np.array(world.gravity()))

nu_k = kindyn.get_model_velocity()
print("nu_k", nu_k)
~~~

It prints:
~~~
nu_k   [0.0000000000 0.0000000000 1.0000000000 0.0000000000 0.0000000000 0.0000000000]
~~~

So it seems that the `reset_base_world_angular_velocity` call is ignored. Inverting the order of the `reset_base_world_angular_velocity` and `reset_base_world_linear_velocity` instead the `reset_base_world_linear_velocity` is the one that is ignored.

diegoferigo, 2020-09-27:
Interesting, logic defined in:

https://github.com/robotology/gym-ignition/blob/0a3d86575940cb222ee405eab9184c8aa3eb1987/cpp/scenario/gazebo/src/Model.cpp#L291-L301

diegoferigo, 2020-09-27:
Ok I know what's happening. The methods `baseWorld{Linear,Angular}Velocity()` read the value of the simulation of the previous run. When you call the `set*` methods, you are not overriding these values that are the components used to extract data from the simulation. You instead create a new `*Cmd` component that will be processed in the next step:

https://github.com/robotology/gym-ignition/blob/0a3d86575940cb222ee405eab9184c8aa3eb1987/cpp/scenario/gazebo/src/Model.cpp#L332-L334

We have two ways to solve this issue:

1. Document that if you want to reset both linear and angular velocity, you need to use the method that sets together both.
2. Improve the logic and check if a `*Cmd` component already exists and use that instead of calling `this->baseWorld*Velocity()`.

I tend towards 2 since it's more user-friendly and does what they expect :)

closed 2020-09-27 by diegoferigo