robotology-legacy/gym-ignition #279: Test conda installation

diegoferigo, 2020-12-10:
Using conda and particularly the `conda-forge` channel could introduce many benefits to our project:

- Possibly prevent [the problems](https://github.com/robotology/gym-ignition/blob/63015c33a1/docs/sphinx/info/faq.rst#interaction-with-tensorflow) of the interaction between protobuf / tensorflow and ign-gazebo
- Provide a cross-platform installation that uses binary packages
- Enable the creation of development environments that can be precisely reproduced in other machines and OSs

About 1), for the sake of testing, manually compiling ign-gazebo with colcon and using the conda environment for the dependencies should suffice.

On conda-forge there are already many of the ign-gazebo dependencies, but few of them are still missing. As soon as the entire ignition distribution will be included in the channel, we'll be able to bypass the manual installation of ign-gazebo from Open Robotics' PPA.

diegoferigo, 2021-02-07:
I am giving conda a try. I did the following in a plain `ubuntu:focal` container:

1. Created a conda environment with micromamba
1. Added the conda-forge channel with strict priority
1. Install buildchain (compilers, cmake, ...)
1. Install dart from sources (I need this for other reasons, this shouldn't affect the following)
1. Install OpenGL stack with `mesa-libgl-devel-cos7-x86_64 libglu libxrandr-devel-cos7-x86_64 xorg-libxrandr`
1. Install ogre 1.9 from sources installing into the conda environment (the ogre package seems too old in conda)
1. Install Ignition Dome from colcon (after the above, all projects build correctly)
1. Build idyntree from sources and installing into the conda environment
1. Build gym-ignition from sources and installing into the conda environment
1. Install `tensorflow-gpu`

After these steps, I hoped that the tensorflow problems related to protobuf would be solved. Sadly, they are not. If tensorflow is imported after scenario, tensorflow will segfault.

<details>
<summary>segfault</summary>

```
$ PYTHONFAULTHANDLER=1 python
Python 3.8.0 | packaged by conda-forge | (default, Nov 22 2019, 19:11:38)                                                                   
[GCC 7.3.0] :: Anaconda, Inc. on linux                                                                                                      
Type "help", "copyright", "credits" or "license" for more information.                                                                      
>>> import faulthandler; faulthandler.enable()                                                                                              
>>> from scenario import gazebo                                                                                                             
>>> import tensorflow                                                                                                                       
Fatal Python error: Segmentation fault

Current thread 0x00007f4cc2ddb740 (most recent call first):
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1101 in create_module
  File "<frozen importlib._bootstrap>", line 556 in module_from_spec
  File "<frozen importlib._bootstrap>", line 657 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 702 in _load
  File "/conda/lib/python3.8/imp.py", line 342 in load_dynamic
  File "/conda/lib/python3.8/imp.py", line 242 in load_module
  File "/conda/lib/python3.8/site-packages/tensorflow/python/pywrap_tensorflow_internal.py", line 24 in swig_import_helper
  File "/conda/lib/python3.8/site-packages/tensorflow/python/pywrap_tensorflow_internal.py", line 28 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 783 in exec_module
  File "<frozen importlib._bootstrap>", line 671 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 975 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 991 in _find_and_load
  File "/conda/lib/python3.8/site-packages/tensorflow/python/pywrap_tensorflow.py", line 58 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 783 in exec_module
  File "<frozen importlib._bootstrap>", line 671 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 975 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 991 in _find_and_load
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1042 in _handle_fromlist
  File "/conda/lib/python3.8/site-packages/tensorflow/python/__init__.py", line 50 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 783 in exec_module
  File "<frozen importlib._bootstrap>", line 671 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 975 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 991 in _find_and_load
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 961 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 991 in _find_and_load
  File "/conda/lib/python3.8/site-packages/tensorflow/__init__.py", line 41 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 783 in exec_module
  File "<frozen importlib._bootstrap>", line 671 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 975 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 991 in _find_and_load
  File "<stdin>", line 1 in <module>
Segmentation fault (core dumped)
```

</details>

The problem is the `libtensorflow_framework.so.2` library that can be found in the `tensorflow` python package. Loading only that library with `ctypes("/path/to/libtensorflow_framework.so.2")` results to the segfault.

At this point, I tried to avoid altering the dlopen flags when loading `scenario.bindings.gazebo` (done [here](https://github.com/robotology/gym-ignition/blob/321df6a2f93da829682ead2f428022ec1c4ef416/bindings/__init__.py#L54-L63)) and the import situation seems solved. However, we need this tweak to make singletons works (unless https://github.com/ignitionrobotics/ign-gazebo/issues/248 get implemented somehow), so this is a solution we cannot use. Even though the dlopen flags are restored after the import, tensorflow doesn't like it.

I think at this point the best workaround that fixed both conda and non-conda environments is detecting if tensorflow is installed and, if found, open the problematic shared library, similarly to https://github.com/apache/arrow/pull/2210. 

cc @traversaro

traversaro, 2021-02-07:
> Install ogre 1.9 from sources installing into the conda environment (the ogre package seems too old in conda)

Strange, conda-forge should contain ogre version that works fine with ignition-rendering, see https://github.com/conda-forge/staged-recipes/pull/13677 .

> Install `tensorflow-gpu`

How did you install `tensorflow-gpu` ? If via the wheel installed by pip, then the problem with the vendored protobuf I guess should be the same that you have when installing gym-ignition dependencies via apt. The `protobuf` problem would be avoided it a tensorflow-gpu build (that used the conda-forge protobuf) was available in conda-forge, but unfortunatly tensorflow for several reasons is quite complicated to package (see https://github.com/conda-forge/tensorflow-feedstock/pull/110, https://github.com/conda-forge/tensorflow-feedstock/issues/107 and https://github.com/conda-forge/tensorflow-feedstock/issues/10).

diegoferigo, 2021-02-07:
> > Install ogre 1.9 from sources installing into the conda environment (the ogre package seems too old in conda)
> 
> Strange, conda-forge should contain ogre version that works fine with ignition-rendering, see [conda-forge/staged-recipes#13677](https://github.com/conda-forge/staged-recipes/pull/13677) .

That is what I thought. Maybe there was something strange in my setup (even though it is a plain docker image), but I got compilation errors. I didn't save the log, my fault, but I remember some failed checks to `nullptr` of ogre pointers. I didn't investigate much, I though it was just an outdated version of ogre, and moved towards a source compilation of the 1.9 version.

> 
> > Install `tensorflow-gpu`
> 
> How did you install `tensorflow-gpu` ? If via the wheel installed by pip, then the problem with the vendored protobuf I guess should be the same that you have when installing gym-ignition dependencies via apt. The `protobuf` problem would be avoided it a tensorflow-gpu build (that used the conda-forge protobuf) was available in conda-forge, but unfortunatly tensorflow for several reasons is quite complicated to package (see [conda-forge/tensorflow-feedstock#110](https://github.com/conda-forge/tensorflow-feedstock/pull/110), [conda-forge/tensorflow-feedstock#107](https://github.com/conda-forge/tensorflow-feedstock/issues/107) and [conda-forge/tensorflow-feedstock#10](https://github.com/conda-forge/tensorflow-feedstock/issues/10)).

I installed the `tensorflow-gpu` conda package (not from the conda-forge channel). I followed one of the issues you posted (particularly https://github.com/conda-forge/tensorflow-feedstock/issues/10) where people reported that the package from the default channel was working fine. I tried on a fresh container and indeed I've got tf with GPU support.

Now that you make me think about it, I originally thought that this `tensorflow` package was built against the conda-forge's protobuf, but actually since it's not coming from the conda-forge channel maybe there's still a mismatch. However, I also did try to dig into the various `.so` files of the tensorflow package but I didn't yet find the one that links against protobuf using `ldd`.

traversaro, 2021-02-07:
> I installed the `tensorflow-gpu` conda package (not from the conda-forge channel). I followed one of the issues you posted (particularly [conda-forge/tensorflow-feedstock#10](https://github.com/conda-forge/tensorflow-feedstock/issues/10)) where people reported that the [package from the default channel](https://anaconda.org/anaconda/tensorflow-gpu) was working fine. I tried on a fresh container and indeed I've got tf with GPU support.

Interesting. I think that the tensorflow from the defaults channel is build from the recipes in https://github.com/AnacondaRecipes/tensorflow_recipes . I think it is probably building against the defaults channel protobuf, built from https://github.com/AnacondaRecipes/protobuf-feedstock . However, to double check it would be interesting to understand which packages are actually installed in your environment with `conda list` .

diegoferigo, 2021-02-07:
I didn't reach yet any conclusion, let's see when I find some more time to continue this experiment. Here below the current list of packages. I have some cos6 and cos7 duplicates, but I guess I'm using the cos7.

<details>
<summary>mamba list</summary>

```

                  __    __    __    __
                 /  \  /  \  /  \  /  \
                /    \/    \/    \/    \
███████████████/  /██/  /██/  /██/  /████████████████████████
              /  / \   / \   / \   / \  \____
             /  /   \_/   \_/   \_/   \    o \__,
            / _/                       \_____/  `
            |/
        ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗
        ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗
        ██╔████╔██║███████║██╔████╔██║██████╔╝███████║
        ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║
        ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║
        ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝

        mamba (0.7.12) supported by @QuantStack

        GitHub:  https://github.com/mamba-org/mamba
        Twitter: https://twitter.com/QuantStack

█████████████████████████████████████████████████████████████

# packages in environment at /conda:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
_sysroot_linux-64_curr_repodata_hack 3                   ha9d2b57_10    conda-forge
_tflow_select             2.1.0                       gpu  
absl-py                   0.11.0           py38h578d9bd_0    conda-forge
ace                       7.0.0                h9c3ff4c_1    conda-forge
alsa-lib                  1.2.3                h516909a_0    conda-forge
ampl-mp                   3.1.0             h616b090_1004    conda-forge
asio                      1.14.1               h9c3ff4c_1    conda-forge
assimp                    5.0.1                hedfc422_5    conda-forge
astor                     0.8.1              pyh9f0ad1d_0    conda-forge
astunparse                1.6.3              pyhd8ed1ab_0    conda-forge
attrs                     20.3.0             pyhd3deb0d_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.1                      py_0    conda-forge
binutils                  2.35.1               hdd6e379_2    conda-forge
binutils_impl_linux-64    2.35.1               h193b22a_2    conda-forge
binutils_linux-64         2.35                hc3fd857_29    conda-forge
blinker                   1.4                        py_1    conda-forge
boost                     1.74.0           py38hc10631b_3    conda-forge
boost-cpp                 1.74.0               h9359b55_0    conda-forge
brotlipy                  0.7.0           py38h497a2fe_1001    conda-forge
bullet                    2.89             py38h51da96c_2    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.17.1               h36c2ea0_0    conda-forge
c-compiler                1.1.3                h7f98852_0    conda-forge
ca-certificates           2020.12.5            ha878542_0    conda-forge
cachetools                4.2.1              pyhd8ed1ab_0    conda-forge
cairo                     1.16.0            h488836b_1006    conda-forge
catkin_pkg                0.4.23             pyh9f0ad1d_0    conda-forge
certifi                   2020.12.5        py38h578d9bd_1    conda-forge
cffi                      1.14.4           py38ha312104_0    conda-forge
chardet                   4.0.0            py38h578d9bd_1    conda-forge
click                     7.1.2              pyh9f0ad1d_0    conda-forge
cloudpickle               1.6.0                      py_0    conda-forge
cmake                     3.19.4               h3020d66_1    conda-forge
colcon-cmake              0.2.26                     py_0    conda-forge
[... 295 lines cut ...]
xorg-libxrandr            1.5.2                h516909a_1    conda-forge
xorg-libxrender           0.9.10            h516909a_1002    conda-forge
xorg-libxt                1.1.5             h516909a_1003    conda-forge
xorg-randrproto           1.5.0             h516909a_1001    conda-forge
xorg-renderproto          0.11.1            h14c3975_1002    conda-forge
xorg-xextproto            7.3.0             h14c3975_1002    conda-forge
xorg-xproto               7.0.31            h14c3975_1007    conda-forge
xz                        5.2.5                h516909a_1    conda-forge
yaml                      0.2.5                h516909a_0    conda-forge
zeromq                    4.3.4                h9c3ff4c_0    conda-forge
zipp                      3.4.0                      py_0    conda-forge
zlib                      1.2.11            h516909a_1010    conda-forge
zstd                      1.4.8                ha95c52a_1    conda-forge
zziplib                   0.13.69              hed695b0_1    conda-forge
```
</details>

In any case, so far this was a kind of successful experiment. Despite the few hacks, I managed to compile:

1. Complete colcon Dome environment against conda-forge packages + dart from sources (could be maybe avoided) + ogre from sources.
1. robotology-superbuild that includes idyntree (also python), blf (also python) and casadi (I enabled python, qpoases and osqp support). Note that conda-forge's casadi only has the ipopt interface enabled (https://github.com/conda-forge/casadi-feedstock/pull/22).
1. tensorflow with gpu support.
1. gym-ignition.

I also need ray and rllib, however I don't think it would be a problem. In earlier experiments I tried to install it from conda-forge together with tensorflow from conda's default channel and they seemed working fine.

diegoferigo, 2021-02-08:
I'm trying to prepare a docker image containing what I've got so far, but now I'm stuck with the following error. Comparing the conda package inside the docker image and what posted in the comment above did not show anything relevant.

<details>
<summary>ignition-rendering4 compilation error</summary>

```
(/conda) root@ff903e9f5fd3:/usr/local/src/workspace/build/ignition-rendering4# ninja                                                                                       
[3/5] Building CXX object ogre/src/CMakeFiles/ignition-rendering4-ogre.dir/OgreRenderEngine.cc.o                                                                           
FAILED: ogre/src/CMakeFiles/ignition-rendering4-ogre.dir/OgreRenderEngine.cc.o                                                                                             
/conda/bin/x86_64-conda-linux-gnu-c++ -Dignition_rendering4_ogre_EXPORTS -DIGN_RENDERING_RESOURCE_PATH=\"/usr/local/src/workspace/install/share/ignition/ignition-rendering
4\" -DOGRE_RESOURCE_PATH=\"/conda/lib/OGRE\" -I/usr/local/src/workspace/src/ign-rendering/include -Iinclude -I/usr/local/src/workspace/src/ign-rendering/ogre/include -Iogr
e/include -Icore/include -isystem /usr/local/src/workspace/install/include/ignition/common3 -isystem /conda/include/uuid -isystem /usr/local/src/workspace/install/include/
ignition/math6 -isystem /usr/local/src/workspace/install/include/ignition/plugin1 -isystem /usr/local/src/workspace/install/include/ignition/cmake2 -isystem /conda/include
/OGRE -isystem /conda/include/OGRE/RenderSystems/GL -isystem /conda/include/OGRE/Paging -isystem /conda/include/OGRE/RTShaderSystem -isystem /conda/include/OGRE/Terrain -i
system /conda/include/OGRE/Overlay -fvisibility-inlines-hidden -std=c++17 -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -
fno-plt -O2 -ffunction-sections -pipe -isystem /conda/include -O2 -g -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-d
efault -Wmissing-include-dirs -pedantic -UNDEBUG -fPIC -msse -msse2 -mfpmath=sse -msse3 -mssse3 -msse4.1 -msse4.2 -fPIC -I/conda/include/uuid -pthread -I/conda/include/OGR
E/RTShaderSystem -I/conda/include -I/conda/include/OGRE -I/conda/include/OGRE/Terrain -I/conda/include/OGRE/Paging -I/conda/include/OGRE/Overlay -MD -MT ogre/src/CMakeFile
s/ignition-rendering4-ogre.dir/OgreRenderEngine.cc.o -MF ogre/src/CMakeFiles/ignition-rendering4-ogre.dir/OgreRenderEngine.cc.o.d -o ogre/src/CMakeFiles/ignition-rendering
4-ogre.dir/OgreRenderEngine.cc.o -c /usr/local/src/workspace/src/ign-rendering/ogre/src/OgreRenderEngine.cc

In file included from /conda/x86_64-conda-linux-gnu/sysroot/usr/include/GL/glx.h:333,                                                                                      
                 from /usr/local/src/workspace/src/ign-rendering/ogre/src/OgreRenderEngine.cc:22:                                                                          
/conda/x86_64-conda-linux-gnu/sysroot/usr/include/GL/glxext.h:522:143: error: 'GLintptr' has not been declared                                                             
  522 | typedef void ( *PFNGLXCOPYBUFFERSUBDATANVPROC) (Display *dpy, GLXContext readCtx, GLXContext writeCtx, GLenum readTarget, GLenum writeTarget, GLintptr readOffset, 
GLintptr writeOffset, GLsizeiptr size);                                                                                                                                    
      |                                                                                                                                               ^~~~~~~~             
/conda/x86_64-conda-linux-gnu/sysroot/usr/include/GL/glxext.h:522:164: error: 'GLintptr' has not been declared                                                             
  522 | typedef void ( *PFNGLXCOPYBUFFERSUBDATANVPROC) (Display *dpy, GLXContext readCtx, GLXContext writeCtx, GLenum readTarget, GLenum writeTarget, GLintptr readOffset, 
GLintptr writeOffset, GLsizeiptr size);                                                                                                                                    
      |                                                                                                                                                                    
^~~~~~~~                                                                                                                                                                   
/conda/x86_64-conda-linux-gnu/sysroot/usr/include/GL/glxext.h:522:186: error: 'GLsizeiptr' has not been declared                                                           
  522 | typedef void ( *PFNGLXCOPYBUFFERSUBDATANVPROC) (Display *dpy, GLXContext readCtx, GLXContext writeCtx, GLenum readTarget, GLenum writeTarget, GLintptr readOffset, GLintptr writeOffset, GLsizeiptr size);                                                                                                                                    
      |                                                                                                                                                                    
                      ^~~~~~~~~~                                                                                                                                           
[... cut ...]
In file included from /usr/local/src/workspace/src/ign-rendering/ogre/include/ignition/rendering/ogre/OgreIncludes.hh:44,
                 from /usr/local/src/workspace/src/ign-rendering/ogre/src/OgreRenderEngine.cc:40:
/conda/include/OGRE/OgreVector3.h:2:62: note: #pragma message: /conda/include/OGRE/OgreVector3.h is deprecated, migrate to Ogre.h
    2 | #pragma message( __FILE__ " is deprecated, migrate to Ogre.h")
      |                                                              ^
ninja: build stopped: subcommand failed.
```

</details>

diegoferigo, 2021-02-08:
Weird, it seems that commenting out the include of `/conda/x86_64-conda-linux-gnu/sysroot/usr/include/GL/glx.h:333` solves the problem. The compilation succeeds. Here below the snippet:

```c
#ifndef GLX_GLXEXT_LEGACY                                                                                                                                                  

#include <GL/glxext.h>  /* This is line 333 */

#endif /* GLX_GLXEXT_LEGACY */
```

traversaro, 2021-02-08:
Related discussion in the ignition-rendering recipe PR : https://github.com/conda-forge/staged-recipes/pull/13677#issuecomment-760006547 .

diegoferigo, 2021-02-08:
Thanks for the link, the problem is the same. I didn't get if a solution was found.

In the meantime, I paste here the Dockerfile I got. I'll commit it somewhere, not sure yet where.

<details>
<summary>Dockerfile</summary>

```dockerfile
ARG from=nvidia/cuda:11.1.1-cudnn8-runtime-ubuntu20.04
FROM ${from}

# Change default shell to bash. This is effective only in the Dockerfile.
SHELL ["/bin/bash", "-i", "-c"]

# Setup locales and timezone
ARG TZ=Europe/Rome
ARG DEBIAN_FRONTEND=noninteractive
RUN rm -f /etc/localtime &&\
    ln -s /usr/share/zoneinfo/"${TZ}" /etc/localtime &&\
    apt-get update &&\
    apt-get install -y --no-install-recommends locales locales-all tzdata &&\
    rm -rf /var/lib/apt/lists/*

# Utilities
RUN apt-get update &&\
    apt-get install -y --no-install-recommends \
        software-properties-common \
        apt-transport-https \
        apt-utils \
        wget \
        nano \
        bash-completion \
        gnupg2 \
        colordiff \
        curl \
        zip \
        unzip \
        lsof \
        net-tools \
        iputils-ping \
        strace \
        less \
        &&\
    rm -rf /var/lib/apt/lists/*

ENV CONDA_PREFIX=/conda
ARG CONDA_PYTHON_VERSION=3.8
ENV MAMBA_ROOT_PREFIX=$CONDA_PREFIX/.mamba

# Install micromamba and create conda environment
RUN cd /usr/local &&\
    wget -qO- https://micromamba.snakepit.net/api/micromamba/linux-64/latest \
        | tar -xvj bin/micromamba &&\
    eval "$(micromamba shell hook -s bash)" &&\    
    micromamba create -y -p $CONDA_PREFIX python==$CONDA_PYTHON_VERSION mamba -c conda-forge &&\
    micromamba activate $CONDA_PREFIX &&\
    conda config --system --add channels conda-forge &&\
    conda config --system --set channel_priority strict

# Enable by default the conda environment for non-root users
RUN echo 'function activate_conda() {' >> /etc/bash.bashrc &&\
    echo '  eval "$(micromamba shell hook -s bash)"' >> /etc/bash.bashrc &&\
    echo '  micromamba activate $CONDA_PREFIX' >> /etc/bash.bashrc &&\
    #echo '  export CMAKE_PREFIX_PATH=$CONDA_PREFIX:$CMAKE_PREFIX_PATH' >> /etc/bash.bashrc &&\
    echo '}' >> /etc/bash.bashrc &&\
    echo '[[ $(id -u) -gt 0 ]] && activate_conda' >> /etc/bash.bashrc

# Install buildchain
RUN activate_conda &&\
    mamba install -y \
[... 126 lines cut ...]

RUN activate_conda &&\
    mamba install -y \
        ipython \
        tensorflow-gpu \
        scipy \
        numpy \
        gym \
        lxml \
        matplotlib \
        setuptools_scm \
        ray-rllib ray-tune ray-dashboard
RUN activate_conda &&\
    cd $SRC_DIR &&\
    git clone https://github.com/robotology/gym-ignition &&\
    mkdir gym-ignition/build && cd gym-ignition/build &&\
    cmake .. \
        -GNinja \
        -DCMAKE_BUILD_TYPE=Release \
        -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX \
        &&\
    ninja install
```

</details>

traversaro, 2021-02-08:
> Thanks for the link, the problem is the same. I didn't get if a solution was found.

No, it was just passed the `-DGLX_GLXEXT_LEGACY` flag to the compiler: https://github.com/conda-forge/libignition-rendering4-feedstock/blob/6f273d6430be1812e2037bb6e976b36aefd5f9ca/recipe/build.sh#L6 .

diegoferigo, 2021-02-09:
I managed to reproduce the ogre compilation error I reported above in :

This error occurs when `ogre==1.10.12` is installed in the conda workspace. Updating ogre to `1.12.10` (the only other available version in conda-forge) does not create problems while building `ign-rendering4`, but I get the error reported in https://github.com/osrf/gazebo/issues/2700 when opening the gui installed by `igni-gui4` (from colcon).

The only alternative I didn't yet check is compiling and installing ogre 1.9 from sources. This should provide the same working configuration with ogre1 of the ignition packages provided by in the osrf's ppa.

<details>
<summary>ogre compilation error</summary>

```
workspace/build/ignition-rendering4 via v3.19.4 via 🅒 /conda 
✦  ❯ ninja
[3/19] Building CXX object ogre/src/CMakeFiles/ignition-rendering4-ogre.dir/OgreMaterialSwitcher.cc.o
FAILED: ogre/src/CMakeFiles/ignition-rendering4-ogre.dir/OgreMaterialSwitcher.cc.o 
/conda/bin/x86_64-conda-linux-gnu-c++ -DOGRE_VERSION_LT_1_12_0 -Dignition_rendering4_ogre_EXPORTS -I/usr/local/src/workspace/src/ign-rendering/include -Iinclude -I/usr/local/src/workspace/src/ign-rendering/ogre/include -Iogre/include -Icore/include -isystem /usr/local/src/workspace/install/include/ignition/common3 -isystem /conda/include/uuid -isystem /usr/local/src/workspace/install/include/ignition/math6 -isystem /usr/local/src/workspace/install/include/ignition/plugin1 -isystem /usr/local/src/workspace/install/include/ignition/cmake2 -isystem /conda/include/OGRE -isystem /conda/include/OGRE/RenderSystems/GL -isystem /conda/include/OGRE/RTShaderSystem -isystem /conda/include/OGRE/Terrain -isystem /conda/include/OGRE/Paging -isystem /conda/include/OGRE/Overlay -fvisibility-inlines-hidden -std=c++17 -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /conda/include -O3 -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic  -fPIC -msse -msse2 -mfpmath=sse -msse3 -mssse3 -msse4.1 -msse4.2 -fPIC -I/conda/include/uuid -pthread -I/conda/include/OGRE/RTShaderSystem -I/conda/include -I/conda/include/OGRE -I/conda/include/OGRE/Terrain -I/conda/include/OGRE/Paging -I/conda/include/OGRE/Overlay -MD -MT ogre/src/CMakeFiles/ignition-rendering4-ogre.dir/OgreMaterialSwitcher.cc.o -MF ogre/src/CMakeFiles/ignition-rendering4-ogre.dir/OgreMaterialSwitcher.cc.o.d -o ogre/src/CMakeFiles/ignition-rendering4-ogre.dir/OgreMaterialSwitcher.cc.o -c /usr/local/src/workspace/src/ign-rendering/ogre/src/OgreMaterialSwitcher.cc
/usr/local/src/workspace/src/ign-rendering/ogre/src/OgreMaterialSwitcher.cc: In member function 'virtual Ogre::Technique* ignition::rendering::v4::OgreMaterialSwitcher::handleSchemeNotFound(uint16_t, const String&, Ogre::Material*, uint16_t, const Ogre::Renderable*)':
/usr/local/src/workspace/src/ign-rendering/ogre/src/OgreMaterialSwitcher.cc:86:53: error: no matching function for call to 'static_pointer_cast<Ogre::Material>(Ogre::ResourcePtr&)'
   86 |         std::static_pointer_cast<Ogre::Material>(res);
      |                                                     ^
In file included from /conda/x86_64-conda-linux-gnu/include/c++/9.3.0/bits/shared_ptr.h:52,
                 from /conda/x86_64-conda-linux-gnu/include/c++/9.3.0/memory:81,
                 from /usr/local/src/workspace/install/include/ignition/common3/ignition/common/Util.hh:21,
                 from /usr/local/src/workspace/install/include/ignition/common3/ignition/common/Console.hh:25,
                 from /usr/local/src/workspace/src/ign-rendering/ogre/src/OgreMaterialSwitcher.cc:18:
/conda/x86_64-conda-linux-gnu/include/c++/9.3.0/bits/shared_ptr_base.h:1532:5: note: candidate: 'template<class _Tp, class _Tp1, __gnu_cxx::_Lock_policy _Lp> std::__shared_ptr<_Tp1, _Lp> std::static_pointer_cast(const std::__shared_ptr<_Tp2, _Lp>&)'
 1532 |     static_pointer_cast(const __shared_ptr<_Tp1, _Lp>& __r) noexcept
      |     ^~~~~~~~~~~~~~~~~~~
/conda/x86_64-conda-linux-gnu/include/c++/9.3.0/bits/shared_ptr_base.h:1532:5: note:   template argument deduction/substitution failed:
/usr/local/src/workspace/src/ign-rendering/ogre/src/OgreMaterialSwitcher.cc:86:53: note:   'Ogre::ResourcePtr' {aka 'Ogre::SharedPtr<Ogre::Resource>'} is not derived from 'const std::__shared_ptr<_Tp2, _Lp>'
   86 |         std::static_pointer_cast<Ogre::Material>(res);
      |                                                     ^
In file included from /conda/x86_64-conda-linux-gnu/include/c++/9.3.0/memory:81,
                 from /usr/local/src/workspace/install/include/ignition/common3/ignition/common/Util.hh:21,
                 from /usr/local/src/workspace/install/include/ignition/common3/ignition/common/Console.hh:25,
                 from /usr/local/src/workspace/src/ign-rendering/ogre/src/OgreMaterialSwitcher.cc:18:
/conda/x86_64-conda-linux-gnu/include/c++/9.3.0/bits/shared_ptr.h:491:5: note: candidate: 'template<class _Tp, class _Up> std::shared_ptr<_Tp> std::static_pointer_cast(const std::shared_ptr<_Tp>&)'
  491 |     static_pointer_cast(const shared_ptr<_Up>& __r) noexcept
      |     ^~~~~~~~~~~~~~~~~~~
/conda/x86_64-conda-linux-gnu/include/c++/9.3.0/bits/shared_ptr.h:491:5: note:   template argument deduction/substitution failed:
[... cut ...]
                 from /usr/local/src/workspace/src/ign-rendering/ogre/include/ignition/rendering/ogre/OgreConversions.hh:26,
                 from /usr/local/src/workspace/src/ign-rendering/ogre/include/ignition/rendering/ogre/OgreDepthCamera.hh:32,
                 from /usr/local/src/workspace/src/ign-rendering/ogre/src/OgreScene.cc:23:
/conda/include/OGRE/OgreRoot.h:392:39: note: declared here
  392 |         OGRE_DEPRECATED SceneManager* createSceneManager(SceneTypeMask typeMask,
      |                                       ^~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
```
</details>

traversaro, 2021-02-09:
> Updating ogre to `1.12.10` (the only other available version in conda-forge) does not create problems while building `ign-rendering4`,

Ok, this make sensei as that ogre is the one used in the ignition-rendering4 official conda build.

> but I get the error reported in osrf/gazebo#2700 when opening the gui installed by igni-gui4 (from colcon).

Did you tried to add `${CONDA_PREFIX}/Media` or `${CONDA_PREFIX}/Media/ShadowVolume` to `OGRE_RESOURCE_PATH` (see https://github.com/ignitionrobotics/ign-rendering/blob/a5f00f91f507cf9cc14134f2fca0f03c4a99a7cc/ogre/src/OgreRenderEngine.cc#L83)?  If that fix is the only necessary, probably we can just add an activation script to `ign-rendering4`  feedstock.

traversaro, 2021-02-09:
> This error occurs when `ogre==1.10.12` is installed in the conda workspace.

I checked a few of the compilation errors, and they seems all to be at points where there is an ifdef for compatibility between 1.9 and 1.12 , such as https://github.com/ignitionrobotics/ign-rendering/blob/ign-rendering4/ogre/src/OgreMaterialSwitcher.cc#L82 . Probably the ifdef logic needs to be change to correctly account for Ogre 1.10.12 .

traversaro, 2021-02-09:
I guess this discussion could be relevant for @JShep1 @wolfv @Tobias-Fischer .

diegoferigo, 2021-02-09:
> > Updating ogre to `1.12.10` (the only other available version in conda-forge) does not create problems while building `ign-rendering4`,
> 
> Ok, this make sensei as that ogre is the one used in the ignition-rendering4 official conda build.

Yep. However as reported above, this ogre version has runtime errors when opening Ignition Gazebo's GUI.

> Did you tried to add `${CONDA_PREFIX}/Media` or `${CONDA_PREFIX}/Media/ShadowVolume` to `OGRE_RESOURCE_PATH` (see [...]

Thanks for the hint, I just tried to compile again ign-rendering4 from colcon against conda-forge's `ogre==1.12.10`. The folder you reported is in `/conda/share/OGRE/Media`. After exporting the environment variable and running the application that opens the Ignition GUI, I get the following error:

```
[GUI] [Err] [OgreRenderEngine.cc:465] Unable to load Ogre Plugin[/conda/share/OGRE/Media/RenderSystem_GL]. Rendering will not be possible.Make sure you have installed OGRE properly.

terminate called after throwing an instance of 'Ogre::RuntimeAssertionException'
  what():  RuntimeAssertionException: Ogre/ShadowExtrudePointLight not found. Verify that you referenced the 'ShadowVolume' folder in your resources.cfg in initialise at /home/conda/feedstock_root/build_artifacts/ogre_1611842210433/work/OgreMain/src/OgreShadowVolumeExtrudeProgram.cpp (line 70)
Aborted (core dumped)
```

The first line is Ignition Rendering 4 that complains about the `RenderSystem_GL` folder that **does not exist** (I guess there is a GL feature in ogre that is not enabled in the package from conda-forge). The second line is an Ogre exception. 

> I checked a few of the compilation errors, and they seems all to be at points where there is an ifdef for compatibility between 1.9 and 1.12 , such as [...]

I might try to alter the compiler definition by hand, let's see if I manage to get something out of this.

traversaro, 2021-02-09:
> I might try to alter the compiler definition by hand, let's see if I manage to get something out of this.

The related CMake code is https://github.com/ignitionrobotics/ign-rendering/blob/ign-rendering4/ogre/src/CMakeLists.txt#L18 .

Tobias-Fischer, 2021-02-09:
Hi @diegoferigo, could you please report on the precise version of ogre that is installed in your conda environment? We fixed the bug about the missing `RenderSystem_GL` one year ago (see https://github.com/conda-forge/ogre-feedstock/issues/15), so I fear for some reason an outdated version is installed.

traversaro, 2021-02-09:
> The first line is Ignition Rendering 4 that complains about the RenderSystem_GL folder that does not exist (I guess there is a GL feature in ogre that is not enabled in the package from conda-forge). The second line is an Ogre exception.

No, actually I think I misunderstood the role of `OGRE_RESOURCE_PATH`  . `OGRE_RESOURCE_PATH`  apparently is used to search for OGRE plugins, and those are in `${CONDA_PREFIX}/lib/OGRE` . We should check if there is any other env variable to inject directory in the ogre resource locator, similar to what `GAZEBO_RESOURCE_PATH`  was in Classic Gazebo (see https://github.com/osrf/gazebo/issues/2700#issuecomment-616319253).

diegoferigo, 2021-02-09:
> Hi @diegoferigo, could you please report on the precise version of ogre that is installed in your conda environment? We fixed the bug about the missing `RenderSystem_GL` one year ago (see [conda-forge/ogre-feedstock#15](https://github.com/conda-forge/ogre-feedstock/issues/15)), so I fear for some reason an outdated version is installed.

For this test I was using `ogre==1.12.10`, but as @traversaro it was caused by a wrong usage of `OGRE_RESOURCE_PATH`. I see `RenderSystem_GL*.so` files in `/conda/lib/OGRE`, therefore the package seems ok.

> > I might try to alter the compiler definition by hand, let's see if I manage to get something out of this.
> 
> The related CMake code is https://github.com/ignitionrobotics/ign-rendering/blob/ign-rendering4/ogre/src/CMakeLists.txt#L18 .

Interestingly, using `ogre==1.10.12` and patching ign-rendering4 as follows, I get a working ogre1-based rendering stack :tada: 

```bash
sed -i "s|if(OGRE_VERSION VERSION_LESS 1.10.3)|if(OGRE_VERSION VERSION_LESS 1.11.0)|g" -i src/ign-rendering/ogre/src/CMakeLists.txt
sed -i "s|if(OGRE_VERSION VERSION_LESS 1.10.1)|if(OGRE_VERSION VERSION_LESS 1.11.0)|g" -i src/ign-rendering/ogre/src/CMakeLists.txt
```

<details>
<summary>ignition gui</summary>

[image]

</details>

diegoferigo, 2021-03-23:
It seems that the patch was applied also to the conda-forge ign-rendering recipe https://github.com/conda-forge/libignition-rendering4-feedstock/pull/8.

closed 2025-01-30 by diegoferigo