robotology-legacy/gym-ignition #281: import issue with panda manipulator example

HorvathDawson, 2020-12-20:
System:
PopOS 20.04 - fresh install
Ignition binary install - fresh install
Stable gym-ignition - fresh install

When running Panda_pick_and_place.py ... I get the following important error,
 
`Traceback (most recent call last):
  File "panda_pick_and_place.py", line 8, in <module>
    from gym_ignition.rbd import conversions
  File "/home/dawson/.local/lib/python3.8/site-packages/gym_ignition/rbd/__init__.py", line 5, in <module>
    from . import idyntree
  File "/home/dawson/.local/lib/python3.8/site-packages/gym_ignition/rbd/idyntree/__init__.py", line 5, in <module>
    from . import numpy
  File "/home/dawson/.local/lib/python3.8/site-packages/gym_ignition/rbd/idyntree/numpy.py", line 5, in <module>
    import idyntree.bindings as idt
  File "/home/dawson/.local/lib/python3.8/site-packages/idyntree/__init__.py", line 1, in <module>
    from . import swig
  File "/home/dawson/.local/lib/python3.8/site-packages/idyntree/swig.py", line 13, in <module>
    from . import _iDynTree
ImportError: libipopt.so.1: cannot open shared object file: No such file or directory
`

HorvathDawson, 2020-12-20:
I found the issue. Looking at the idyntree repo I found after installing the dependencies listed there,

`sudo apt-get install libeigen3-dev libxml2-dev coinor-libipopt-dev qtbase5-dev qtdeclarative5-dev qtmultimedia5-dev qml-module-qtquick2 qml-module-qtquick-window2 qml-module-qtmultimedia qml-module-qtquick-dialogs qml-module-qtquick-controls qml-module-qt-labs-folderlistmodel qml-module-qt-labs-settings` 

it was able run the example

diegoferigo, 2020-12-20:
Thanks for tracking this problem @HorvathDawson. As you found out, it was a dependency-of-a-dependency problem. 

I am the one that packaged `idyntree` for Python in its current form, and unfortunately we found limitations in what the existing standards require when building a wheel. As for gym-ignition, where the manual installation of the Ignition suite is required, also iDyntree requires the manual installation of its dependencies. We know this is a limitation but luckily the tooling supporting the new PEP600 standard (that hopefully will allow us to vendor all the dependencies in the wheel without compromising the C++ ABI) are being finalized in these days.

I'm closing this issue since the problem has been solved updating the system configuration.

closed 2020-12-20 by diegoferigo