robotology-legacy/gym-ignition #287: Capturing images from the environment

YagoGG, 2021-01-04:
Hi!

I was wondering if it is possible to programatically capture rendered images of the environment.

I have seen in #199 that there is a "camera sensor" that is yet to be exposed in the API, but I believe that is not exactly what I am looking for. I do not need it to be a sensor in one of the robots, but just a visualization like the one you get with Ignition's GUI.

Not sure if this is even out of `gym-ignition`'s scope, but since this is also the repo for the `scenario` bindings I thought it could fit here.

Please let me know if you need further details, and congratulations on the great work!

diegoferigo, 2021-01-05:
Hi @YagoGG, thanks for passing by! I'm not an expert in the rendering stack of Ignition Gazebo, if I don't mistake the GUI's view is actually a camera that does not differ from any on-board camera. If you see check the documentation on [`GzScene3D`](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo4/tutorials/gui_config.md), you can see that the view is defined by a `<camera_pose>` entry.

This being said, I think what you mean is compatible with #199. I still haven't thought about the API to expose, but I'd like to have a method that queries the available cameras of the world. Maybe the view point could even be a special case deserving a different method, or being the default of a `world.getCamera(cameraName)` when the argument is not given.

A while ago I started working on #249 but then I had to pause the PR due to other priorities. I'm not happy of that implementation, I had to vendor and modify the `Sensors` system to make it compatible with programmatic steps (otherwise image acquisition would not be synchronous and deterministic) and performances were pretty bad. I understand that having cameras is an important feature and it could be worth even if performance are not the best initially, and then iterating after bottlenecks are found.

Note that all other rendering cameras (rgb, thermal, etc) share the same infrastructure of the implemented depth camera, adding them would be trivial once that PR is finalized.

YagoGG, 2021-01-07:
Hey @diegoferigo! Thank you for your detailed response.

I have been taking a detailed look at the pointers you provided, and your reasoning sounds completely logical to me. Regarding the API design, this is what makes the most sense to me too:

> Maybe the view point could even be a special case deserving a different method, or being the default of a `world.getCamera(cameraName)` when the argument is not given.

Unfortunately, and after checking out the code in #249, I am afraid that I am not familiar enough with Ignition to continue working on it myself. But I'll definitely keep an eye on the repo, as I am really excited about `gym-ignition` and I might be able to use it in the future.

Cheers!

diegoferigo, 2021-01-11:
> Unfortunately, and after checking out the code in #249, I am afraid that I am not familiar enough with Ignition to continue working on it myself.

This is understandable. Despite I'm already familiar with most of the Ignition stack, I remember it took me a considerable effort to get #249 (kind of) working.

Ok! I cannot guarantee a prompt finalization of the PR, I'm going to leave this issue open so that you and others can track the progress of this feature.

FirefoxMetzger, 2021-03-11:
It is possible to get camera images in pure python (+ compiling the protobuf ignition messages to python), since Ignition is talking protobuf passed over zmq sockets. Although, this usage may not be intended by Ignition Gazebo, so it's probably a hack 🤷.

Here is a (hacky/dirty) example:

```python
from ignition.msgs.image_pb2 import Image
from scenario import gazebo as scenario_gazebo
from scenario import core as scenario_core
from itertools import count
import gym_ignition_environments
import numpy as np
import zmq
import subprocess
import socket
import os
import pwd
import numpy as np
import matplotlib.pyplot as plt 

ctx = zmq.Context()
camera_socket = ctx.socket(zmq.SUB)

# scenario_gazebo.set_verbosity(scenario_gazebo.Verbosity_info)

gazebo = scenario_gazebo.GazeboSimulator(
    step_size=0.001,
    rtf=1.0,
    steps_per_run=1
)
assert gazebo.insert_world_from_sdf("./panda_world_expanded.sdf")
gazebo.initialize()

world = gazebo.get_world()
panda = gym_ignition_environments.models.panda.Panda(
    world=world, position=[0.2, 0, 1.025])

# gazebo.gui()
gazebo.run(paused=True)

# this is a bad hack and should be implemented
# by talking the ign-transport discovery protocol
ign_topic = "/camera"
result = subprocess.check_output(f"ign topic -i -t {ign_topic}", shell=True)
address = result.decode("utf-8").split("\n")[1].split(",")[0].replace("\t", "").replace(" ", "")
camera_socket.connect(address)

# this could also be streamlined by speaking the ign-transport discovery protcol
host_name = socket.gethostname()
user_name = pwd.getpwuid(os.getuid())[0]
camera_socket.subscribe(f"@/{host_name}:{user_name}@{ign_topic}")

# weird hack to encourage ign-transport to actually publish camera messages
# start an echo subscriber and print the messages into the void to make ign 
# realize that something is listening to the topic
# tracking issue: https://github.com/ignitionrobotics/ign-transport/issues/225
echo_subscriber = subprocess.Popen(
    ["ign", "topic", "-e", "-t", ign_topic], 
    stdout=open(os.devnull, 'w')
)

# execute a bunch of steps (proof of concept)
for _ in range(int(1000)):
    gazebo.run(paused=False)

# fetch all camera messages published during that time
for frame_idx in count():
    try:
        msg = camera_socket.recv_multipart(zmq.NOBLOCK)
        proto_msg = Image()
        proto_msg.ParseFromString(msg[2])
[... 496 lines cut ...]
              <height>240</height>
            </image>
            <clip>
              <near>0.1</near>
              <far>100</far>
            </clip>
          </camera>
          <always_on>1</always_on>
          <update_rate>30</update_rate>
          <visualize>1</visualize>
          <topic>camera</topic>
          <plugin name='__default__' filename='__default__'/>
        </sensor>
      </link>
      <plugin name='__default__' filename='__default__'/>
    </model>
  </world>
</sdf>

```
</details>

For this to work, you will need to install the necessary python packages (most noteworthy `pyzmq`) and you will have to create python bindings for the protobuf messages (so you can deserialize them in python). To do this I've cloned the [ign-msgs](https://github.com/ignitionrobotics/ign-msgs) repo which has all the messages in the `proto` folder. From there it is a one-liner `protoc -I=<path/to/proto/folder> --python_out=<path/to/script/folder> *.proto` (from memory, ymmv).

One caveat of this approach is that the camera publishes at a much slower rate than the simulator steps (the example runs 1000 steps, and produces ~30 frames) so ensuring that the current simulator step and sensor message are in sync would require additional effort. It may be enough for the use-case of "just want to create some images/video" though.