robotology-legacy/gym-ignition #288: ImportError: libignition-gazebo4.so.4: cannot open shared object file: No such file or directory

YijiongLin, 2021-01-23:
Hi,

After installing the `pip install gym-ignition`, I ran the `example.py` from [here](https://robotology.github.io/gym-ignition/master/getting_started/scenario.html), and got this issue:

`Traceback (most recent call last):
  File "example.py", line 3, in <module>
    from scenario import gazebo as scenario_gazebo
  File "/home/th20988/py38ign/lib/python3.8/site-packages/scenario/__init__.py", line 93, in <module>
    from .bindings import core
  File "/home/th20988/py38ign/lib/python3.8/site-packages/scenario/bindings/core.py", line 12, in <module>
    import scenario.bindings.gazebo
  File "/home/th20988/py38ign/lib/python3.8/site-packages/scenario/bindings/gazebo.py", line 13, in <module>
    from . import _gazebo
ImportError: libignition-gazebo4.so.4: cannot open shared object file: No such file or directory
`
Before meeting this error, I also came across this one :
`Traceback (most recent call last):
  File "gymign_example.py", line 3, in <module>
    from scenario import gazebo as scenario_gazebo
  File "/home/th20988/py38/lib/python3.8/site-packages/scenario/__init__.py", line 93, in <module>
    from .bindings import core
  File "/home/th20988/py38/lib/python3.8/site-packages/scenario/bindings/core.py", line 12, in <module>
    import scenario.bindings.gazebo
  File "/home/th20988/py38/lib/python3.8/site-packages/scenario/bindings/gazebo.py", line 13, in <module>
    from . import _gazebo
ImportError: libpython3.8.so.1.0: cannot open shared object file: No such file or directory
`
I solved this error by solutions provided [here](https://stackoverflow.com/a/43623903/11194600) and [here](https://bugs.python.org/issue38987). I'm not sure if these solutions result in the first error.

I'm using Ubuntu 18 and install python 3.8 according to this [webpage](https://medium.com/@akhilmathew_/python3-8-from-installation-to-setting-up-a-virtual-environment-on-ubuntu-9491c016480), then python 3.8 is used as the interpreter to create a virtual env with this command `virtualenv -p python3.8 py38ign `

Any help and ideas would be highly appreciated!!

Cheers,

YijiongLin, 2021-01-23:
It seems the scenario lib is missing, or the builder cannot link to this lib.

I have checked that the `scenario` are in:
`
:~/py38ign$ find . -name scenario
./lib/python3.8/site-packages/scenario
./lib/python3.8/site-packages/scenario/include/scenario
./lib/python3.8/site-packages/scenario/lib/scenario
./lib/python3.8/site-packages/gym_ignition/scenario
`

diegoferigo, 2021-01-24:
@YijiongLin How did you install Ignition Gazebo? [Here](https://robotology.github.io/gym-ignition/master/installation/support_policy.html) you can find the requirements of both Stable and Nightly, the latest release requires Dome.

Note that Ubuntu Bionic is not officially supported. However, with minor extra steps wrt those listed in the installation guide, gym-ignition should be able to run also there.

YijiongLin, 2021-01-24:
@diegoferigo Hi, thanks for the quick reply!

I re-installed the Ignition Gazebo with the [Dome](https://ignitionrobotics.org/docs/dome/install_ubuntu) version, and then installed `pip install gym-ignition` with the Stable one.  Now another similar but different error showed up:
`
(py_test) th20988@IT080176:~$ python
Python 3.8.6 (default, Jan 24 2021, 11:16:39) 
[GCC 7.5.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
\>>> import scenario
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/th20988/py_test/lib/python3.8/site-packages/scenario/__init__.py", line 93, in <module>
    from .bindings import core
  File "/home/th20988/py_test/lib/python3.8/site-packages/scenario/bindings/core.py", line 12, in <module>
    import scenario.bindings.gazebo
  File "/home/th20988/py_test/lib/python3.8/site-packages/scenario/bindings/gazebo.py", line 13, in <module>
    from . import _gazebo
ImportError: libprotobuf.so.17: cannot open shared object file: No such file or directory
`
I tried `pip install protobuf` but it didn't work.

I also checked the GCC version in the terminal with `gcc -v` and it showed:

`
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/8/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 8.4.0-1ubuntu1~18.04' --with-bugurl=file:///usr/share/doc/gcc-8/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-8 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 8.4.0 (Ubuntu 8.4.0-1ubuntu1~18.04) 
`

Also, I checked clang version with `clang -v`

`
clang version 6.0.0-1ubuntu2 (tags/RELEASE_600/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/4.8
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/4.8.5
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/7
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/8
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/4.8
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/4.8.5
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/7
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/7.5.0
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/8
Selected GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/8
Candidate multilib: .;@m64
Selected multilib: .;@m64
`

I'm not sure what's the problem behind them, could you please give me some ideas?

Cheers,

YijiongLin, 2021-01-24:
I just install `sudo apt-get install libprotobuf-dev`, and used `find . -name *libprotobuf*` to find the library, and get these:

>./usr/share/doc/libprotobuf-lite10
./usr/share/doc/libprotobuf-dev
./usr/share/doc/libprotobuf10
./usr/lib/x86_64-linux-gnu/libprotobuf.so
./usr/lib/x86_64-linux-gnu/libprotobuf.a
./usr/lib/x86_64-linux-gnu/libprotobuf-lite.a
./usr/lib/x86_64-linux-gnu/libprotobuf-lite.so
./usr/lib/x86_64-linux-gnu/libprotobuf.so.10.0.0
./usr/lib/x86_64-linux-gnu/libprotobuf.so.10
./usr/lib/x86_64-linux-gnu/libprotobuf-lite.so.10
./usr/lib/x86_64-linux-gnu/libprotobuf-lite.so.10.0.0

It seems there is no `libprotobuf.so.17`.

diegoferigo, 2021-01-24:
As I said, Ubuntu Bionic is not officially supported, and the problems you're encountering are among the reasons. I'll try to explain the situation.

gym-ignition is a hybrid Python / C++ package that requires finding a number of system dependencies (we will lift some of these requirements as soon as the tools for PEP600 will be finalized). You can install gym-ignition in two ways from PyPI: using the `wheel` (no compilation necessary) or the `sdist` (requires the C++ buildchain and all the build dependencies). The former is of course more user friendly, but it has to find some system libraries installed. Among them, it looks for [`libprotobuf17`](https://packages.ubuntu.com/search?keywords=libprotobuf17), and bionic does not seem to have it in its repos.

You can find all the libraries that are missing by locating the file `_gazebo.so` in your venv and calling `ldd`. E.g., in my system I have the following.

<details>
<summary>ldd /path/to/scenario/bindings/_gazebo.so</summary>

```
❯ ldd /venv/lib/python3.8/site-packages/scenario/bindings/_gazebo.so
        linux-vdso.so.1 (0x00007ffea34e9000)
        libpython3.8.so.1.0 => /lib/x86_64-linux-gnu/libpython3.8.so.1.0 (0x00007efe4af9d000)
        libECMSingleton.so => /venv/lib/python3.8/site-packages/scenario/bindings/../lib/libECMSingleton.so (0x00007efe4af74000)
        libignition-gazebo4.so.4 => /rl/sources/workspace/install/lib/libignition-gazebo4.so.4 (0x00007efe4ace0000)
        libignition-fuel_tools5.so.5 => /rl/sources/workspace/install/lib/libignition-fuel_tools5.so.5 (0x00007efe4ac40000)
        libignition-common3-events.so.3 => /rl/sources/workspace/install/lib/libignition-common3-events.so.3 (0x00007efe4ac17000)
        libignition-common3.so.3 => /rl/sources/workspace/install/lib/libignition-common3.so.3 (0x00007efe4abcd000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007efe4abaa000)
        libignition-transport9.so.9 => /rl/sources/workspace/install/lib/libignition-transport9.so.9 (0x00007efe4aae7000)
        libignition-msgs6.so.6 => /rl/sources/workspace/install/lib/libignition-msgs6.so.6 (0x00007efe4a6c3000)
        libsdformat10.so.10 => /rl/sources/workspace/install/lib/libsdformat10.so.10 (0x00007efe4a4c6000)
        libignition-math6.so.6 => /rl/sources/workspace/install/lib/libignition-math6.so.6 (0x00007efe4a462000)
        libprotobuf.so.17 => /lib/x86_64-linux-gnu/libprotobuf.so.17 (0x00007efe4a16a000)
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007efe49f89000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007efe49e3a000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007efe49e1d000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007efe49c2b000)
        libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007efe49bfd000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007efe49be1000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007efe49bdb000)
        libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007efe49bd6000)
        libignition-plugin1-loader.so.1 => /rl/sources/workspace/install/lib/libignition-plugin1-loader.so.1 (0x00007efe49bbe000)
        libignition-plugin1.so.1 => /rl/sources/workspace/install/lib/libignition-plugin1.so.1 (0x00007efe49bb1000)
        libQt5Core.so.5 => /lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007efe49668000)
        libtinyxml2.so.6 => /lib/x86_64-linux-gnu/libtinyxml2.so.6 (0x00007efe49651000)
        /lib64/ld-linux-x86-64.so.2 (0x00007efe4b729000)
        libjsoncpp.so.1 => /lib/x86_64-linux-gnu/libjsoncpp.so.1 (0x00007efe4961b000)
        libyaml-0.so.2 => /lib/x86_64-linux-gnu/libyaml-0.so.2 (0x00007efe495f9000)
        libzip.so.5 => /lib/x86_64-linux-gnu/libzip.so.5 (0x00007efe495dc000)
        libcurl.so.4 => /lib/x86_64-linux-gnu/libcurl.so.4 (0x00007efe4954b000)
        libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007efe49542000)
        libzmq.so.5 => /lib/x86_64-linux-gnu/libzmq.so.5 (0x00007efe4949c000)
        liburdfdom_model.so.1.0 => /lib/x86_64-linux-gnu/liburdfdom_model.so.1.0 (0x00007efe49477000)
        libicui18n.so.66 => /lib/x86_64-linux-gnu/libicui18n.so.66 (0x00007efe49176000)
        libicuuc.so.66 => /lib/x86_64-linux-gnu/libicuuc.so.66 (0x00007efe48f90000)
        libpcre2-16.so.0 => /lib/x86_64-linux-gnu/libpcre2-16.so.0 (0x00007efe48f0d000)
        libdouble-conversion.so.3 => /lib/x86_64-linux-gnu/libdouble-conversion.so.3 (0x00007efe48ef7000)
        libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007efe48dce000)
        libbz2.so.1.0 => /lib/x86_64-linux-gnu/libbz2.so.1.0 (0x00007efe48db9000)
        libcrypto.so.1.1 => /lib/x86_64-linux-gnu/libcrypto.so.1.1 (0x00007efe48ae3000)
        libnghttp2.so.14 => /lib/x86_64-linux-gnu/libnghttp2.so.14 (0x00007efe48aba000)
        libidn2.so.0 => /lib/x86_64-linux-gnu/libidn2.so.0 (0x00007efe48a99000)
        librtmp.so.1 => /lib/x86_64-linux-gnu/librtmp.so.1 (0x00007efe48a79000)
        libssh.so.4 => /lib/x86_64-linux-gnu/libssh.so.4 (0x00007efe48a0b000)
        libpsl.so.5 => /lib/x86_64-linux-gnu/libpsl.so.5 (0x00007efe489f6000)
        libssl.so.1.1 => /lib/x86_64-linux-gnu/libssl.so.1.1 (0x00007efe48963000)
        libgssapi_krb5.so.2 => /lib/x86_64-linux-gnu/libgssapi_krb5.so.2 (0x00007efe48916000)
        libldap_r-2.4.so.2 => /lib/x86_64-linux-gnu/libldap_r-2.4.so.2 (0x00007efe488c0000)
[... cut ...]
A possible workaround to make gym-ignition run in Ubuntu Bionic, or any other non-supported distribution, is to install it from `sdist`. You can do it with the following command:

```
pip3 install gym-ignition --no-binary "gym-ignition"
```

This could not be straighforward. In fact, beyond the runtime dependencies like Ignition Dome, installing from `sdist` requires the build dependencies. To mention one, Ubuntu Bionic has SWIG 3 in its repo, therefore you should install version 4 from sources.

diegoferigo, 2021-02-09:
Closing this issue since it was referring to an unsupported platform. Feel free to comment again in case you need more advice.

closed 2021-02-09 by diegoferigo

Gaoyuan-Liu, 2021-03-09:
Hey @diegoferigo,
I met the same issue and thanks for your explanation I understand what's happening.
But that means the gym-ignition only support ubuntu OS later than 18.04 which is not very old actually, a lot of ros projects are build on ubuntu 18.04 even older version. 
I think it would be pretty nice to make gym-ignition work easily on Ubuntu Bionic, or give a building instruction.

Thanks.

diegoferigo, 2021-03-09:
> Hey @diegoferigo,
> I met the same issue and thanks for your explanation I understand what's happening.
> But that means the gym-ignition only support ubuntu OS later than 18.04 which is not very old [...]

Hi @Gaoyuan-Liu, thanks for the feedback. I understand your point. We supported Ubuntu Bionic for almost 2 years and the installation instructions were quite complicated. As soon as Focal has been released, we started migrating our systems and recommend all those who cannot to rely on docker. Nowadays, it's pretty simple 1) creating a container 2) disabling network isolation and 3) enabling X11 support. In this way, the host could be your preferred and relatively old distribution.

This being said, since you're not the first asking some steps for Ubuntu Bionic, you can try the following:

- Install Python 3.8 (you can use [this ppa](https://launchpad.net/~deadsnakes/+archive/ubuntu/ppa))
- Install SWIG >= 4.0 [from sources](https://github.com/swig/swig)
- Install Ignition Dome as you prefer (either from ppa or colcon)
- Build the gym-ignition wheel locally:
   ```bash
   cd /tmp
   git clone https://github.com/robotology/gym-ignition
   cd gym-ignition
   python3 setup.py bdist_wheel
   pip3 install dist/*.whl
   ```

I went open loop with the commands above, please let me know if something is wrong.

diegoferigo, 2021-03-09:
For those reading this issue in the future, I also cross-link https://github.com/robotology/gym-ignition/issues/293#issuecomment-789545130. In our organization we started exploring conda and conda-forge. Such setup would provide a unified, distro-agnostic, multi-platform installation scheme. We're still in exploration phase, but on the long run it could become the preferred way to install also gym-ignition.