robotology-legacy/gym-ignition #295: Failed building wheel for gym-ignition

Gaoyuan-Liu, 2021-03-09:
Hey,
I'm using ubuntu 18.04 + python3.6. 
I followed the setup instruction to install the [demo](https://ignitionrobotics.org/docs/dome), it doesn't give any error, but after that when I run `pip install gym-ignition` in the terminal, I got the following error and install failed:
` ----------------------------------------
  Failed building wheel for gym-ignition
  Running setup.py clean for gym-ignition
Failed to build gym-ignition`

Maybe I do something wrong, any ideas?
Thanks!

closed 2021-03-09 by Gaoyuan-Liu

Seiwert, 2021-03-21:
@diegoferigo I ran into this issue as well and it wasn't immediately clear what the problem was. I would recommend adding a few more details to the README to clarify the necessity of python =>3.8. I attempted to do this with a PR but do not have access.

diegoferigo, 2021-03-22:
The README was structured to be a distilled version of the website, where information have been centralized. The system requirements are listed in [Support Policy](https://robotology.github.io/gym-ignition/master/installation/support_policy.html).