robotology-legacy/gym-ignition #301: Failed to spawn panda

Gaoyuan-Liu, 2021-03-17:
Hey,
I followed the instruction and installed the gym-ignition without any error. (ubuntu 20.04, python3.8, virtual env)
And when I try to test the panda example given [here](https://robotology.github.io/gym-ignition/master/getting_started/manipulation.html), the gazebo can popup but not panda, instead, a bunch of weird triangle shapes show in gazebo. 
[image]
Any ideas if I missing some steps?

Thanks!

Gaoyuan-Liu, 2021-03-17:
FYI, the pendulum example works fine.

FirefoxMetzger, 2021-03-17:
Could you share the exact code that you are using to spawn panda and the logs produced by the call?

wolfgangschwab, 2021-03-17:
I don't know whether it's the same issue. But I saw a similar issue in another package which also brought pictures with triangles:

[image]

The reason there were german region settings. It seems ign-fuel gets confused if you use a comma for the decimal point in your system. Do you used the standard english settings?

Gaoyuan-Liu, 2021-03-18:
Hey @FirefoxMetzger @wolfgangschwab,
Thanks a lot for your response. 
I think the reason for the "triangle error" is indeed the region settings. For my case I changed the setting from Belgium to US, now I can see the panda in my gazebo!
Here are still some small problems:
1. The robot doesn't show color or details:
[image]
No related error shows in the terminal. 
2. The error related to "insert_bucket":

> [Err] [SystemPaths.cc:444] Could not resolve file [texture.png]

> [GUI] [Err] [SystemPaths.cc:462] Could not resolve file [texture.png]

[image]
which means if I comment out `bucket = insert_bucket(world=world)`, there will be no error.

What I did are:
1. Install ignition_demo & gym_ignition (virtual env);
2. Copy the python code provided [here](https://robotology.github.io/gym-ignition/master/getting_started/manipulation.html) in my /home directory as test.py;
3. Activate virtual env;
4. `python test.py`

I get the following info after I run `python test.py` and press Ctrl-c to stop it.
[image]

Thanks for your help!
Gaoyuan

traversaro, 2021-03-18:
>  I think the reason for the "triangle error" is indeed the region settings. For my case I changed the setting from Belgium to US, now I can see the panda in my gazebo!

Possibly related issue: https://github.com/ignitionrobotics/ign-rendering/issues/136 .

Gaoyuan-Liu, 2021-03-19:
Hey @traversaro,
I understand now that the "triangle error" is because of the locale setting, but did you mean this "no color issue" is also for the same reason?

Thanks.

traversaro, 2021-03-19:
>  I read the link you point, it's very confusing to me, maybe because I don't have enough knowledge of C++. So how I can solve this issue as a user of ignition-gazebo and gym-ignition? Now what I'm doing is changing the whole ubuntu language system to US, but as I mentioned, there are still some problems.

Hi @Gaoyuan-Liu, those links are meant to developers to solve the problem that you are solving by setting the whole ubuntu locale to US, they were not meant to your specific issue. Furthermore, they are not related to the other problems that you mention in https://github.com/robotology/gym-ignition/issues/301#issuecomment-801785099 .

diegoferigo, 2021-03-19:
> I think the reason for the "triangle error" is indeed the region settings. For my case I changed the setting from Belgium to US, now I can see the panda in my gazebo!

I know this makes @traversaro happy, he's collecting the occurrence of these kind of problems :smile: Jokes apart, it seems an upstream problem and regarding it we have to wait for a fix. Changing temporarily the locale should work as @Gaoyuan-Liu confirmed.

> I understand now that the "triangle error" is because of the locale setting, but did you mean this "no color issue" is also for the same reason?

For the color problem I don't have many ideas. Can you please report what is your graphics stack? Can you try changing the rendering backend from `ogre2` to `ogre` as [described in FAQ](https://github.com/robotology/gym-ignition/blob/master/docs/sphinx/info/faq.rst#ogre2-and-opengl)? For example, on my laptop, if I disable the Nvidia GPU, I have to switch to `ogre` otherwise the GUI segfaults. It all depends on which OpenGL version your video driver supports. It's a know limitation and in this moment this is the first workaround I can think of.

FirefoxMetzger, 2021-03-19:
> The error related to "insert_bucket":

The `texture.png` error is related to general problems with handling Fuel models; though I can't tell you the exact root of the problem. If you check the model [on the fuel website](https://app.ignitionrobotics.org/GoogleResearch/fuel/models/Threshold_Basket_Natural_Finish_Fabric_Liner_Small) you can see that there is indeed a `Texture.png` among the files of the basket model and that it indeed contains the texture for the basket as it should. You have downloaded this file, and it is being used by the rendering engine (otherwise your basked would not show the correct texture); however, the system still complains that the file is missing. I've noticed this for other models, too, but - as far as I am aware - this is a false alarm and you are _probably_ safe ignoring it.

> The robot doesn't show color or details

The robot (and the rest of the scene) are actually showing color details. You can see this particularly well when inspecting the basket (that displays the exact texture it claims it can't find 🤷). What is missing are shadows, and reflections. This is typical for scenes lit by ambient light only and it could be that you didn't add a directional light source. Either that or the rendering engine struggles to render the light source. Latter would be surprising, given that ambient light renders fine.

Could you use the Gazebo GUI to export the world into a new file and share that here? 

Additionally, you can get more verbose logs (that might contain a clue) by adding 

```python
scenario_gazebo.set_verbosity(scenario_gazebo.Verbosity_info)
# or
scenario_gazebo.set_verbosity(scenario_gazebo.Verbosity_debug)
```

to your script and/or calling it via `IGN_VERBOSE=1 python test.py` (very verbose). The output of `IGN_VERBOSE` is different from `set_verbosity`, so both are complementing each other.

traversaro, 2021-03-19:
> I know this makes @traversaro happy, he's collecting the occurrence of these kind of problems 😄

[image]

Gaoyuan-Liu, 2021-03-19:
Hey @FirefoxMetzger,
1. I saved the world through GUI as .sdf file:
`<sdf version='1.7'>
  <world name='default'>
    <physics default='1' type='dart'>
      <max_step_size>0.001</max_step_size>
      <real_time_factor>2</real_time_factor>
      <real_time_update_rate>-1</real_time_update_rate>
    </physics>
    <light name='sun' type='directional'>
      <cast_shadows>1</cast_shadows>
      <pose>0 0 10 0 -0 0</pose>
      <diffuse>1 1 1 1</diffuse>
      <specular>0.5 0.5 0.5 1</specular>
      <attenuation>
        <range>1000</range>
        <constant>0.9</constant>
        <linear>0.01</linear>
        <quadratic>0.001</quadratic>
      </attenuation>
      <direction>-0.5 0.1 -0.9</direction>
      <spot>
        <inner_angle>0</inner_angle>
        <outer_angle>0</outer_angle>
        <falloff>0</falloff>
      </spot>
    </light>
    <gravity>0 0 -9.8</gravity>
    <magnetic_field>6e-06 2.3e-05 -4.2e-05</magnetic_field>
    <atmosphere type='adiabatic'/>
    <scene>
      <ambient>0.4 0.4 0.4 1</ambient>
      <background>0.7 0.7 0.7 1</background>
      <shadows>1</shadows>
    </scene>
    <plugin name='__default__' filename='__default__'/>
    <model name='ground_plane'>
      <static>1</static>
      <link name='link'>
        <collision name='collision'>
          <geometry>
            <plane>
              <normal>0 0 1</normal>
              <size>1 1</size>
            </plane>
          </geometry>
          <surface>
            <friction>
              <ode>
                <mu>100</mu>
              </ode>
            </friction>
            <bounce>
              <restitution_coefficient>1</restitution_coefficient>
            </bounce>
            <contact/>
          </surface>
        </collision>
        <visual name='visual'>
          <geometry>
            <plane>
              <normal>0 0 1</normal>
              <size>100 100</size>
            </plane>
          </geometry>
          <material>
            <ambient>0.8 0.8 0.8 1</ambient>
            <diffuse>0.8 0.8 0.8 1</diffuse>
            <specular>0.8 0.8 0.8 1</specular>
          </material>
[... 811 lines cut ...]
          <geometry>
            <box>
              <size>0.05 0.05 0.05</size>
            </box>
          </geometry>
          <material>
            <script>
              <uri>file://media/materials/scripts/gazebo.material</uri>
              <name>Gazebo/Wood</name>
            </script>
          </material>
          <plugin name='__default__' filename='__default__'/>
        </visual>
      </link>
      <plugin name='__default__' filename='__default__'/>
      <pose>0.243278 0.086602 1.015 0 -0 0</pose>
    </model>
  </world>
</sdf>`

2. Followed your instruction (`IGN_VERBOSE`+`set_verbosity`), I got no extra error info except one warning:
[image]
But it seems not related to the light problem.

Thanks!

FirefoxMetzger, 2021-03-20:
I tried loading your scene, and it loads fine (with exceptions of those that use mesh paths specific to your machine) including a light source.

Do the Ignition example worlds render fine? 

`ign gazebo shapes.sdf` should look something like this:
[image]

If not, there is a general problem with your ignition installation.

Gaoyuan-Liu, 2021-03-21:
Hey @FirefoxMetzger,
I can spawn the shapes and it looks good:
[image]
I try to drag a panda model into this world and it doesn't have the problem:
[image]
It feels like the problem is that the world initialized by:
 `gazebo, world = gym_ignition.utils.scenario.init_gazebo_sim( step_size=0.001, real_time_factor=2.0, steps_per_run=1)`
dose't initialize the light conditions.
So is there any python API that can open a specified pre-defined world?

FirefoxMetzger, 2021-03-21:
What about building the world from scratch in `gym-ignition`? Does that give you the missing light source?

> So is there any python API that can open a specified pre-defined world?

There is ... sort of: `scenario_gazebo.GazeboSimulator(...).insert_world_from_sdf` Fuel support is limited at the moment though. It is something I plan to look into after my deadline in early April (and after I manage to figure out why the development build of gym-ignition keeps complaining about my protobuf version).

You can check https://github.com/robotology/gym-ignition/issues/296#issuecomment-795998022 , but the entire issue is related.

Gaoyuan-Liu, 2021-03-21:
> What about building the world from scratch in gym-ignition? Does that give you the missing light source?
Could you give some instruction about how to build a world from scratch in gym-ignition?

Thanks.

FirefoxMetzger, 2021-03-22:
> Could you give some instruction about how to build a world from scratch in gym-ignition?

For example the code in the scenarIO example :)

https://robotology.github.io/gym-ignition/master/getting_started/scenario.html

Gaoyuan-Liu, 2021-03-22:
No light either:
[image]

diegoferigo, 2021-03-22:
> For the color problem I don't have many ideas. Can you please report what is your graphics stack? Can you try changing the rendering backend from `ogre2` to `ogre` as [described in [...]

Ogre and ogre-next (ogre2) handle materials in different ways and lightning could differ from published screenshots. It would be helpful if you could provide the info I asked above and maybe try to change rendering engine.

Gaoyuan-Liu, 2021-03-23:
Hey,
My graphics info:
  *-display                 
       description: VGA compatible controller
       product: Iris Graphics 540
       vendor: Intel Corporation
       physical id: 2
       bus info: pci@0000:00:02.0
       version: 0a
       width: 64 bits
       clock: 33MHz
       capabilities: pciexpress msi pm vga_controller bus_master cap_list rom
       configuration: driver=i915 latency=0
       resources: irq:135 memory:db000000-dbffffff memory:90000000-9fffffff ioport:f000(size=64) memory:c0000-dffff
The version of OpenGL:
       OpenGL version string: 4.6 (Compatibility Profile) Mesa 20.2.6

Changing <engine> from ogre2 to ogre didn't help and got the error:
[image]
and the world became black and white:
[image]

2 observation:
1. Even though there is no actual light in the world, but there is a green box which (I think) is the light sign:
[image]
2. I try to use GUI to manually add extra light, it can show the expecting effect:
[image]

Thanks.