robotology-legacy/gym-ignition #314: Enable self-collision for Panda

FirefoxMetzger, 2021-03-31:
I don't know if this issue should live here or in the `gym_ignition_models` repo. 

Currently, Panda doesn't seem to have self-collisions enabled. Collision with other objects works fine. See for example this video (at around 0:08 - 0:09):

https://user-images.githubusercontent.com/4402489/113131592-43b49580-91d2-11eb-863c-84bcdb292fee.mp4

Would it be possible to enable it by default, or provide an option to enable it?

diegoferigo, 2021-03-31:
Contacts and collisions are part of what I call the bottleneck features. When enabled, simulations could slow down considerably.

For this reason, contact detection and self collisions are disabled by default ([contacts](https://github.com/robotology/gym-ignition/blob/bba62416e97fe5788019e32cae66355bcb9fe70a/cpp/scenario/gazebo/src/Link.cpp#L168-L171) [1], [collisions](https://github.com/robotology/gym-ignition/blob/bba62416e97fe5788019e32cae66355bcb9fe70a/cpp/scenario/gazebo/src/Model.cpp#L175-L178)). You can enable self collisions model-wise with the following method:

https://github.com/robotology/gym-ignition/blob/bba62416e97fe5788019e32cae66355bcb9fe70a/cpp/scenario/gazebo/include/scenario/gazebo/Model.h#L174-L183

[1] I actually realized that contacts are enabled by default. We should understand what is the impact to the real time factor, and if it makes sense to disable them by default.

FirefoxMetzger, 2021-03-31:
In python this will need to be cast `to_gazebo()` first:

```
panda.to_gazebo().enable_self_collisions(True)
```

I should first read the docs more thoroughly before opening new issues 🥇 I totally forgot to check the Gazebo API and just looked at the core API. Closing...

closed 2021-03-31 by FirefoxMetzger

diegoferigo, 2021-03-31:
No worries, we can consider issues as part of a project's documentation :wink: One nice thing with SWIG (much more difficult to get with pybind11) is that the C++ doxygen documentation is copied to the Python's docstring. I admit that, as you, I rarely rely on the online documentation. Nowadays autocompletion is much faster.