robotology-legacy/gym-ignition #329: CanonicalLink updates

diegoferigo, 2021-04-09:
There is some upstream activity in https://github.com/ignitionrobotics/ign-gazebo/pull/736 that touches the `CanonicalLink` component. Refer to https://github.com/ignitionrobotics/ign-gazebo/pull/736#discussion_r609805967 and other comments.

Changes will not break the functionality within Edifice (the target branch of the PR) and nothing will stop working from our side, though I open this issue as reminder to check what's the outcome of the PR and adapt, if needed, align our logic. Currently, our `devel` branch supports Edifice.

Currently we use the `CanonicalLink` component in the following methods:

- [`Model::resetBasePose`](https://github.com/robotology/gym-ignition/blob/50caccfa1b5ddb617ec0b1aea98bfec72c8fbadf/cpp/scenario/gazebo/src/Model.cpp#L256-L289)
- [`Model::resetBaseWorldVelocity`](https://github.com/robotology/gym-ignition/blob/50caccfa1b5ddb617ec0b1aea98bfec72c8fbadf/cpp/scenario/gazebo/src/Model.cpp#L343-L377)
- [`Model::baseFrame`](https://github.com/robotology/gym-ignition/blob/50caccfa1b5ddb617ec0b1aea98bfec72c8fbadf/cpp/scenario/gazebo/src/Model.cpp#L952-L974)
- [`Model::baseWorldLinearVelocity`](https://github.com/robotology/gym-ignition/blob/50caccfa1b5ddb617ec0b1aea98bfec72c8fbadf/cpp/scenario/gazebo/src/Model.cpp#L1024-L1061)

adlarkin, 2021-05-14:
Just so you're aware, we changed how things were implemented in https://github.com/ignitionrobotics/ign-gazebo/pull/736 so that the `CanonicalLink` component remains unchanged (https://github.com/ignitionrobotics/ign-gazebo/pull/736#issuecomment-830184043). So, hopefully, you don't need to change anything :slightly_smiling_face:

diegoferigo, 2021-05-14:
Hi @adlarkin, thanks for chiming in! Awesome, it seems a much more neat and future proof solution moving the logic into the Physics system. Great work! Closing.

closed 2021-05-14 by diegoferigo