robotology-legacy/gym-ignition #402: After updating to v1.3.0 stable render now instantly closes and physics engine path isn't found.

HorvathDawson, 2021-10-04:
- [ ] I've read the [FAQ][FAQ]
- [ ] I've read the [Support Policy][Support Policy]
- [ ] I've already searched similar [issues][Issues] and [discussions][Discussions]
- [ ] I've already updated to the most recent supported [release][Releases] (either Stable or Nightly)
    
## Description:

I just updated to gym-ignition v1.3.0 stable. After upgrading gym-ignition and ignition-gazebo to fortress from dome the first issue I encountered was 
```
[Err] [Physics.cc:680] Failed to find plugin [ignition-physics5-dartsim-plugin]. Have you checked the IGN_GAZEBO_PHYSICS_ENGINE_PATH environment variable?
```
 which was solved by,
```
export IGN_GAZEBO_PHYSICS_ENGINE_PATH=${IGN_GAZEBO_PHYSICS_ENGINE_PATH}:/usr/lib/x86_64-linux-gnu/ign-physics-5/engine-plugins/
```

However there is another issue... rendering does not work anymore. The GUI loads for half a second then instantly closes. This is an issue on both my code and on the launch_cartpole example. Any ideas on how to solve this?

## Environment

- OS: PopOS 20.04
- GPU: RTX 1650   
- Python: 3.8.10
- Version: v1.3.0
- Channel:
  - Stable
- Installation type:
  - User

HorvathDawson, 2021-10-04:
After running Debug I get the output.
```
DEBUG: GazeboEnvRandomizer: {'kwargs': {'reward_class_name': 'BalancingV1'}}
INFO: Making new env: Monopod-v1 ({'reward_class_name': 'BalancingV1', 'physics_engine': 0})
DEBUG: {'self': <gym_ignition.runtimes.gazebo_runtime.GazeboRuntime object at 0x7fcc5710bdc0>, 'task_cls': <class 'gym_bb.tasks.monopod_builder.MonopodBuilder'>, 'agent_rate': 1000, 'physics_rate': 1000, 'real_time_factor': 3.4028234663852886e+38, 'physics_engine': 0, 'world': None}
[Msg] [GazeboSimulator.cpp:461] No world file passed, using the default empty world
[Msg] Loading SDF string. File path not available.
[Msg] Loaded level [3]
[Msg] No systems loaded from SDF, loading defaults
[Wrn] [ServerConfig.cc:860] IGN_GAZEBO_SERVER_CONFIG_PATH set but no file found, no plugins loaded
[Msg] Serving world controls on [/world/default/control] and [/world/default/playback/control]
[Msg] Serving GUI information on [/world/default/gui/info]
[Msg] World [default] initialized with [default_physics] physics profile.
[Msg] Serving world SDF generation service on [/world/default/generate_world_sdf]
[Msg] Serving world names on [/gazebo/worlds]
[Msg] Resource path add service on [/gazebo/resource_paths/add].
[Msg] Resource path get service on [/gazebo/resource_paths/get].
[Msg] Resource paths published on [/gazebo/resource_paths].
[Dbg] [GazeboSimulator.cpp:623] World 'default' successfully processed by ECMProvider
[Dbg] [GazeboSimulator.cpp:741] Starting the gazebo server
[Msg] Found no publishers on /stats, adding root stats topic
[Msg] Found no publishers on /clock, adding root clock topic
[Dbg] [SimulationRunner.cc:514] Creating PostUpdate worker threads: 1
[Dbg] [GazeboSimulator.cpp:790] Creating and caching World 'default'
[Dbg] [World.cpp:225] Initializing world 'default' with physics parameters:
rtf=3.40282e+38
step=0.001
type=ignored
[Msg] [World.cpp:232] Models:
[Msg] [Model.cpp:147] Model: [5] ground_plane
[Msg] [Model.cpp:160] Links:
[Msg] [Link.cpp:147]   [6] link
[Msg] [Model.cpp:169] Joints:
[Dbg] [Physics.cc:723] Loaded [ignition::physics::dartsim::Plugin] from library [/usr/lib/x86_64-linux-gnu/ign-physics-5/engine-plugins/libignition-physics5-dartsim-plugin.so]
[Dbg] [SimulationRunner.cc:877] Loaded system [scenario::plugins::gazebo::Physics] for entity [1]
/home/capstone/.local/lib/python3.8/site-packages/gym/logger.py:34 UserWarning: WARN: Box bound precision lowered by casting to float64
[Dbg] [GazeboSimulator.cpp:339] Starting the SceneBroadcaster plugin
[Dbg] [SimulationRunner.cc:877] Loaded system [ignition::gazebo::systems::SceneBroadcaster] for entity [1]
[Dbg] [GazeboSimulator.cpp:374] Waiting GUI to show up... 
[Dbg] [GazeboSimulator.cpp:374] Waiting GUI to show up... 
[Msg] Ignition Gazebo GUI    v6.0.0
[Dbg] [GazeboSimulator.cpp:374] Waiting GUI to show up... 
[Dbg] [Application.cc:87] Initializing application.
[GUI] [Dbg] [Application.cc:427] Create main window
[Dbg] [GazeboSimulator.cpp:374] Waiting GUI to show up... 
[Dbg] [GazeboSimulator.cpp:374] Waiting GUI to show up... 
[GUI] [Wrn] [Application.cc:669] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: Failed to initialize QSettings instance. Status code is: 1
[GUI] [Wrn] [Application.cc:669] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: The following application identifiers have not been set: QVector("organizationName", "organizationDomain")
[GUI] [Wrn] [Application.cc:669] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: Failed to initialize QSettings instance. Status code is: 1
[GUI] [Wrn] [Application.cc:669] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: The following application identifiers have not been set: QVector("organizationName", "organizationDomain")
[GUI] [Dbg] [PathManager.cc:66] Requesting resource paths through [/gazebo/resource_paths/get]
[GUI] [Wrn] [Application.cc:669] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:669] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: Failed to initialize QSettings instance. Status code is: 1
[GUI] [Wrn] [Application.cc:669] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: The following application identifiers have not been set: QVector("organizationName", "organizationDomain")
[GUI] [Wrn] [Application.cc:669] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[GUI] [Dbg] [Gui.cc:149] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[Dbg] [GazeboSimulator.cpp:374] Waiting GUI to show up... 
[Dbg] [GazeboSimulator.cpp:374] Waiting GUI to show up... 
[GUI] [Dbg] [PathManager.cc:55] Received resource paths.
[GUI] [Dbg] [Gui.cc:208] Requesting GUI from [/world/default/gui/info]...
[GUI] [Dbg] [GuiRunner.cc:96] Requesting initial state from [/world/default/state]...
[GUI] [Wrn] [Filesystem.cc:376] Failed to create file [/home/capstone/.ignition/gazebo/6/gui.config]: No such file or directory
[... cut ...]
[Dbg] [GazeboSimulator.cpp:374] Waiting GUI to show up... 
[Dbg] [GazeboSimulator.cpp:374] Waiting GUI to show up... 
[Dbg] [GazeboSimulator.cpp:374] Waiting GUI to show up... 
[Dbg] [GazeboSimulator.cpp:374] Waiting GUI to show up... 
[Dbg] [GazeboSimulator.cpp:374] Waiting GUI to show up... 
[Dbg] [GazeboSimulator.cpp:374] Waiting GUI to show up... 
[Dbg] [GazeboSimulator.cpp:374] Waiting GUI to show up... 
[Dbg] [GazeboSimulator.cpp:374] Waiting GUI to show up... 
[Dbg] [GazeboSimulator.cpp:374] Waiting GUI to show up...
```

diegoferigo, 2021-10-04:
The problem about the plugin path is [kind of known](https://robotology.github.io/gym-ignition/master/installation/developer.html#system-configuration), even though I never figured out what's the source of it. It used to affect only the installation from sources, now apparently also the installation from binaries.

For what regards the error with the GUI, these lines seem relevant:

> [GUI] [Wrn] [Filesystem.cc:376] Failed to create file [/home/capstone/.ignition/gazebo/6/gui.config]: No such file or directory
[GUI] [Err] [Gui.cc:269] Failed to copy installed config [/usr/share/ignition/ignition-gazebo6/gui/gui.config] to default config [/home/capstone/.ignition/gazebo/6/gui.config].

Can you try the following? The first one could suffice:

```bash
# Remove the config folder so that Ignition creates a new one the next time it starts
mv $HOME/.ignition $HOME/.ignition_bak

# If there are still folder errors, try to create the folder yourself
mkdir -p $HOME/.ignition/gazebo/6
```

xref https://github.com/ignitionrobotics/ign-gazebo/pull/1050, https://github.com/ignitionrobotics/ign-gui/issues/294 https://github.com/ignitionrobotics/ign-gazebo/pull/1070

HorvathDawson, 2021-10-05:
@diegoferigo  

The possible solution you provided worked as a solution.

Thank you for the quick response. Was this cause by having an older version of ignition installed prior?

diegoferigo, 2021-10-05:
Yes, until fortress the configuration that is stored in the dot folder has (always?) been compatible, at least I don't remember a time in which I had a real crash like in your case. As you can read from the PR I cross-referenced, few things have changed in fortress and in upstream they are implementing a per-version dot folder in order to prevent future similar breakings. Though, this seems not fully backward compatible since you experienced this problem. I don't exclude that it is our issue though, maybe by launching Ignition Gazebo from the command line it does some migration operation that we don't.

closed 2021-10-05 by diegoferigo