robotology-legacy/gym-ignition #423: Failed to find Ignition Gazebo ~=6.0.0.dev

dnovischi, 2021-11-15:
## Description:

Running the example for ScenarIO form the docs in [1] fails:

```
Traceback (most recent call last):
  File "/home/dan/workspace/mrsrl_ws/test/main.py", line 3, in <module>
    from scenario import gazebo as scenario_gazebo
  File "/home/dan/anaconda3/envs/mrsrl/lib/python3.8/site-packages/scenario/__init__.py", line 227, in <module>
    check_gazebo_installation()
  File "/home/dan/anaconda3/envs/mrsrl/lib/python3.8/site-packages/scenario/__init__.py", line 175, in check_gazebo_installation
    raise RuntimeError(msg)
RuntimeError: Failed to find Ignition Gazebo ~=6.0.0.dev (found incompatible version(s): [<Version('6.1.0')>, <Version('5.2.0')>, <Version('4.12.0')>, <Version('3.12.0')>])
```

[1][https://robotology.github.io/gym-ignition/master/getting_started/scenario.html#python](https://robotology.github.io/gym-ignition/master/getting_started/scenario.html#python)

## Steps to reproduce

1. `sudo apt install ignition-*`
1. `pip install gym-ignition`
1. `python example.py`

## Environment

- OS: <Ubuntu 20.04 Focal>
- GPU: <NVidia 1050Ti>    
- Python: <3.8.X from conda environment -->
- Version: 
- Channel:
  - [x] Stable
- Installation type:
  - [x] User
  - [ ] Developer
- Conda environment packages:
 ```
conda list
# packages in environment at ../anaconda3/envs/mrsrl:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main  
_openmp_mutex             4.5                       1_gnu  
absl-py                   0.15.0             pyhd8ed1ab_0    conda-forge
aiohttp                   3.7.4.post0      py38h497a2fe_0    conda-forge
anyio                     3.3.4            py38h578d9bd_1    conda-forge
argon2-cffi               20.1.0           py38h27cfd23_1  
async-timeout             3.0.1            py38h06a4308_0  
async_generator           1.10                       py_0    conda-forge
attrs                     21.2.0             pyhd8ed1ab_0    conda-forge
babel                     2.9.1              pyh44b312d_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
blas                      1.0                         mkl  
bleach                    4.1.0              pyhd8ed1ab_0    conda-forge
blinker                   1.4                        py_1    conda-forge
brotli                    1.0.9                h7f98852_5    conda-forge
brotli-bin                1.0.9                h7f98852_5    conda-forge
brotlipy                  0.7.0           py38h497a2fe_1001    conda-forge
bzip2                     1.0.8                h7b6447c_0  
c-ares                    1.17.1               h27cfd23_0  
ca-certificates           2021.10.8            ha878542_0    conda-forge
cachetools                4.2.4              pyhd8ed1ab_0    conda-forge
certifi                   2021.10.8        py38h578d9bd_1    conda-forge
cffi                      1.14.5           py38h261ae71_0  
chardet                   4.0.0            py38h578d9bd_2    conda-forge
charset-normalizer        2.0.0              pyhd8ed1ab_0    conda-forge
click                     8.0.3            py38h578d9bd_1    conda-forge
cloudpickle               2.0.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
[... 153 lines cut ...]
testpath                  0.5.0              pyhd8ed1ab_0    conda-forge
threadpoolctl             2.1.0              pyh5ca1d4c_0    anaconda
tk                        8.6.11               h1ccaba5_0  
torchaudio                0.10.0               py38_cu113    pytorch
torchvision               0.11.1               py38_cu113    pytorch
tornado                   6.1              py38h497a2fe_1    conda-forge
tqdm                      4.62.3             pyhd8ed1ab_0    conda-forge
traitlets                 5.1.1              pyhd8ed1ab_0    conda-forge
typing-extensions         3.10.0.2             hd3eb1b0_0  
typing_extensions         3.10.0.2           pyh06a4308_0  
urllib3                   1.26.7             pyhd8ed1ab_0    conda-forge
wcwidth                   0.2.5              pyh9f0ad1d_2    conda-forge
webencodings              0.5.1                      py_1    conda-forge
websocket-client          1.1.0            py38h578d9bd_1    conda-forge
werkzeug                  2.0.1              pyhd8ed1ab_0    conda-forge
wheel                     0.37.0             pyhd3eb1b0_1  
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xz                        5.2.5                h7b6447c_0  
yarl                      1.6.3            py38h497a2fe_2    conda-forge
zeromq                    4.3.4                h9c3ff4c_0    conda-forge
zipp                      3.6.0              pyhd8ed1ab_0    conda-forge
zlib                      1.2.11               h7b6447c_3  
zstd                      1.4.9                haebb681_0  
```

diegoferigo, 2021-11-15:
Hi @dnovischi, we are aware that `master` only supports the 6.0 version, while the last Ignition Gazebo release is 6.1[^1]. 

Waiting for a new release, you can install the project from its Nightly channel.

[^1]: https://github.com/robotology/gym-ignition/discussions/414#discussioncomment-1597672

diegoferigo, 2021-11-15:
I noted that you used conda to install the dependencies. We are in the process to package this project in conda-forge, if you're interested you can follow https://github.com/conda-forge/staged-recipes/pull/16582.

diegoferigo, 2021-11-16:
I'll keep this issue open so that it has visibility in case other users try to install the Stable channel with the most recent Ignition Gazebo. I'll close it as soon as a new release will be rolled out.

dnovischi, 2021-11-19:
That's probably for the best until migration. I can confirm that it works with nightly.

psh9002, 2022-08-10:
When I tried install with the Nightly version, there was `TypeError: metaclass conflict: the metaclass of a derived class must be a (non-strict) subclass of the metaclasses of all its bases` which is started from `gym_ignition_environments`.

How can I fix it?