robotology-legacy/gym-ignition #425: gym_ignition.scenario: ServerConfig and Component Warning Issues

dnovischi, 2021-11-29:
## Description:

Whenever I use `from gym_ignition.scenario import model_with_file, model_wrapper` to define a model class, e.g.

```python

from gym_ignition.scenario import model_with_file, model_wrapper

class CartPole(model_wrapper.ModelWrapper, model_with_file.ModelWithFile):
``` 

I get the following warnings from ignition-gazebo:

``` bash
[Wrn] [ServerConfig.cc:861] IGN_GAZEBO_SERVER_CONFIG_PATH set but no file found, no plugins loaded
[Wrn] [Component.hh:144] Trying to serialize component with data type [NSt6chrono8durationIlSt5ratioILl1ELl1000000000EEEE], which doesn't have `operator<<`. Component will not be serialized.
[Wrn] [Component.hh:144] Trying to serialize component with data type [N3sdf3v125WorldE], which doesn't have `operator<<`. Component will not be serialized.
[Wrn] [Component.hh:144] Trying to serialize component with data type [N8ignition4math2v63PIDE], which doesn't have `operator<<`. Component will not be serialized.
[Wrn] [Component.hh:144] Trying to serialize component with data type [N8scenario4core16JointControlModeE], which doesn't have `operator<<`. Component will not be serialized.
```

## Steps to reproduce

1. `sudo apt install ignition-fortress` 
1. `pip install --pre scenario gym-ignition`
1. `export IGN_GAZEBO_PHYSICS_ENGINE_PATH=/lib/x86_64-linux-gnu/ign-physics-5/engine-plugins`
1. Use code below:
```python
from typing import List
from gym_ignition.scenario import model_with_file, model_wrapper
from gym_ignition.utils.scenario import get_unique_model_name
from scenario import core as scenario

class CartPole(model_wrapper.ModelWrapper, model_with_file.ModelWithFile):
    def __init__(
        self,
        world: scenario.World,
        position: List[float] = (0.0, 0.0, 0.0),
        orientation: List[float] = (1.0, 0, 0, 0),
        model_file: str = None,
    ):
pass
```

## Environment

- OS: Ubuntu 20.04
- GPU: 1050Ti    
- Python: 3.8.X from conda env
- Version: 1.3.2.dev15
- Channel:
  - [ ] Stable
  - [x] Nightly
- Installation type:
  - [x] User
  - [ ] Developer

diegoferigo, 2021-11-29:
These warning are generated when the content of the entity component manager (ECM), that is the "database" containing all the simulation data, is serialized to be sent from the server to the GUI. In order to make ScenarIO work, we store custom components in the ECM, and some of them cannot be serialized. Serializing them would give no real benefit, it would just increase the communication overhead to the GUI.

This being said, in upstream there is an ongoing activity to run the server and the GUI in the same process. It will prevent us to run the GUI in a different process. It will also mean that the ECM will no longer be duplicated in both server and GUI, and there won't be any need to serialize data through network.

The warnings you read are just informative. I'm not sure that the Warning verbosity is what I would personally chose for this type of logging (maybe Info would be more appropriate). Though, this is something that we do not have control, it's a upstream policy. If they bother you, you can restrict the verbosity of the whole application to Error with:

https://github.com/robotology/gym-ignition/blob/fa14f7f8f667cf84bf226756cea158ec0f5c3d23/examples/panda_pick_and_place.py#L17-L18

dnovischi, 2021-11-29:
Thanks for the fast response and detailed explanation.