robotology-legacy/gym-ignition #428: TypeError: metaclass conflict: the metaclass of a derived class must be a (non-strict) subclass of the metaclasses of all its bases

psh9002, 2022-08-09:
## Description:

There is a TypeError when I type 'python panda_pick_and_place.py'.

## Steps to reproduce

1. Install Ignition Gazebo (Fortress) from source
2. Install gym-ignition from [https://robotology.github.io/gym-ignition/master/installation/nightly.html](url)

## Additional context
[image]

Traceback (most recent call last):
  File "panda_pick_and_place.py", line 7, in <module>
    import gym_ignition_environments
  File "/home/bak/venv/lib/python3.8/site-packages/gym_ignition_environments/__init__.py", line 8, in <module>
    from . import models, randomizers, tasks
  File "/home/bak/venv/lib/python3.8/site-packages/gym_ignition_environments/randomizers/__init__.py", line 5, in <module>
    from . import cartpole, cartpole_no_rand
  File "/home/bak/venv/lib/python3.8/site-packages/gym_ignition_environments/randomizers/cartpole.py", line 8, in <module>
    from gym_ignition import randomizers, utils
  File "/home/bak/venv/lib/python3.8/site-packages/gym_ignition/randomizers/__init__.py", line 5, in <module>
    from . import abc, gazebo_env_randomizer, model, physics
  File "/home/bak/venv/lib/python3.8/site-packages/gym_ignition/randomizers/abc.py", line 7, in <module>
    import gym_ignition.base.task
  File "/home/bak/venv/lib/python3.8/site-packages/gym_ignition/base/__init__.py", line 6, in <module>
    from . import runtime, task
  File "/home/bak/venv/lib/python3.8/site-packages/gym_ignition/base/runtime.py", line 11, in <module>
    class Runtime(gym.Env, abc.ABC):
TypeError: metaclass conflict: the metaclass of a derived class must be a (non-strict) subclass of the metaclasses of all its bases

<details>
<summary>Title</summary>

Content

</details>

## Environment

- OS: Ubuntu 20.04
- GPU: Nvidia GeForce 1080 
- Python: both 3.8.10 from 'virtualenv' and 'anaconda'
- Version: Nightly
- Channel:
  - [ ] Stable
  - [ v] Nightly
- Installation type:
  - [ v] User
  - [ ] Developer

psh9002, 2022-08-10:
I fix the problem by modifying `runtime.py` and `gazebo_env_randomizer.py`.

In `runtime.py`,
`# Copyright (C) 2019 Istituto Italiano di Tecnologia (IIT). All rights reserved.
# This software may be modified and distributed under the terms of the
# GNU Lesser General Public License v2.1 or any later version.

import abc

import gym
from gym_ignition.base.task import Task

# class Runtime(gym.Env, abc.ABC):  # Previous line
class Runtime(gym.Env, abc.ABCMeta):    # Modified line
    """
    Base class for defining executors of :py:class:`~gym_ignition.base.task.Task` objects.
..............`.

In `gazebo_env_randomizer.py`,
`
from gym_ignition.randomizers.abc import PhysicsRandomizer, TaskRandomizer
from gym_ignition.randomizers.physics import dart
from gym_ignition.runtimes import gazebo_runtime
from gym_ignition.utils import typing

MakeEnvCallable = Callable[[Optional[Dict]], gym.Env]

################# Added Line #################
class GazeboEnvRandomizer_Meta(type(gym.Wrapper), type(TaskRandomizer)):
    pass
################# Added Line #################

# class GazeboEnvRandomizer(gym.Wrapper, TaskRandomizer, abc.ABC):  # Previous line
class GazeboEnvRandomizer(gym.Wrapper, TaskRandomizer, metaclass=GazeboEnvRandomizer_Meta):   # Modified line
    """
    Base class to implement an environment randomizer for Ignition Gazebo.
...............................`.

psh9002, 2022-08-10:
However, there is another issue.
`[Err] [Conversions.cc:1757] Tried to convert SDF [world] into [plugin]
[Err] [Joint.cpp:756] The active joint control mode does not accept a velocity target
[Err] [Model.cpp:1259] Failed to set force of joint 'panda_joint1'
Traceback (most recent call last):
  File "/home/bak/Projects/ignition-robot/examples/manipulation.py", line 268, in <module>
    add_panda_controller(panda=panda, controller_period=gazebo.step_size())
  File "/home/bak/Projects/ignition-robot/examples/manipulation.py", line 52, in add_panda_controller
    assert panda.set_joint_velocity_targets(panda.joint_velocities())
AssertionError`

Gazebo Gui window was opened but, there is only empty space. It is also the same on the [pendulum example](https://robotology.github.io/gym-ignition/master/getting_started/scenario.html) at ScenarIO.

giu950, 2023-02-09:
Hello @psh9002,
I'm having the same issue, do you have any tips to solve it?

psh9002, 2023-02-21:
Hello @giu950 ,
Unfortunately no. Now, I'm not using Gazebo

DanielePucci, 2023-02-21:
CC @diegoferigo

diegoferigo, 2023-03-14:
Hi everyone, thanks for the feedback. I described the current status of the project in https://github.com/robotology/gym-ignition/issues/430 in case someone wants to start addressing the major problems affecting the current status of the project. 

Unfortunately there have been major changes of our main dependencies (both Ignition Gazebo and OpenAI Gym), and my research focus on the past 2 years shifted towards solutions providing parallelization on hardware accelerators like GPUs and TPUs. I'm not sure if there is anyone in AMI interested in the short term to pick up what's left, for sure in the close future I don't think I will find enough bandwidth for a proper upstream alignment, but I'd be happy to provide all the needed support and guidance.